arendst/Tasmota #8935: Using PowerDelta doesn't provide meaningful rule triggers

n1tehawk, 2020-07-19:
### PROBLEM DESCRIPTION

There doesn't seem to be a way to trigger rules from the _SENSOR_ messages that get generated when using `PowerDelta`. This came up in a Discord discussion between _sfromis_ and me (_NiteHawk_) on 2020/07/18 (search for "tele-ENERGY" by _sfromis_  or me).

The practical use case for this was inspired by _Euton_'s problem of wanting to send MQTT messages to Domoticz on (possibly variable and) shorter intervals than `TelePeriod`. To achieve this, Domoticz requires rules - publishling MQTT messages in a custom format to `domoticz/in`.

When trying to test this in conjunction with `PowerDelta`, I realized that there doesn't seem to be a way to trigger rules on the power changes taking place; i.e. rules on `tele-ENERGY#<x>` would not trigger as expected, and rules on `ENERGY#<x>` do of course trigger, but rather frequently (on every sensor update / polling cycle) and independent of `PowerDelta`.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): gosund SP111
- [X] Tasmota binary firmware version number used: _tasmota.bin_ v8.3.1.7
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: webUI firmware upgrade (file upload)
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
13:07:11 CMD: Backlog Template; Module; GPIO 255
13:07:11 MQT: stat/gosund1/RESULT = {"NAME":"Gosund SP111 2","GPIO":[57,0,56,0,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45}
13:07:11 MQT: stat/gosund1/RESULT = {"Module":{"0":"Gosund SP111 2"}}
13:07:11 MQT: stat/gosund1/RESULT = {"GPIO0":{"57":"Led2i"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"132":"HLWBL CF1"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
13:32:04 CMD: Backlog Rule1; Rule2; Rule3
13:32:04 MQT: stat/gosund1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":42,"Free":469,"Rules":"on tele-ENERGY#Power do VAR1 Gotcha. endon"}
13:32:04 MQT: stat/gosund1/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
13:32:05 MQT: stat/gosund1/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [X] Provide the output of this command: `Status 0`:
```
13:32:47 CMD: status 0
13:32:47 MQT: stat/gosund1/STATUS = {"Status":{"Module":0,"DeviceName":"gosund1","FriendlyName":["418EA7"],"Topic":"gosund1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:32:47 MQT: stat/gosund1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:33:03","StartupUTC":"2020-07-19T11:59:44","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2020-06-23T17:23:02","SaveCount":105,"SaveAddress":"FA000"}}
13:32:47 MQT: stat/gosund1/STATUS2 = {"StatusFWR":{"Version":"8.3.1.7(2c6198f-tasmota)","BuildDateTime":"2020-07-18T19:02:23","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"383/699"}}
13:32:47 MQT: stat/gosund1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MySSID",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008008","2805C8000100060000005A00000000000000","00000000","00006000"]}}
13:32:47 MQT: stat/gosund1/STATUS4 = {"StatusMEM":{"ProgramSize":591,"Free":412,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
13:32:47 MQT: stat/gosund1/STATUS5 = {"StatusNET":{"Hostname":"gosund1-3751","IPAddress":"192.168.0.207","Gateway":"192.168.0.250","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.250","Mac":"24:62:AB:41:8E:A7","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
13:32:47 MQT: stat/gosund1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_418EA7","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:32:47 MQT: stat/gosund1/STATUS7 = {"StatusTIM":{"UTC":"2020-07-19T12:32:47","Local":"2020-07-19T13:32:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:08","Sunset":"20:44"}}
13:32:47 MQT: stat/gosund1/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:32:47 MQT: stat/gosund1/STATUS10 = {"StatusSNS":{"Time":"2020-07-19T13:32:47","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.073,"Yesterday":0.004,"Today":0.005,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
13:32:47 MQT: stat/gosund1/STATUS11 = {"StatusSTS":{"Time":"2020-07-19T13:32:47","Uptime":"0T00:33:03","UptimeSec":1983,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WLAN-00025A","BSSId":"01:23:45:67:89:AB","Channel":13,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
13:40:00 CMD: powerdelta 102
13:40:00 MQT: stat/gosund1/RESULT = {"PowerDelta":102}
13:40:06 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-19T13:40:06","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.073,"Yesterday":0.004,"Today":0.006,"Power":2,"ApparentPower":20,"ReactivePower":20,"Factor":0.10,"Voltage":237,"Current":0.086}}
13:40:16 MQT: tele/gosund1/STATE = {"Time":"2020-07-19T13:40:16","Uptime":"0T00:40:32","UptimeSec":2432,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"MySSID","BSSId":"01:23:45:67:89:AB","Channel":13,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:05"}}
13:40:16 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-19T13:40:16","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.073,"Yesterday":0.004,"Today":0.006,"Period":0,"Power":3,"ApparentPower":29,"ReactivePower":29,"Factor":0.09,"Voltage":237,"Current":0.123}}
13:40:16 RUL: TELE-ENERGY#POWER performs "VAR1 Gotcha."
13:40:16 MQT: stat/gosund1/RESULT = {"Var1":"Gotcha."}
13:40:32 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-19T13:40:32","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.073,"Yesterday":0.004,"Today":0.006,"Power":2,"ApparentPower":21,"ReactivePower":21,"Factor":0.09,"Voltage":237,"Current":0.088}}
13:40:34 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-19T13:40:34","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.073,"Yesterday":0.004,"Today":0.006,"Power":5,"ApparentPower":23,"ReactivePower":23,"Factor":0.19,"Voltage":237,"Current":0.098}}
13:40:36 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-19T13:40:36","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.073,"Yesterday":0.004,"Today":0.006,"Power":2,"ApparentPower":22,"ReactivePower":22,"Factor":0.08,"Voltage":237,"Current":0.093}}
13:40:44 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-19T13:40:44","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.073,"Yesterday":0.004,"Today":0.006,"Power":4,"ApparentPower":21,"ReactivePower":21,"Factor":0.20,"Voltage":237,"Current":0. [... cut ...]
* There should be _some_ way to trigger rules on PowerDelta events (not necessarily by `tele-ENERGY#<x>`).
* If current Tasmota already has a way to achieve this, it seems at least to be lacking the documentation for it.
* Alternative implementations may be possible. For example, it's obvious that the _PowerDelta_ messages have no `Period` field. Maybe it's feasible to include a `PowerDelta` field instead, and allow a (JSON) trigger on `ENERGY#PowerDelta` that way?

### SCREENSHOTS
_N/A_

### ADDITIONAL CONTEXT
_N/A_

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-19:
Looking at my code written long long time ago I see that a tele-Energy is reported if power delta is triggered. 

Your issue is probably related to:

- Way to low power differences (either 2W or 4W max loads makes calculating power delta very difficult as it almost equals the noise measured. As backgound information you might want to know that power delta triggers only of three samples have been taken which in turn can take several seconds on low loads simply because low power loads take several seconds to measure by the hardware used)
- You might want to use percentage instead of actual power differences. See PowerDelta command for correct syntax.

n1tehawk, 2020-07-19:
Hello arendst!

Thank you for your reply and the additional input. I have repeated the experiment with a `tele-Energy#Power` rule, this time using a halogen lamp (50W, power factor ~1.0) and `PowerDelta 145` (a 45W "delta").
Switching the lamp on and off, I can reliably trigger the *PowerDelta* messages - but the rule still doesn't fire. Can you suggest a different/specific event to use as a trigger? (I tried `tele-Energy#Factor` as an alternative without success.)

`weblog` didn't provide any additional insight into what's going on (just an additional `WIF: Checking connection...` on the console).

arendst, 2020-07-20:
Verified. Investigating...

arendst, 2020-07-20:
Pls try latest commit where a valid would be:

``on PowerDelta do VAR1 Gotcha2 endon``

so without the ``tele-`` part resulting in 
```
16:23:35 MQT: tele/shp10/MARGINS = {"PowerDelta":5,"PowerHigh":"ON"}
16:23:35 RUL: POWERDELTA performs "VAR1 Gotcha2"
16:23:35 MQT: stat/shp10/RESULT = {"Var1":"Gotcha2"}
16:23:35 MQT: tele/shp10/SENSOR = {"Time":"2020-07-20T16:23:35","ENERGY":{"TotalStartTime":"2020-06-07T14:29:21","Total":1.655,"Yesterday":0.070,"Today":0.005,"Power":5.28,"ApparentPower":9.89,"ReactivePower":8.40,"Factor":0.53,"Voltage":231.7,"Current":0.043,"BL0940":{"Temperature":44.7}}}
16:23:43 MQT: tele/shp10/MARGINS = {"PowerDelta":-5,"PowerHigh":"OFF"}
16:23:43 RUL: POWERDELTA performs "VAR1 Gotcha2"
16:23:43 MQT: stat/shp10/RESULT = {"Var1":"Gotcha2"}
16:23:43 MQT: tele/shp10/SENSOR = {"Time":"2020-07-20T16:23:43","ENERGY":{"TotalStartTime":"2020-06-07T14:29:21","Total":1.655,"Yesterday":0.070,"Today":0.005,"Power":0.00,"ApparentPower":0.00,"ReactivePower":0.00,"Factor":0.00,"Voltage":231.9,"Current":0.000,"BL0940":{"Temperature":44.7}}}
```

arendst, 2020-07-20:
Reconsidering the last commit I'll change it so it will be reported as

```
16:23:35 MQT: tele/shp10/MARGINS = {"MARGINS":{"PowerDelta":5,"PowerHigh":"ON"}}
```
This solves issues when using rules and commands like PowerDelta.

Hold on...

arendst, 2020-07-20:
So the latest commit triggers on a rule like

``on Margins#PowerDelta do VAR1 Gotcha2 endon``

n1tehawk, 2020-07-20:
Thanks for looking into this. As it seemed relevant, I've been testing the new logic with commit a84ebe55:
```
18:29:07 CMD: status 2
18:29:07 MQT: stat/gosund1/STATUS2 = {"StatusFWR":{"Version":"8.3.1.7(a84ebe5-tasmota)","BuildDateTime":"2020-07-20T18:00:52","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"383/699"}}
18:29:10 CMD: rule1 on tele-Energy#Power do Var1 TelePeriod endon on Margins#PowerDelta do Var1 PowerDelta endon
18:29:10 RUL: Stored uncompressed, would compress from 92 to 53 (-42%)
18:29:10 MQT: stat/gosund1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":92,"Free":419,"Rules":"on tele-Energy#Power do Var1 TelePeriod endon on Margins#PowerDelta do Var1 PowerDelta endon"}
18:29:14 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-20T18:29:14","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.096,"Yesterday":0.027,"Today":0.001,"Period":0,"Power":55,"ApparentPower":55,"ReactivePower":0,"Factor":1.00,"Voltage":236,"Current":0.232}}
18:29:14 RUL: TELE-ENERGY#POWER performs "Var1 TelePeriod"
18:29:14 MQT: stat/gosund1/RESULT = {"Var1":"TelePeriod"}
18:29:38 MQT: tele/gosund1/MARGINS = {"MARGINS":{"PowerDelta":61}}
18:29:38 RUL: MARGINS#POWERDELTA performs "Var1 PowerDelta"
18:29:38 MQT: stat/gosund1/RESULT = {"Var1":"PowerDelta"}
18:29:38 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-20T18:29:38","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.096,"Yesterday":0.027,"Today":0.001,"Power":26,"ApparentPower":26,"ReactivePower":0,"Factor":1.00,"Voltage":236,"Current":0.110}}
18:29:44 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-20T18:29:44","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.096,"Yesterday":0.027,"Today":0.001,"Period":0,"Power":64,"ApparentPower":64,"ReactivePower":0,"Factor":1.00,"Voltage":236,"Current":0.271}}
18:29:44 RUL: TELE-ENERGY#POWER performs "Var1 TelePeriod"
18:29:44 MQT: stat/gosund1/RESULT = {"Var1":"TelePeriod"}
18:29:56 MQT: tele/gosund1/MARGINS = {"MARGINS":{"PowerDelta":-56}}
18:29:56 RUL: MARGINS#POWERDELTA performs "Var1 PowerDelta"
18:29:56 MQT: stat/gosund1/RESULT = {"Var1":"PowerDelta"}
18:29:56 MQT: tele/gosund1/SENSOR = {"Time":"2020-07-20T18:29:56","ENERGY":{"TotalStartTime":"2020-06-23T17:08:40","Total":0.096,"Yesterday":0.027,"Today":0.001,"Power":2,"ApparentPower":44,"ReactivePower":44,"Factor":0.05,"Voltage":234,"Current":0.189}}
```
The console is reporting the new `MARGINS` message, and an associated rule triggers as expected. Well done.
From my perspective, the original problem is resolved. Feel free to close the issue as you see fit, otherwise I'll probably do that in a few days.

Regards, NiteHawk

closed 2020-07-20 by ascillato2