arendst/Tasmota #8936: Featrue request for tuya-convert users.

tonyfitzs, 2020-07-19:
First let me say, I really love what you have done here fantastic project great work. But I would like to make a request to help support those of us who are using tuya-convert (which is a large community). 

One of the challenges we face using the tuya based hardware is the hardware is built to a budget and can sometimes have unknown faults. Because the process to install Tasmota using tuya-convert recommends NOT initially connecting the unit to the Tuya app as this may make it impossible to update, you don’t discover any issues until it’s too late and have installed the new firmware.

The problem then is you can not return the device under warrantee because it no longer has the original firmware. So you need to roll the firmware back first. The only method I have seen to do this is not helpful in the use case I am describing because it relies on the user rolling back the original dumped firmware via a serial connection.

The challenge here is that not all the Tuya devices allow you to do this easily, so you may need to modify the internal circuit board (at least solder some connections) to achieve this. The problem here is that just as upgrading the firmware makes it challenging to return, modifying the hardware is even worse. 

So this got me thinking. It would be great if there were a way to install OTA a firmware stub that doesn’t do anything other than allow tuya-convert to connect to the device.

So what I mean in more detail. After updating the device to Tasmota, you use OTA support to load a firmware shell onto the device that just turns on the WIFI pair mode the same as if it was getting ready to receive the firmware from tuya-convert. 
Then in tuya-convert, you connect to the device but rather than deploy Tasmota you use the original dump file back to the device. This would effectively enable the capability to re-flash the OG firmware OTA thereby solve the challenge of what to do when you need to roll back the firmware for any reason. 

I would love to help out in any way I can, but I would not have a clue where to start. Thanks for your indulgence.

closed 2020-07-19 by close-issue-app[bot]

ascillato, 2020-07-20:
Hi,

We don't manage tuya convert. Please, address your request to them. They should know how to help you. Thanks.

tonyfitzs, 2020-07-20:
Hi Adrian, 
I appreciate your response, my apologies for not being more clear in my request, I have also sent more or less the same request to tuya-convert as well as to ESPurna, so let me try to explain more clearly why this is a correct request for Tasmota in this case.
While your correct based on the title, my request relates to tuya-convert, my request applies to the firmware part of the process, not the firmware delivery. 

Tuya-convert is a firmware delivery process, whereas both Tasmota and ESPurna are actually firmware. 
For what I am suggesting to work, it would require a change on the firmware side. **Not** the firmware delivery side so much. That's because my suggestion only impacts tuya devices that **ALREADY** have Tasmota or ESPurna firmware installed, and the only way to roll them back would be to re-apply the firmware via serial connection or some type of OTA method. 

What I am requesting is a very small firmware package that has the bare minimum code to enable a tuya device to boot up in pair mode to allow tuya-convert to connect so you can re-deploy the original firmware. 

I hope that is more clear. If this case remains closed, I will re-open it again and rewrite until I am sure my request has been read and fully understood. If what I am requesting is impossible, that's understandable, but I hope it isn't. 

I have very little experience in developing firmware, but I am willing to take this on as a project of my own if I have too. But would prefer not to.