arendst/Tasmota #8938: IRSend Issues Greater than 32bits

t3h-n00b, 2020-07-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Using Tasmota 8.3.1, with both an ircustom.bin and ir.bin firmware files.

I have noticed that when sending IR messages greater than 32 bits (Samsung36 or Pioneer), the end device is unable to read the message.

Interestingly, another Tasmota device is able to read/decode the message successfully.

If I change SetOption57 to 1 and then send the RAW IR command via IRSend, the command **is** successfully read by the end device (and a second Tasmota device reads/decodes this message the same as the first one)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [Y] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [Y] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [Y] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [Y] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos D1 Mini / D1R2 with TSAL6400/TSOP38536
- [ ] Tasmota binary firmware version number used: 8.3.1
  - [Y] Pre-compiled
  - [Y] Self-compiled
    - [ ] IDE / Compiler used: Visual Studio Code
- [ ] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
01:57:42 MQT: tasmota/stat/room-XXXX/RESULT = {"NAME":"Whole Room Controller","GPIO":[0,0,0,0,106,105,0,0,51,2,8,0,0],"FLAG":0,"BASE":18}
01:57:42 MQT: tasmota/stat/room-XXXX/RESULT = {"Module":{"0":"Whole Room Controller"}}
01:57:42 MQT: tasmota/stat/room-XXXX/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"106":"RFrecv"},"GPIO5":{"105":"RFSend"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"51":"IRrecv"},"GPIO13":{"2":"AM2301"},"GPIO14":{"8":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
Not Applicable

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
01:55:33 CMD: status 0
01:55:33 MQT: tasmota/stat/room-XXXX/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"room-XXXX","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:55:33 MQT: tasmota/stat/room-XXXX/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T01:55:34","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"1970-01-01T00:00:00","SaveCount":13,"SaveAddress":"F7000"}}
01:55:33 MQT: tasmota/stat/room-XXXX/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-07-19T20:56:33","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"403/699"}}
01:55:33 MQT: tasmota/stat/room-XXXX/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["44S_IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000100","00000000"]}}
01:55:33 MQT: tasmota/stat/room-XXXX/STATUS4 = {"StatusMEM":{"ProgramSize":617,"Free":384,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","8FDAE797","043783A1","000000CD","010093C0","D030F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,17,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,37,58,59,63"}}
01:55:33 MQT: tasmota/stat/room-XXXX/STATUS5 = {"StatusNET":{"Hostname":"room-XXXX-2599","IPAddress":"192.168.Y.Y","Gateway":"192.168.Y.Y","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"XXXXX,"Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
01:55:33 MQT: tasmota/stat/room-XXXX/STATUS6 = {"StatusMQT":{"MqttHost":"XXXXX","MqttPort":1883,"MqttClientMask":"room-%06X","MqttClient":"room-XXXX","MqttUser":"XXXXX","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
01:55:33 MQT: tasmota/stat/room-7B0A27/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T01:55:33","Local":"1970-01-01T01:55:33","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
01:55:33 MQT: tasmota/stat/room-XXXX/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T01:55:33","AM2301":{"Temperature":20.5,"Humidity":48.4,"DewPoint":9.2},"TempUnit":"C"}}
01:55:33 MQT: tasmota/stat/room-XXXX/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T01:55:33","Uptime":"0T01:55:34","UptimeSec":6934,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"XXXXX","BSSId":"02:9F:C2:34:08:69","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Example below is the MUTE command from a Samsung Sound Bar. Also experience similar with Pioneer devices.

1. Capture IR command from Samsung/Pioneer remote
Example capture from original remote:
`{"IrReceived":{"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF008E71","DataLSB":"0x0F300718E","Repeat":0,"RawData":[4544,4458,500,498,500,500,496,496,500,498,500,1464,530,1464,522,502,494,502,496,1470,524,1470,504,1464,522,1476,520,500,496,500,498,500,498,498,500,4460,530,492,494,504,496,500,496,500,496,1472,524,500,498,498,500,498,468,1474,522,1472,524,1470,526,498,500,498,500,1468,528,1466,530,1464,500,524,474,498,500,500,498,1466,530],"RawDataInfo":[77,77,0]}}`

[... 15 lines cut ...]
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

It is worth noting that all 32 bit commands captured/resent appear to work fine. IRHvac commands also seem fine - just IRSend messsages >32 bits that dont.

Sample known bad commands:
```
Samsung_Soundbar_PWR {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF000EF1","DataLSB":"0x0F300708F","Repeat":0}
Samsung_Soundbar_DIN {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF00C936","DataLSB":"0x0F300936C","Repeat":0}
Samsung_Soundbar_Up {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF00EE11","DataLSB":"0x0F3007788","Repeat":0}
Samsung_Soundbar_Down {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF001EE1","DataLSB":"0x0F3007887","Repeat":0}
Samsung_Soundbar_Mute {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF008E71","DataLSB":"0x0F300718E","Repeat":0}
AV_Power_On {"Protocol":"PIONEER","Bits":64,"Data":"0xA55A58A7A55A58A7","DataLSB":"0xA55A1AE5A55A1AE5","Repeat":0}
AV_Power_Off {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AD827A55AD827","DataLSB":"0xA55A1BE4A55A1BE4","Repeat":0}
AV_Sat_Cable {"Protocol":"PIONEER","Bits":64,"Data":"0xA55A08F7A55A08F7","DataLSB":"0xA55A10EFA55A10EF","Repeat":0}
AV_Vol_Up {"Protocol":"PIONEER","Bits":64,"Data":"0xA55A50AFA55A50AF","DataLSB":"0xA55A0AF5A55A0AF5","Repeat":0}
AV_Vol_Down {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AD02FA55AD02F","DataLSB":"0xA55A0BF4A55A0BF4","Repeat":0}
AV_Bass_Up {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AEA15A55A00FF","DataLSB":"0xA55A57A8A55A00FF","Repeat":0}
AV_Bass_Down {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AEA15A55A807F","DataLSB":"0xA55A57A8A55A01FE","Repeat":0}

```
**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-07-20:
Sending IR messages is supposed to work up to 64 bits.

Since the `SAMSUNG36` message is correctly received by another Tasmota, I believe the problem is in the `IRRemoteESP8266` lib which encodes differently the message. Looking at raw messages, I don't see any difference, except that the pulses are longer in the Tasmota generated (1650 us instead of 1460 us), which can explain why it is wrongly interpreted.

I suggest you open an issue here: https://github.com/crankyoldgit/IRremoteESP8266

Can you also give more data about the PIONEER issue?

t3h-n00b, 2020-07-20:
Thanks for the response. Will create an issue there as well.

For Pioneer, example commands:
Pioneer Power On (From Remote):
`{"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0xA55A58A7A55A58A7","DataLSB":"0xA55A1AE5A55A1AE5","Repeat":0,"RawData":[8504,4196,588,1536,562,464,578,1544,586,464,578,470,584,1540,580,470,588,1536,580,470,584,1514,582,470,582,1538,582,1542,588,462,580,1542,554,496,558,466,588,1536,582,466,586,1538,582,1542,588,462,580,468,584,466,588,1512,584,464,578,1546,586,464,580,470,586,1538,584,1540,560,1538,582,25190,8508,4190,586,1538,584,466,576,1546,552,496,560,466,578,1546,588,462,582,1542,580,468,586,1538,584,464,580,1518,580,1544,578,472,582,1542,582,468,586,462,582,1542,556,492,564,1536,586,1538,584,466,578,472,586,462,578,1546,586,462,582,1518,582 ...`

From Tasmota-sent command:
`{"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0xA55A58A7A55A58A7","DataLSB":"0xA55A1AE5A55A1AE5","Repeat":0,"RawData":[8520,4282,564,1590,556,520,558,1586,556,518,580,496,560,1584,560,514,564,1580,552,522,554,1590,554,520,556,1588,556,1588,558,518,560,1584,560,512,554,520,558,1586,560,510,556,1588,556,1588,556,514,564,512,554,522,554,1590,560,516,562,1582,558,512,554,522,554,1584,558,1580,564,1580,564,25210,8580,4242,558,1586,560,516,562,1582,564,512,554,520,556,1588,556,518,558,1586,558,518,560,1584,562,514,562,1582,562,1580,564,506,560,1582,562,510,556,520,558,1586,558,514,562,1576,558,1588,558,514,564,512,554,524,564,1576,556,518,558,1586,560 ...
`
(I'll work on getting the full dump that isn't cut off in the console buffer)

t3h-n00b, 2020-07-20:
Updated above comment with full Pioneer commands

PR #8992 (2020-07-29): Change IRRemoteESP8266 IR lib to pre-2.7.9, fixing Samsung and Pioneer protocols (#8938)
## Description:

Pre-2.7.9 version of IR lib

**Related issue (if applicable):** fixes #8938

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.2.1
  - [ ] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor- [... cut]
closed 2020-07-29 by arendst