arendst/Tasmota #8939: Featrue request for tuya-convert users. THIS IS A FIRMWARE REQUEST!

tonyfitzs, 2020-07-20:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

### **The is a Re Opening Ticket as the Bot closed it**

Also, people may only read the title and decide I have posted a tuya-convert in the wrong place.   **I HAVNT !**

### **First, what am I actually asking for!**
Basically a very small firmware package that has the bare minimum code to enable a tuya device to boot up in pair mode to allow tuya-convert to connect so you can re-deploy the original firmware. 

### **Second Why am I asking for it?**
One of the challenges we face using the tuya based hardware is the hardware is built to a budget and can sometimes have unknown faults. Because the process to install Tasmota using tuya-convert recommends **NOT** initially connecting the unit to the Tuya app as this may make it impossible to update, you don't discover any issues until it's too late and has installed the new firmware.

The problem then is you can not return the device under warrantee because it no longer has the original firmware. So you need to roll the firmware back first. The only method I have seen to do this is not helpful in the use case I am describing because it relies on the user rolling back the original dumped firmware via a serial connection.

The challenge here is that not all the Tuya devices allow you to do this easily, so you may need to modify the internal circuit board (at least solder some connections) to achieve this. The problem here is that just as upgrading the firmware makes it challenging to return, modifying the hardware is even worse.

So this got me thinking. It would be great if there were a way to install OTA a firmware stub that doesn't do anything other than allow tuya-convert to connect to the device.

So what I mean in more detail. After updating the device to Tasmota, you use OTA support to load a firmware shell onto the device that just turns on the WIFI pair mode the same as if it was getting ready to receive the firmware from tuya-convert.
Then in tuya-convert, you connect to the device but rather than deploy Tasmota you use the original dump file back to the device. This would effectively enable the capability to re-flash the OG firmware OTA thereby solve the challenge of what to do when you need to roll back the firmware for any reason.

### **Who else have I sent the same request too?**
Just so its absolutely clear, I have also sent more or less the same request to tuya-convert as well as to ESPurna, I expect that there would need to be some level of collaboration between the Tasmota Dev folks and the tuya-convert dev folks. But based on my very limited understanding of what's involved here,

I think more of the heavy lifting os on the firmware side. 

Tuya-convert is a firmware delivery process, whereas both Tasmota and ESPurna are actually firmware. For what I am suggesting to work, it would require a change on the firmware side, Not so much the firmware delivery side.

That's because my suggestion only impacts tuya devices that **ALREADY** have Tasmota or ESPurna firmware installed, and the only way to roll them back would be to re-apply the firmware via serial connection or some type of OTA method.

I hope that is more clear. Just in case a member of the development community is tempted to close this request without fully understanding it.

Jason2866, 2020-07-20:
And which features should be in this TUYA build? There are so many different TUYA devices. 
Building a firmware to statisfy all TUYA wishes will be greater than 500k.

Solutions:
- Use for the Tuya-Convert the lite Tasmota build. This version is < 500k and can be updated to any Tasmota version. 
  There is no need for other builds.
- Compile your build you want / need yourself

tonyfitzs, 2020-07-20:
Hi Jason, thanks for the responce, but its still clear that what I am asking is not understood. the firmware should have NO feature! its just a stub so that Tuya Convert can connect to it to role back the firmware to the stock firmware.

Jason2866, 2020-07-20:
Tasmota crew will not provide such a feature. Reason for. It is wasting time. Sooner or layter Tuya exploit will be fixed.
That means all invested time in this feature is lost. Tasmotas goal is NOT to do programming work for closed commercial software.

tonyfitzs, 2020-07-20:
fare enough.

ascillato2, 2020-07-20:
For making such intermediate firmware you are asking for, it is needed a lot of information from the original tuya firmware (memory distribution, etc etc) in order to make a working OTA.
The people that knows that, are the guys from tuya convert. That is why It is better you ask them to make this new tuya-unconvert firmware. They should know how to do that.
Sorry we can't help you on this.

closed 2020-07-20 by ascillato2