arendst/Tasmota #8946: Dimmer Decoupling

alexvenom, 2020-07-20:
**Have you looked for this feature in other issues and in the docs?**
Yes

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
I have a Moes DSC01 dimmer switch and a Yeelight 1S Smart Bulb. When the dimmer goes below a certain level, the bulb starts flickering.
Having the bulb "always on" is not an option, so I have to use the physical switch to turn it on and off, but I'd like the dimmer to be decoupled so I could dim up and down using MQTT.

**Describe the solution you'd like**
An option for decoupling the dimmer so I can:

- operate the physical dimmer and reflect on MQTT, but no voltage change on the line
- operate set the MQTT value for the dimmer and reflect on the physical dimmer, but no voltage change on the line
- keep the on/off switch relay working

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Tried Groups and kind works, but the bulb has to be "always on", which is not an option here.

**Additional context**
_Add any other context or screenshots about the feature request here._
TuyaMCU Docs Page: https://tasmota.github.io/docs/TuyaMCU/#fnid
(Please, remember to close the issue when the problem has been addressed)

digiblur, 2020-07-21:
I don't see that being possible with the TuyaMCU as it controls the relay and dimming.

blakadder, 2020-07-22:
In your place I'd disconnect live from the dimmer and use software dimming since you're operating a smart bulb.

alexvenom, 2020-07-22:
That is not an option here due to the law. I cannot have a light socket always energized.
Decoupling the dimmer would be ideal. I know I can decouple the main relay in some Tasmotized switches. I guess if it can be done, it would help everyone that has to conform to this law.

digiblur, 2020-07-22:
You would need a dimmer without a TuyaMCU then.

stefanbode, 2020-07-24:
I just recently submitted setoption99 1 to use a AC Dimmer with Zero-Crossing. Search for setoption99 in the documentation. Maybe this helps in your case. Anyhow the TuyaMCU is at it is. No way to control it further.

closed 2020-07-25 by ascillato2