arendst/Tasmota #8948: Ping improvement - dns resolution

kevinkahn, 2020-07-21:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Currently the ping command will accept a symbolic dns name as well as an IP address.  This is great for targeting a well known internet server to check connectivity (presumably the motivation for the command).  However, in the check for success you will need to use the IP address that the name resolved to which defeats using a well known name.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I can imaging a number of solutions.  The simplest would be to keep the symbolic name in the rule report so that it will match in the ping#xxx#success==1 check.  An alternative might be to have a way to default the check to the "last" ping although this would not work with the multiple ping enhancement that was made.  (I could live with that but it wouldn't be a clean solution.)  A third possibility might be to allow a separate name to ip resolution to a var that would then be used in the ping and the check.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Only current approach is to pre-resolve the well known name but this has the real risk of that address changing.
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2020-07-21:
might also not work with CNAME DNS entries or round robin DNS

s-hadinger, 2020-07-21:
Good suggestion, I will look at it. It should be a low impact change. Indeed adding a `String` to the `Ping_t` to save the original hostname.

There should not be any issue with round-robin, the response always comes back from the same IP that received the request, otherwise firewalls would block it.

PR #8955 (2020-07-21): Change ``Ping`` now reports the hostname instead of IP address (#8948)
## Description:

`Ping` now reports the hostname as key and IP address in `IP` field.

Example:
```
Ping tasmota.github.io

xx:xx:xx CMD: Ping tasmota.github.io
xx:xx:xx RSL: stat/tasmota_xxxxxx/RESULT = {"Ping":"Done"}
xx:xx:xx RSL: tele/tasmota_xxxxxx/RESULT = {"Ping":{"tasmota.github.io":{"Reachable":true,"IP":"185.199.110.153","Success":1,"Timeout":0,"MinTime":9,"MaxTime":9,"AvgTime":9}}}
```

For numerical address, the result is similar:
```
Ping 192.168.1.1

xx:xx:xx CMD: Ping 192.168.1.1
xx:xx:xx RSL: stat/tasmota_xx [... cut]
closed 2020-07-21 by arendst