arendst/Tasmota #8958: Request For FSLP (Force Sensing Linear Potentiometer) sensor

lacherni14, 2020-07-22:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Add the FSLP sensor (Force Sensing Linear Potentiometer) from Interlink
[Datasheet_FSLP.pdf](https://github.com/arendst/Tasmota/files/4959617/Datasheet_FSLP.pdf)

**Describe alternatives you've considered**  
Using TasmotaSlave with an Arduino to integrate the sensor. This way of integrating the sensor turned out to be slow.

**Additional context**  
_Add any other context or screenshots about the feature request here._
The Interlink Electronics FSLP Sensor can measure position and pressure. The
connection to the measuring microprocessor is very simple and requires only one
external component. The microprocessor will need two general purpose IO (GPIO)
pins and two GPIO/analog-to-digital converter (ADC) pins. The GPIOs should be
able to go into high-impedance mode (>1MΩ) and the ADCs should be able to
measure from 0 to Vcc.

Integration Guid:
https://www.pololu.com/file/0J750/FSLP-Integration-Guide-13.pdf
[FSLP-Integration-Guide-13.pdf](https://github.com/arendst/Tasmota/files/4959647/FSLP-Integration-Guide-13.pdf)

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-07-22:
> The microprocessor will need two general purpose IO (GPIO)
pins and two GPIO/analog-to-digital converter (ADC) pins. The GPIOs should be
able to go into high-impedance mode (>1MΩ) and the ADCs should be able to
measure from 0 to Vcc.

Tasmota is a firmware for the esp8266 chip and that chip has only one ADC. That sensor needs 2 adc

lacherni14, 2020-07-22:
> > The microprocessor will need two general purpose IO (GPIO)
> > pins and two GPIO/analog-to-digital converter (ADC) pins. The GPIOs should be
> > able to go into high-impedance mode (>1MΩ) and the ADCs should be able to
> > measure from 0 to Vcc.
> 
> Tasmota is a firmware for the esp8266 chip and this wifi chip has only one ADC. The FDLP sensor needs 2 ADC pins.

Therefore it is necessary to use an external ADC. The necessary is already integrated in Tasmota