arendst/Tasmota #8961: ESP32 | Sensor SHT1x not detect

NetHans, 2020-07-22:
### PROBLEM DESCRIPTION
The SHT1x sensor no longer works with the ESP32. This is no longer detect. On an ESP8266, on the other hand, it works correctly. Other I2C devices are detected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32
- [ ] Tasmota binary firmware version number used: v8.3.1.7
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Visual Studio Code / PlatformIO
- [x] Flashing tools used: PlatformIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,640,608,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
{"Module":{"1":"ESP32-DevKit"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"416":"PWM1"},"GPIO13":{"0":"None"},"GPIO14":{"417":"PWM2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"33":"Button2"},"GPIO19":{"0":"None"},"GPIO21":{"640":"I2C SDA"},"GPIO22":{"608":"I2C SCL"},"GPIO23":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"418":"PWM3"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
{"Time":"2020-07-22T17:11:49","Uptime":"0T00:33:31","UptimeSec":2011,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"xxxxx","Channel":xx,"RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:05"}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-23:
The SHT1x is not really an I2C device. In fact it's a pain in the ass device as it violates the I2C protocol leading to I2C issues if real I2C devices are connected.

For the ESP8266 direct digitalWrites are used to get the thing working in some way. As timing on an ESP32 is different I suspect that's the main reason it doesn't work.

Any volunteer willing to solve be my guest.

arendst, 2020-07-23:
Nice starting point here https://github.com/espressif/arduino-esp32/issues/1064

Jason2866, 2020-07-23:
The Digital Humidity Sensor SHT1x is End of Life.

closed 2020-07-31 by ascillato2