arendst/Tasmota #8965: MCCP23017 outputs and DS18B20 double telemetry

hebdam, 2020-07-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-07-23:
Dont know if this behaviour can or will be changed. You can solve with a more specific JSON parsing in OpenHab.

sfromis, 2020-07-23:
Yeah, when I parse /SENSOR payloads, I check if the JSON contain the object for the specific sensor before trying to access keys for individual values. Which is basically as designed when all sensors get their readings published under the same topic.

arendst, 2020-07-23:
How about I append the output data to the first telemetry JSOn message where it should have been all the time anyway like:

```
10:17:27 MQT: tele/garage_control/SENSOR = {"Time":"2020-07-23T10:17:27","DS18B20":{"Id":"020A92464368","Temperature":22.31},"MCP230XX":{"D0":0,"D1":0,"D2":1,"D3":1,"D4":0,"D5":0,"D6":0,"D7":0,"D8":1,"D9":1,"D10":1,"D11":1,"D12":0,"D13":0,"D14":0,"D15":0},"MCP230_OUT":{"OUT_D0":"OFF","OUT_D2":"OFF","OUT_D3":"OFF","OUT_D4":"OFF","OUT_D5":"OFF","OUT_D6":"OFF","OUT_D7":"OFF","END":1},"TempUnit":"C"}
```

hebdam, 2020-07-23:
**How about I append the output data to the first telemetry JSOn message where it should have been all the time anyway like:**

Yes that looks great to me as  one time telemetry is easier to handle (at least for me)

My apology to bug you but I just noticed similar situation with status reporting of outputs and inputs. Both output
stat/garage_control/RESULT

so output switching looks like:
13:52:39 CMD: sensor29 5,0
13:52:39 MQT: stat/garage_control/RESULT = {"S29cmnd_D5":{"COMMAND":"OFF","STATE":"OFF"}

and Input 
MQT: stat/garage_control/RESULT = {"Time":"2020-07-19T16:08:46","MCP230XX_INT":{"D0":1,"MS":18101}}
Here again OPENHAB recieves MQTT /RESULT and confuses transformation path one for the other. It's just a heads up for you. Perhaps I have to find a way on OPENHAB side but at this moment I have no clue. In OPENHAB I specify ITEMS or THINGS with JSON transformation pattern to read status of both input or output.
This feature I/O expansion with MCP23017 is a great tool and it works great for my remote house. I'm still testing it before I use it for my gate and garage open, and other things. Must make sure that my house  doesn't end up to be wide opened without me knowing it :-)

closed 2020-07-23 by hebdam