arendst/Tasmota #8968: TSL2591 shows -1 when illuminance is over about 2600 lux

Ingenieur89, 2020-07-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
TSL2591 shows -1 when illuminance is over about 2600 lux

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
16:05:52 RSL: stat/tasmota_972054/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
16:05:52 RSL: stat/tasmota_972054/RESULT = {"Module":{"1":"ESP32-DevKit"}}
16:05:52 RSL: stat/tasmota_972054/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO21":{"640":"I2C SDA"},"GPIO22":{"608":"I2C SCL"},"GPIO23":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
16:06:21 RSL: stat/tasmota_972054/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_972054","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:06:21 RSL: stat/tasmota_972054/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:03:51","StartupUTC":"2020-07-23T15:02:30","Sleep":50,"CfgHolder":4617,"BootCount":88,"BCResetTime":"2020-07-19T09:38:08","SaveCount":161}}
16:06:21 RSL: stat/tasmota_972054/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-21T19:56:47","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":80,"Hardware":"ESP32","CR":"338/699"}}
16:06:21 RSL: stat/tasmota_972054/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PWLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000"]}}
16:06:21 RSL: stat/tasmota_972054/STATUS4 = {"StatusMEM":{"ProgramSize":1067,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CD","010013C0","40080181","40004000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,82","Sensors":"1,2,3,5,6,57"}}
16:06:21 RSL: stat/tasmota_972054/STATUS5 = {"StatusNET":{"Hostname":"tasmota_972054-0084","IPAddress":"192.168.178.60","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"30:AE:A4:97:20:54","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:06:21 RSL: stat/tasmota_972054/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_972054","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:06:21 RSL: stat/tasmota_972054/STATUS7 = {"StatusTIM":{"UTC":"2020-07-23T15:06:21","Local":"2020-07-23T16:06:21","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:12","Sunset":"20:40"}}
16:06:21 RSL: stat/tasmota_972054/STATUS10 = {"StatusSNS":{"Time":"2020-07-23T16:06:21","TSL2591":{"Illuminance":   36.167}}}
16:06:21 RSL: stat/tasmota_972054/STATUS11 = {"StatusSTS":{"Time":"2020-07-23T16:06:21","Uptime":"0T00:03:51","UptimeSec":231,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":85,"MqttCount":0,"Wifi":{"AP":1,"SSId":"PWLAN","BSSId":"CC:CE:1E:BB:E2:4C","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect a TSL2591 to an ESP32 via i2c and shine a flashlight at the sensor until the lux value increases over about 2600 lux. After that the value will be -1. Normally the sensor should work up to 88000 lux.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The value should increase up to 88000 lux and maybe after that show -1.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Maybe an overflow issue? The sensor shows 3 decimals.
[image]

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-23:
I guess it has to do with the chosen default gain/timing range of gain 25x and timing range 300 millis.

As these parameters are currently fixed the only solution is changing them yourselves in file ``xsns_57_tls2591.ino`` lines 47 and 48

Remember that due to these parameters you will never get a reading from 188uLux to 88000Lux with one set of gain/timing.

Ingenieur89, 2020-07-23:
I set it to

Adafruit_TSL2591.cpp (lines 115-124)
```
Adafruit_TSL2591::Adafruit_TSL2591()
{
  _initialized = false;
  _integration = TSL2591_INTEGRATIONTIME_100MS;
  _gain        = TSL2591_GAIN_LOW;

  // we cant do wire initialization till later, because we havent loaded Wire yet
}
```
but I still get values over 2600. This should be the setting for highest brightness.
But maybe I changed it in the wrong place.

closed 2020-08-29 by stale[bot]