arendst/Tasmota #8974: Tasmota32 TelePeriod for MJ_HT_V1 sensor

31warren, 2020-07-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When the MJ_HT_V1 sensor is not present the MQTT Tele Period works at the default 300s but when the sensor comes into Bluetooth range MQTT sensor data is transmitted every 3 to 5 seconds. Increasing the TelePeriod or MI32Period value makes no difference.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____ESP32 WROOM
- [x ] Tasmota binary firmware version number used: _____8.3.1.7(sensors)
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [ x] IDE / Compiler used: _____Gitpod
- [x ] Flashing tools used: _____Expressif Flash Tool
- [ x] Provide the output of command: `Backlog Template; Module; 
```
  Configuration output here: 
GPIO 255`:00:02:17 MQT: stat/tasmota_50C0B8/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:02:17 MQT: stat/tasmota_50C0B8/RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:02:17 MQT: stat/tasmota_50C0B8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}

09:05:25 MQT: stat/tasmota_50C0B8/RESULT = {"TelePeriod":300}
09:20:32 MQT: stat/tasmota_50C0B8/RESULT = {"MI32Period":300}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ x] Provide the output of this command:
  STATUS 0 output here:
 `Status 0`:00:04:02 MQT: stat/tasmota_50C0B8/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_50C0B8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
```

```
- [ x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Console output shows a continuous stream of MQTT messages from the sensor.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_ Compile using Tasmota32-Sensors with the use MI_ESP32 setting. No changes to the default settings other than setting the MQTT server details. Bring a working MJ_HT_V1 Bluetooth sensor into range and notice a continuous flow of MQTT sensor readings start. Remove the sensor and the MQTT transmissions return to the default 300s value. Check TelePeriod and MI32Period are still set at 300. Attempt to increase the MI32Period - no difference observed until a point where the MI32Period changes from the set value to -99 and the interval between MQTT messages decreases to around 5 messages per second.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._ I expected a sensor reading and MQTT transmission to take place at the default period of 300s.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._ All other functions of the module are working correctly.

**(Please, remember to close the issue when the problem has been addressed)**

31warren, 2020-07-25:
Some further information.....

On the same ESP32 module/settings Flora and LYWSD03MMC sensors send MQTT messages at the correct time interval. Only the MJ_HT_V1 sensor sends messages every 2-5 seconds.

Staars, 2020-07-26:
What you see now, is the real sending behaviour of the sensors as designed by the different vendors. Under the hood this has always been the case in this driver. The frequency of sended data lies between 3 seconds and 1 hour.
The "old" method simply aggregated the (multiple) received messagesinto one  and published it via MQTT at TELEINTERVAL.
That always meant an undefined temporal state. You could never know, how old your data was.
Second problem for me was that the data did not fit in one big MQTT-message anymore.

The "new" method works like a real-time-bridge, where one received bluetooth message is translated into one immediately published MQTT-message. This solves the temporal problem.

As this is still very much a work in progress, the wiki is not updated yet. Everything related to ESP32 is still alpha in Tasmota.

Besides the fact that this is unexpected behaviour of the driver, does this raise a severe problem in your setup?

You can try to change line 642 to:
MI32.option.minimalSummary = 0;

This should show the old behaviour.

31warren, 2020-07-26:
Thank you for the advice. I will try the 'old' method and see what happens.

When receiving data from BLE sensors that transmit continuously we would normally only scan at pre-set intervals but I appreciate that this won't work if sensors have variable transmission times.

Thanks again, other functionality with Tasmota32 is working very well.

closed 2020-07-26 by 31warren