arendst/Tasmota #8977: Connecting to alternate Wifi Access Point on the wrong channel

MarcG046, 2020-07-25:
### PROBLEM DESCRIPTION

When switching Acces Point, a connection is made to the wrong Wifi channel (the channel that belongs to the other AP).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP-01S
- [ ] Tasmota binary firmware version number used: 8.3.1
  - [X] Pre-compiled
  - [X] Self-compiled
    - [] IDE / Compiler used: PIO/VSCODE
- [ ] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
14:53:14 CMD: Backlog Template; Module; GPIO 255
14:53:15 RSL: stat/tasmota_263E0C/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
14:53:15 RSL: stat/tasmota_263E0C/RESULT = {"Module":{"1":"Sonoff Basic"}}
14:53:15 RSL: stat/tasmota_263E0C/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"Geen"},"GPIO2":{"0":"Geen"},"GPIO3":{"0":"Geen"},"GPIO4":{"0":"Geen"},"GPIO5":{"0":"Geen"},"GPIO9":{"0":"Geen"},"GPIO10":{"0":"Geen"},"GPIO12":{"21":"Relais1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"Geen"},"GPIO15":{"0":"Geen"},"GPIO16":{"0":"Geen"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
14:54:58 CMD: Status 0
14:54:58 RSL: stat/tasmota_263E0C/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_263E0C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:54:58 RSL: stat/tasmota_263E0C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:14:44","StartupUTC":"2020-07-25T13:40:14","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-07-25T14:38:23","SaveCount":14,"SaveAddress":"F6000"}}
14:54:58 RSL: stat/tasmota_263E0C/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-07-25T15:30:29","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"358/699"}}
14:54:58 RSL: stat/tasmota_263E0C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wlan_marc_24","AndroidAP"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
14:54:58 RSL: stat/tasmota_263E0C/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000413","8FDAE797","043683A1","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
14:54:58 RSL: stat/tasmota_263E0C/STATUS5 = {"StatusNET":{"Hostname":"tasmota_263E0C-7692","IPAddress":"192.168.0.110","Gateway":"192.168.0.1","Subnetmask":"255.255.248.0","DNSServer":"192.168.0.1","Mac":"BC:DD:C2:26:3E:0C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:54:58 RSL: stat/tasmota_263E0C/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_263E0C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:54:58 RSL: stat/tasmota_263E0C/STATUS7 = {"StatusTIM":{"UTC":"2020-07-25T13:54:58","Local":"2020-07-25T14:54:58","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:15","Sunset":"20:37"}}
14:54:58 RSL: stat/tasmota_263E0C/STATUS10 = {"StatusSNS":{"Time":"2020-07-25T14:54:58"}}
14:54:58 RSL: stat/tasmota_263E0C/STATUS11 = {"StatusSTS":{"Time":"2020-07-25T14:54:58","Uptime":"0T00:14:44","UptimeSec":884,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wlan_marc_24","BSSId":"78:D2:94:47:DE:EE","Channel":2,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:17"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
00:00:00 CFG: Geladen vanuit flash op F9, Aantal 11
00:00:00 Project tasmota Tasmota Versie 8.3.1(tasmota)-2_7_1
00:00:00 WIF: Verbinden met AP2 AndroidAP Channel 2 BSSId 78:D2:94:47:DE:EE in stand 11N als tasmota_263E0C-7692...
00:00:09 WIF: Verbinding mislukt omdat AP onbereikbaar is
00:00:09 WIF: Verbinden met AP1 wlan_marc_24 in stand 11N als tasmota_263E0C-7692...
00:00:15 WIF: Verbonden

```

(I compiled the Dutch version. I can repeat the experiment with the standard English firmware if needed)
SSID wlan_marc_24 has channel 2 and SSID AndroidAP has channel 1, note that it is trying to connect to AP2 AndroidAP on channel 2 and fails, then switches back to AP1

### TO REPRODUCE
_Steps to reproduce the behavior:_

Setup to SSIDs that are on different channels.
Use the AP console command to switch to the other AP. It will fail to do so.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

A correct switch to the other AP.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

There is a related post on the forum, but it has no replies
"Wi-Fi AP 0 command uses wrong BSSID and Channel"

I have tried removing the channel selection from 
in `support_wifi.ino`, Line 210, the call to `WiFi.begin`, I have removed the channel and Wifi.bssid argument.
After recompiling and running the modified firmware, the AP changing works.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-29:
Pls try lates dev version and report back.

arendst, 2020-07-29:
I cannot reproduce:
```
00:00:00 CFG: Loaded from flash at F4, Count 117
00:00:00 SRC: Restart
00:00:00 Project tasmota Wemos4 Version 8.4.0(tasmota)-2_7_2_1
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 indebuurt2 Channel 5 BSSId A0:AB:1B:7D:42:AC in mode 11N as wemos4...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Connected
12:21:43 NTP: UTC 2020-07-29T10:21:42, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
12:21:43 HTP: Web server active on wemos4 with IP address 192.168.2.223
12:21:44 MQT: Attempting connection...
12:21:44 MQT: Connected
12:21:44 MQT: tele/wemos4/LWT = Online (retained)
12:21:44 MQT: cmnd/wemos4/POWER = 
12:21:44 MQT: Subscribe to cmnd/wemos4/#
12:21:44 MQT: Subscribe to cmnd/tasmotas/#
12:21:44 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
12:21:44 MQT: tele/wemos4/INFO1 = {"Module":"Generic","Version":"8.4.0(tasmota)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
12:21:44 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
12:21:44 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
12:21:44 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
12:21:44 MQT: stat/wemos4/POWER = OFF
12:21:45 QPC: Reset
12:21:47 APP: Boot Count 45
12:21:47 CFG: Saved to flash at FB, Count 118, Bytes 4096
12:21:48 MQT: tele/wemos4/STATE = {"Time":"2020-07-29T12:21:48.411+02:00","Uptime":"0T00:00:11","UptimeSec":11,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":20,"Color":"2B3300","HSBColor":"69,100,20","Channel":[17,20,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:05"}}
12:21:54 CMD: ap 0
12:21:54 SRC: Serial
12:21:54 CMD: Group 0, Index 1, Command "AP", Data "0"
12:21:54 MQT: stat/wemos4/RESULT = {"Ap":{"2":"indebuurt_IoT"}}
12:21:54 CFG: Saved to flash at FA, Count 119, Bytes 4096
12:21:55 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
v0cc1cc0b
~ld
00:00:00 CFG: Loaded from flash at FA, Count 119
00:00:00 SRC: Restart
00:00:00 Project tasmota Wemos4 Version 8.4.0(tasmota)-2_7_2_1
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 indebuurt_IoT in mode 11N as wemos4...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:07 WIF: Connect failed with AP timeout
00:00:08 WIF: Attempting connection...
00:00:08 APP: Boot Count 46
00:00:08 CFG: Saved to flash at F9, Count 120, Bytes 4096
00:00:09 WIF: Attempting connection...
00:00:10 WIF: Connected
00:00:10 HTP: Web server active on wemos4 with IP address 192.168.2.223
00:00:10 CFG: Saved to flash at F8, Count 121, Bytes 4096
12:22:07 NTP: UTC 2020-07-29T10:22:06, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
12:22:07 MQT: Attempting connection...
12:22:07 MQT: Connected
12:22:07 MQT: tele/wemos4/LWT = Online (retained)
12:22:07 MQT: cmnd/wemos4/POWER = 
12:22:07 MQT: Subscribe to cmnd/wemos4/#
12:22:07 MQT: Subscribe to cmnd/tasmotas/#
12:22:07 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
[... 27 lines cut ...]
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on wemos4 with IP address 192.168.2.223
00:00:07 CFG: Saved to flash at F6, Count 123, Bytes 4096
12:22:33 NTP: UTC 2020-07-29T10:22:32, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
12:22:33 MQT: Attempting connection...
12:22:33 MQT: Connected
12:22:33 MQT: tele/wemos4/LWT = Online (retained)
12:22:33 MQT: cmnd/wemos4/POWER = 
12:22:33 MQT: Subscribe to cmnd/wemos4/#
12:22:33 MQT: Subscribe to cmnd/tasmotas/#
12:22:33 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
12:22:33 MQT: tele/wemos4/INFO1 = {"Module":"Generic","Version":"8.4.0(tasmota)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
12:22:33 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
12:22:33 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
12:22:33 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
12:22:33 MQT: stat/wemos4/POWER = OFF
12:22:33 APP: Boot Count 47
12:22:33 CFG: Saved to flash at F5, Count 124, Bytes 4096
12:22:37 MQT: tele/wemos4/STATE = {"Time":"2020-07-29T12:22:37.673+02:00","Uptime":"0T00:00:14","UptimeSec":14,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":20,"Color":"2B3300","HSBColor":"69,100,20","Channel":[17,20,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:08"}}
```

MarcG046, 2020-07-29:
The issue is fixed in the development branch and now also in 8.4.0 (with the fix to issue #8635)
Sorry for not trying the development branch myself earlier, I didn't realize that it was not the master branch.

closed 2020-07-29 by MarcG046