arendst/Tasmota #8979: Perform command when device switched from SRC: Switch (Sonoff D1 Dimmer)

digitlength, 2020-07-25:
### PROBLEM DESCRIPTION
Sonoff D1 Dimmers have a known ghost switching issue. Can this be trapped and acted upon?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff D1 Dimmer
- [x] Tasmota binary firmware version number used: 8.3.1.3
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Gitpod
- [x] Flashing tools used: espEasy
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
20:36:15 MQT: stat/swUpstairsHallwayLight/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:36:15 MQT: stat/swUpstairsHallwayLight/RESULT = {"Module":{"74":"Sonoff D1"}}
20:36:16 MQT: stat/swUpstairsHallwayLight/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
Not applicable

```
- [x] Provide the output of this command: `Status 0`:
```
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS = {"Status":{"Module":74,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"swUpstairsHallwayLight","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T10:15:09","StartupUTC":"2020-07-25T09:21:30","Sleep":0,"CfgHolder":4617,"BootCount":62,"BCResetTime":"2020-06-07T10:02:52","SaveCount":445,"SaveAddress":"F8000"}}
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS2 = {"StatusFWR":{"Version":"8.3.1.3(tasmota)","BuildDateTime":"2020-06-14T10:49:16","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"398/699"}}
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ACPRO1",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000000060000005A00000000000000","00008000","00000000"]}}
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS4 = {"StatusMEM":{"ProgramSize":589,"Free":412,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00004024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS5 = {"StatusNET":{"Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"C8:2B:96:5A:2B:32","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.92","MqttPort":1883,"MqttClientMask":"swUpstairsHallwayLight","MqttClient":"swUpstairsHallwayLight","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS7 = {"StatusTIM":{"UTC":"2020-07-25T19:36:39","Local":"2020-07-25T20:36:39","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:15","Sunset":"20:37"}}
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS10 = {"StatusSNS":{"Time":"2020-07-25T20:36:39"}}
20:36:39 MQT: stat/swUpstairsHallwayLight/STATUS11 = {"StatusSTS":{"Time":"2020-07-25T20:36:39","Uptime":"0T10:15:09","UptimeSec":36909,"Heap":22,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
20:08:56 WIF: Checking connection...
20:09:16 WIF: Checking connection...
20:09:36 WIF: Checking connection...
20:09:54 DMP: AA 55 01 04 00 0A 01 64 FF FF FF FF FF FF FF FF 6C
20:09:54 SRC: Switch
20:09:54 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON"}
20:09:54 MQT: stat/swUpstairsHallwayLight/POWER = ON
20:09:55 CFG: Saved to flash at FB, Count 442, Bytes 4096
20:09:56 WIF: Checking connection...
20:10:16 WIF: Checking connection...
20:10:36 WIF: Checking connection...

```

### TO REPRODUCE
Wire in a Sonoff D1 Dimmer.
Power it up.
Ensure light is OFF.
Wait a while.
Light will turn ON at random, producing the above log (ghost switching ON, as described in #7598 )

### EXPECTED BEHAVIOUR
Is it possible to somehow pickup the fact that the light was turned ON by `SRC: Switch`, and immediately command OFF?

I only intend to use this device through MQTT:

```
20:44:01 MQT: Received Topic "cmnd/swUpstairsHallwayLight/POWER", Data Size 2, Data "ON"
20:44:01 SRC: MQTT
20:44:01 CMD: Group 0, Index 1, Command "POWER", Data "ON"
20:44:01 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON"}
20:44:01 MQT: stat/swUpstairsHallwayLight/POWER = ON
20:44:01 DMP: AA 55 01 04 00 00 05
20:44:01 CFG: Saved to flash at F4, Count 449, Bytes 4096
```

### SCREENSHOTS
Not applicable

### ADDITIONAL CONTEXT
Not applicable

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-07-25:
Yes, by the use of rules, you can detach the switch and make the logic you want. 

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-07-25 by ascillato2

digitlength, 2020-07-25:
> Yes, by the use of rules, you can detach the switch and make the logic you want.

Can you give me a hint, or point me to a similar example? I'm not managing to use the correct search terms, evidently.

I do not have a Discord account.

ascillato, 2020-07-26:
> Can you give me a hint, or point me to a similar example?

Yes, of course. No problem.

There is a cookbook with a lot of examples on the use of rules in the Tasmota documentation at: https://tasmota.github.io/docs/Rules/#perform-any-action-on-singledouble-press-for-switches-and-buttons

> I do not have a Discord account.

The chat community is really awesome, you should check it.