arendst/Tasmota #8984: Sleep timer / Inching / Auto turn-off functionality on button press (watering plants)

bentolor, 2020-07-27:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_I migrated my Sonoff S20 lately to Tasmota and miss an important functionality: My problem is, that I use the S20 plug to water plants outdoor: With the EWELink firmware there was an "inching" functionality available which allowed me to define a custom time span after which the plug reliably would turn off automatically._

_My Wifi connection outdoor is only sporadic available if at all. On demand-programming a new cycle via web interface turned out to be rather cumbersome (UI UX) or not possible (broken Wifi Link)._ 

_Therefore restoring my former functionality of just pressing the button to start on demand a new watering cycle which gets automatically turned off i.e. after 10 minutes would be a big win. Yet I drained my water barrels empty more than once…_ 

**Describe the solution you'd like**  
_I want to be able to configure a custom time period after which the plug should turned off itself reliably._ 

_This means i.e. after a turning on the plug by a short button press (default behavior) it will automatically turn off after my custom configured time (i.e. 10 mins). This auto-turn-off period could also apply to schedule-triggered turn-ons._

**Describe alternatives you've considered**  
_1. Manually programming a new time table entry: Doesn't work due to crappy Wifi connection._
_2. Manually turning of: I forget it EVERY time!_
_3. Looking at the button config options: Didn't see any way to copy my desired behavior_

**Additional context**  
_Nothing._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-07-27:
Hi Bentolor,
should be easy by using rules.
Set timer on button press and turn off when timer elapsed.
Search for rules and timer.
Regards Joerg

bentolor, 2020-07-27:
Hi @schrej-zz : Thanks for your quick reply. Hmmm… how did I miss that?

Nevertheless: *Easy* does not match at least for me: I'm a little lost on how to get things working.  I would access the web console an there paste something like the following ????:

```
Rule1  ON Power#state=1 DO RuleTimer1 4 ENDON ON Rules#Timer=1 DO Power 0 ENDON
```

But this is not working. Also tried `Power1` instead of `Power`.

bentolor, 2020-07-27:
Thanks @schrej-zz : Found a cookbook recipe similar to my use case: https://tasmota.github.io/docs/Rules/#time-delayed-auto-off-switch

Using this I found out I must enter
```
Rule1  ON Power1#state=1 DO RuleTimer1 4 ENDON ON Rules#Timer=1 DO Power1 Off ENDON
Rule1 1
```

closed 2020-07-27 by bentolor

jonathon-love, 2023-11-26:
what about `PulseTime`?

https://tasmota.github.io/docs/Commands/#control

cheers