arendst/Tasmota #8985: Material Design Icons for co2 and TVOC are swapped in Home Assistant auto discovery

pashdown, 2020-07-28:
### PROBLEM DESCRIPTION
When Home Assistant does auto-discovery of co2 and TVOC sensors, the mdi (material design icons) are swapped.  TVOC should be mdi:air-filter but is set to mdi:periodic-table-co2 and co2 should be mdi:periodic-table-co2 but is mdi:air-filter.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): d1mini
- [ X] Tasmota binary firmware version number used: 8.3.1.2
  - [ ] Pre-compiled
  - [ X] Self-compiled
    - [X ] IDE / Compiler used: Visual Studio Code
- [X ] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
18:19:33 CMD: Backlog Template; Module; GPIO 255
18:19:33 MQT: stat/attic/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
18:19:34 MQT: stat/attic/RESULT = {"Module":{"18":"Generic"}}
18:19:34 MQT: stat/attic/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"6":"I2C SDA"},"GPIO3":{"0":"None"},"GPIO4":{"5":"I2C SCL"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
18:19:59 MQT: stat/attic/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":122,"Free":389,"Rules":"on Switch2#state=1 do publish stat/%topic%/PIR1 ON endon on Switch2#state=0 do Publish stat/%topic%/PIR1 OFF endon Rule1 1"}
18:19:59 MQT: stat/attic/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
18:19:59 MQT: stat/attic/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
18:20:01 MQT: tele/attic/STATE = {"Time":"2020-07-28T18:20:00","Uptime":"12T00:36:01","UptimeSec":1038961,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"ON","Wifi":{"AP":1,"SSId":"pashnet-IoT","BSSId":"18:E8:29:94:A2:29","Channel":6,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:20:01 MQT: tele/attic/SENSOR = {"Time":"2020-07-28T18:20:01","BMP280":{"Temperature":96.4,"Pressure":860.1,"SeaPressure":1022.7},"CCS811":{"eCO2":1144,"TVOC":113},"HDC1080":{"Temperature":95.9,"Humidity":28.1,"DewPoint":57.6},"PressureUnit":"hPa","TempUnit":"F"} (retained)

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
18:20:46 CMD: Status 0
18:20:46 MQT: stat/attic/STATUS = {"Status":{"Module":18,"DeviceName":"Attic","FriendlyName":["Attic"],"Topic":"attic","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[2,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0}}
18:20:46 MQT: stat/attic/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-sensors.bin","RestartReason":"Software/System restart","Uptime":"12T00:36:47","StartupUTC":"2020-07-16T16:43:59","Sleep":50,"CfgHolder":4617,"BootCount":117,"BCResetTime":"2020-05-27T00:14:52","SaveCount":384,"SaveAddress":"F7000"}}
18:20:46 MQT: stat/attic/STATUS2 = {"StatusFWR":{"Version":"8.3.1.2(sensors)","BuildDateTime":"2020-06-27T13:25:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"360/699"}}
18:20:46 MQT: stat/attic/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["pashnet-IoT",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A8309","2805C8000100060000005A00000000000000","00000200","00000000"]}}
18:20:46 MQT: stat/attic/STATUS4 = {"StatusMEM":{"ProgramSize":617,"Free":384,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000809","8FDAA787","0415A3A5","B7FFBFCD","01DA9BC4","64367CC7","0000007A"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,65,66,67"}}
18:20:47 MQT: stat/attic/STATUS5 = {"StatusNET":{"Hostname":"attic-3029","IPAddress":"10.0.68.146","Gateway":"10.0.68.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.68.1","Mac":"EC:FA:BC:C0:8B:D5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:20:47 MQT: stat/attic/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.67.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C08BD5","MqttUser":"","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:20:47 MQT: stat/attic/STATUS7 = {"StatusTIM":{"UTC":"2020-07-28T17:20:47","Local":"2020-07-28T18:20:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:19","Sunset":"20:33"}}
18:20:47 MQT: stat/attic/STATUS10 = {"StatusSNS":{"Time":"2020-07-28T18:20:47","BMP280":{"Temperature":96.5,"Pressure":860.1,"SeaPressure":1022.6},"CCS811":{"eCO2":1195,"TVOC":121},"HDC1080":{"Temperature":96.0,"Humidity":28.1,"DewPoint":57.7},"PressureUnit":"hPa","TempUnit":"F"}}
18:20:47 MQT: stat/attic/STATUS11 = {"StatusSTS":{"Time":"2020-07-28T18:20:47","Uptime":"12T00:36:48","UptimeSec":1039008,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"ON","Wifi":{"AP":1,"SSId":"pashnet-IoT","BSSId":"18:E8:29:94:A2:29","Channel":6,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1) Connect CJMCU-8128 to d1mini
2) Compile Tasmota Sensors to include the necessary sensors
3) Configure Tasmota wifi
4) Configure Tasmota MQTT
5) SetOption19 1
6) Look at Home Assistant and notice the transposed icons

### EXPECTED BEHAVIOUR
eco2 sensor should be mdi:periodic-table-co2
TVOC sensor should be mdi:air-filter

### SCREENSHOTS
What it should be:
[image]

What it is:
[image]

### ADDITIONAL CONTEXT
I'd imagine this line needs to have something swapped, but I don't know how this relates to what sensor they are assigned to.

`./xdrv_12_home_assistant.ino:  "ic\":\"mdi:periodic-table-co2|ic\":\"mdi:air-filter|ic\":\"mdi:periodic-table-co2|";
`

**(Please, remember to close the issue when the problem has been addressed)**

pashdown, 2020-07-28:
Actually, after looking at it again, I think the line should be:
`"ic\":\"mdi:periodic-table-co2|ic\":\"mdi:periodic-table-co2|":\"mdi:air-filter|ic\";`

effelle, 2020-07-28:
Thanks. Will be fixed on next update.

closed 2020-07-28 by effelle