arendst/Tasmota #8989: Tasmota / HC-SR501 PIR sensor / no mqtt messages

Seqway, 2020-07-29:
### PROBLEM DESCRIPTION
I have a NodeMcu v3 and connected several PIR sensors like HC-SR501 onto it (connected to GPIO 4 and 5).
I tried all the solutions in the issues and also the description according to the documentation to implement a rule including Switcmode and Switchtopic.
In the console the PIR sensor is working fine and gives messages ON / OFF when motion is detected.
BUT i do not have any messages on my mqtt server neither on the datapoint SWITCH1 nor on the datapoint POWER.
I really followd the instructions but i could not get it working :-(
Where is my problem ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMcu v3 with HC-SR501
- [ ] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: ESP Easy Flash
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
08:03:58 MQT: stat/NodeMcu_Teich/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:03:58 MQT: stat/NodeMcu_Teich/RESULT = {"Module":{"18":"Generic"}}
08:03:58 MQT: stat/NodeMcu_Teich/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"9":"Switch1"},"GPIO5":{"10":"Switch2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
08:04:31 MQT: stat/NodeMcu_Teich/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":114,"Free":397,"Rules":"on switch1#state=1 do publish stat/%topic%/PIR1 ON endon on switch1#state=0 do publish stat/%topic%/PIR1 OFF endon"}
08:04:31 MQT: stat/NodeMcu_Teich/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":114,"Free":397,"Rules":"on switch2#state=1 do publish stat/%topic%/PIR2 ON endon on switch2#state=0 do publish stat/%topic%/PIR2 OFF endon"}
08:04:31 MQT: stat/NodeMcu_Teich/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:04:58 MQT: stat/NodeMcu_Teich/STATUS = {"Status":{"Module":18,"DeviceName":"NodeMcu_Teich","FriendlyName":["Tasmota"],"Topic":"NodeMcu_Teich","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"BWM_Teich_1","SwitchMode":[1,1,0,1,1,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:04:58 MQT: stat/NodeMcu_Teich/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T08:05:00","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":75,"BCResetTime":"1970-01-01T00:00:00","SaveCount":179,"SaveAddress":"F9000"}}
08:04:58 MQT: stat/NodeMcu_Teich/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020.05.18 15:45:12","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"397/699"}}
08:04:58 MQT: stat/NodeMcu_Teich/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
08:04:58 MQT: stat/NodeMcu_Teich/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000407","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:04:58 MQT: stat/NodeMcu_Teich/STATUS5 = {"StatusNET":{"Hostname":"NodeMcu_Teich-6598","IPAddress":"10.10.8.201","Gateway":"10.10.8.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.40.8","Mac":"E0:98:06:86:19:C6","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:04:58 MQT: stat/NodeMcu_Teich/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.40.7","MqttPort":1883,"MqttClientMask":"NodeMcu_Teich","MqttClient":"NodeMcu_Teich","MqttUser":"iobroker","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:04:58 MQT: stat/NodeMcu_Teich/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T08:04:58","Local":"1970-01-01T08:04:58","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
08:04:58 MQT: stat/NodeMcu_Teich/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T08:04:58","Switch1":"OFF","Switch2":"ON"}}
08:04:58 MQT: stat/NodeMcu_Teich/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T08:04:58","Uptime":"0T08:05:00","UptimeSec":29100,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"IoT","BSSId":"2A:E8:29:91:69:50","Channel":11,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:03"}}}
08:05:07 MQT: tele/NodeMcu_Teich/STATE = {"Time":"1970-01-01T08:05:07","Uptime":"0T08:05:09","UptimeSec":29109,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"IoT","BSSId":"2A:E8:29:91:69:50","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}
08:05:07 MQT: tele/NodeMcu_Teich/SENSOR = {"Time":"1970-01-01T08:05:07","Switch1":"OFF","Switch2":"ON"}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-07-29:
Please close, answered in Discord

Seqway, 2020-07-29:
Will be discussed in Discord

closed 2020-07-31 by ascillato2