arendst/Tasmota #8993: Sonoff 4CH Pro R2 physical pushbuttons not switching relays

parautenbach, 2020-07-29:
### PROBLEM DESCRIPTION
When I actuate the pushbuttons, the corresponding relays don't toggle. It does work when toggling the buttons via the web interface (or MQTT which I have configured).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff 4CH Pro R2
- [ ] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: N/A
- [x] Flashing tools used: `esptool.py`
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

13:14:45 MQT: stat/yard/irrigation/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
13:14:45 MQT: stat/yard/irrigation/RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
13:14:46 MQT: stat/yard/irrigation/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"23":"Relay3"},"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"19":"Button3"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"20":"Button4"},"GPIO15":{"24":"Relay4"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

13:15:54 MQT: stat/yard/irrigation/STATUS = {"Status":{"Module":23,"DeviceName":"Irrigation","FriendlyName":["Irrigation Zone 1","Irrigation Zone 2","Irrigation Zone 3","Irrigation Zone 4"],"Topic":"yard/irrigation","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:15:54 MQT: stat/yard/irrigation/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:34:26","StartupUTC":"2020-07-29T11:41:28","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-07-25T17:32:15","SaveCount":77,"SaveAddress":"FB000"}}
13:15:54 MQT: stat/yard/irrigation/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"424/699"}}
13:15:54 MQT: stat/yard/irrigation/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Pegasus",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00800200","00000000"]}}
13:15:54 MQT: stat/yard/irrigation/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
13:15:54 MQT: stat/yard/irrigation/STATUS5 = {"StatusNET":{"Hostname":"yard/irrigation-1811","IPAddress":"192.168.0.128","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"E0:98:06:C2:27:13","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:15:54 MQT: stat/yard/irrigation/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.104","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C22713","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:15:54 MQT: stat/yard/irrigation/STATUS7 = {"StatusTIM":{"UTC":"2020-07-29T12:15:54","Local":"2020-07-29T13:15:54","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:20","Sunset":"20:32"}}
13:15:54 MQT: stat/yard/irrigation/STATUS10 = {"StatusSNS":{"Time":"2020-07-29T13:15:54"}}
13:15:54 MQT: stat/yard/irrigation/STATUS11 = {"StatusSTS":{"Time":"2020-07-29T13:15:54","Uptime":"0T00:34:26","UptimeSec":2066,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Pegasus","BSSId":"5C:96:9D:6A:7E:9D","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
1. Turn on the device
2. Press any button

### EXPECTED BEHAVIOUR
I expect the corresponding relay and LED to switch toggle.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
The only related issue I could find was someone that powered their device incorrectly. I read elsewhere to "simply" reboot the device a few times. In a third instance, someone said to just re-upload the firmware (which I've done).

**(Please, remember to close the issue when the problem has been addressed)**

kugelkopf123, 2020-07-29:
Same with the 4ch r2.

effelle, 2020-07-29:
Please check if setoption73 is off and post the log (using weblog 4) when you press any of the buttons.

arendst, 2020-07-29:
Verified. With ``SO73 1`` the buttons don't work for obvious reasons.

kugelkopf123, 2020-07-29:
Found the reason with me:
https://github.com/arendst/Tasmota/pull/8927
Thought this was only for switches, not switches and buttons. 
If I comment it out of the script the buttons work again. 
SO73 is 0.

parautenbach, 2020-07-29:
> Please check if setoption73 is off and post the log (using weblog 4) when you press any of the buttons.

```
15:18:19 CMD: SetOption73
15:18:19 MQT: stat/yard/irrigation/RESULT = {"SetOption73":"ON"}
15:19:15 CMD: SetOption73 0
15:19:15 MQT: stat/yard/irrigation/RESULT = {"SetOption73":"OFF"}
15:19:23 CMD: Weblog 4
15:19:23 MQT: stat/yard/irrigation/RESULT = {"WebLog":4}
15:19:23 CFG: Saved to flash at FB, Count 85, Bytes 4096
15:19:27 APP: Button1 level 1-0
15:19:27 MQT: stat/yard/irrigation/BUTTON1T = {"TRIG":"SINGLE"}
15:19:27 SRC: Button
15:19:27 MQT: tele/yard/irrigation/STATE = {"Time":"2020-07-29T15:19:27","Uptime":"0T02:37:59","UptimeSec":9479,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Pegasus","BSSId":"5C:96:9D:6A:7E:9D","Channel":1,"RSSI":72,"Signal":-64,"LinkCount":2,"Downtime":"0T00:00:33"}}
15:19:27 MQT: stat/yard/irrigation/RESULT = {"POWER1":"ON"}
15:19:27 MQT: stat/yard/irrigation/POWER1 = ON
15:19:27 APP: Button1 level 0-1
15:19:27 CFG: Saved to flash at FA, Count 86, Bytes 4096
15:19:33 CMD: Weblog 4
15:19:33 SRC: WebConsole from 192.168.0.112
15:19:33 CMD: Group 0, Index 1, Command "WEBLOG", Data "4"
15:19:33 MQT: stat/yard/irrigation/RESULT = {"WebLog":4}
```

Turning it off means the buttons work as expected. 

I don't know why `SO73` was on (neither am I sure that I understand what exactly that option does).

parautenbach, 2020-07-29:
I think it's the [Tasmota Home Assistant docs](https://tasmota.github.io/docs/Home-Assistant/) that lead me to turn it on. 

(Also, really appreciate the quick feedback.)

closed 2020-07-29 by parautenbach

effelle, 2020-07-29:
> UPDATE: Thinking about it more: Does it mean if I receive a button press command it's up to me to send a command back to turn the LED on? Is that what "decouple" means here?

Yes. SO73 was made to decouple the buttons from their respective power devices (relays) and let the user manage them as unlinked entities.
Home Assistant Discovery needs so73 enabled if you want to use the buttons as triggers for automation, then in Tasmota you can set a rule to enable just what you want. 
For example you could decide to have a button react in Tamsota just when you press it 3 times or when you hold it and leave all the rest of the combinations out (or managed by Home Assistant automation).