arendst/Tasmota #8995: Home Assistant Light Names

mk-81, 2020-07-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Since Tasmota 8.1.3 (commit 950a4dc998b37cdb0ec04edda874a07f2902be44, issue #8462), the names of the home assistant lights (home assistant auto discovery) are generated from device name and friendly name. This results to "double" names in home assistant for simple switches. 

E.g:
If the friendly name is **light1**, than the entity will be named **light1 light1**, because the device name defaults to friendly name 1. As I can clearly see the need for sensors and multi switch devices, on single switch devices (e.g. sonoff basic) this is a bit of a overshoot. Also it changes (needlessly) the already existing light entity names.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____ Sonoff Basic R2
- [X] Tasmota binary firmware version number used: _____8.4.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: _____Tasmota Flasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

17:47:23 MQT: home/livingroom/plants-light/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:47:24 MQT: home/livingroom/plants-light/stat/RESULT = {"Module":{"1":"Sonoff Basic"}}
17:47:24 MQT: home/livingroom/plants-light/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"9":"Switch1"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
n/a

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

17:48:14 MQT: home/livingroom/plants-light/stat/STATUS = {"Status":{"Module":1,"DeviceName":"Beleuchtung-Pflanzen","FriendlyName":["Beleuchtung-Pflanzen"],"Topic":"home/livingroom/plants-light","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:48:14 MQT: home/livingroom/plants-light/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:13","StartupUTC":"2020-07-29T16:37:01","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2020-06-10T15:41:04","SaveCount":97,"SaveAddress":"F8000"}}
17:48:14 MQT: home/livingroom/plants-light/stat/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:05:28","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"472/699"}}
17:48:14 MQT: home/livingroom/plants-light/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ORCA-INT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["401AA00B","2805C800010006000A005AFF000000000000","00000229","00006000"]}}
17:48:14 MQT: home/livingroom/plants-light/stat/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:48:14 MQT: home/livingroom/plants-light/stat/STATUS5 = {"StatusNET":{"Hostname":"Beleuchtung-Pflanzen.mk-home","IPAddress":"192.168.0.239","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:92:B8:2A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:48:14 MQT: home/livingroom/plants-light/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_92B82A","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:48:14 MQT: home/livingroom/plants-light/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-07-29T16:48:14","Local":"2020-07-29T17:48:14","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:20","Sunset":"20:32"}}
17:48:14 MQT: home/livingroom/plants-light/stat/STATUS10 = {"StatusSNS":{"Time":"2020-07-29T17:48:14","Switch1":"OFF"}}
17:48:14 MQT: home/livingroom/plants-light/stat/STATUS11 = {"StatusSTS":{"Time":"2020-07-29T17:48:14","Uptime":"0T00:11:13","UptimeSec":673,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"<removed>","BSSId":"<removed>","Channel":1,"RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
n/a

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set Friendly Name1 of a single switch device

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would expect a senseful light entity name, in case of a single switch device, where the device name defaults to friendly name 1. For example, if device name and friendly name 1 is identical, than only one of them is used as name in HA auto discovery.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

danmed, 2020-07-31:
I solve this by just putting a space character in the friendly name and putting the full name in the device Name field.

Seems to work though it's a little annoying.

mk-81, 2020-08-01:
thanks for the tip (@danmed). It works, even if it's more like a workaround. I've changed my devices. But still I think the implementation is wrong (this is more for the devs) because of two reasons:

first: 
The default implementation (the implementation most of the devices will have) changes needlessly their device names. This applies to all sonoff basic switches, smart plugs, lights etc. As far as I understood the change to device name, this should have been compatible to the most one light devices (device name is friendly name 1, so for most lights nothing would happen if the name would not be doubled before the patch for the other entity types)

second:
the generated names contains (obviously) a space at the end. It seems that homeassistant remove trailing spaces (didn't check in code).
[image]

all in all I'll personally think that a it is the most compatible way to send only one of the names in case devicename and friendlyname1 are identical. Kepps current names, doesn't change "new" names and keeps also the workaround from danmed.

effelle, 2020-08-01:
Once upon a time a Home Assistant dev asked to use a different name to identify a device under HAss to avoid repetitions, then people started to open issues about it here.
We started to use module name as a base, people started to open issues about it.
When we added a real device name, people started to open issues about it.
Seems a bit hard to make everyone happy folks! 

Fear not, I'll remove the device name (just from relays/lights entities) in the next update and will use just a Friendlyname. Sensors will continue to use the DeviceName.
This will obviously force you to delete completely the devices from HAss before use again So19 to clean the HAss DB.
When the update will be ready you will receive a notification and this issue will be automatically closed.

mk-81, 2020-08-01:
Hello @effelle,

don't get me wrong. I'm not complaining and I'm completly not unhappy with tasmota. This was only my thinking based on my experiences with some devices where I only set the friendly name (more of an upgrade issue). And I thought that this might be unintended behaviour.

I can clearly see a need for device name (switch from module to device) and I see the need also for lights. So, if you think your change is a good way to go, than there is no need for a change, besides the docu maybe. Please don't change anything only to change it back again because we missed something here. I don't want to force something. There is a workaround available, so I'm fine. I had to remove the devices from home assistant DB anyway - this is possible via GUI now. 

Maybe it's enough for 99% of the devices to send only device name, in case the friendly name 1 equals the device name. If you didn't name you device than no harm should be done. Or maybe it is enough to have the possibility to have an empty friendly name 1.  Something like that is may be sufficient to. 

Please take your time to think about it. I'm not pussing anything, and hopefully everone also, as there is a working workaround in place.

PR #9085 (2020-08-14): HAss update
## Description:
- Removed `DeviceName` from Relay based entities;
- Correct some icons;
- Discovery will now show just the relevant topics for the configuration, hiding all the unnecessary MQTT messages in console. 
Those messages will be transmitted to the broker as normal but will be not shown to help reducing CPU load and avoid pollution in console. Set `weblog` (or `seriallog` if connected with serial) to `4` to see all of them.

**Related issue (if applicable):** fixes #8995 

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant fi [... cut]
closed 2020-08-14 by arendst