arendst/Tasmota #8996: WifiConfig changes to 2 on reboot on iFan02

rishabmehta7, 2020-07-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have 3 iFan02 devices.

All 3 have been set to WifiConfig 5. On a power failure (which I have been seeing a lot recently - our generator is having issues) the WifiConfig changes to 2 on all 3 devices, my other SonOff devices work as expected (SonOff Basic, SonOff 4CH).

I have tried changing the same to 5 multiple times. But on every 2 3rd reboot it Just switches to 2 and then takes forever to connect to MQTT broker.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____SonOff iFan02
- [x] Tasmota binary firmware version number used: _____8.2, 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: _____esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
12:41:37 MQT: stat/iFan02_02/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:41:37 MQT: stat/iFan02_02/RESULT = {"Module":{"44":"Sonoff iFan02"}}
12:41:37 MQT: stat/iFan02_02/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"15":"Switch7"},"GPIO2":{"0":"None"},"GPIO3":{"14":"Switch6"},"GPIO4":{"23":"Relay3"},"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"19":"Button3"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"20":"Button4"},"GPIO15":{"24":"Relay4"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
12:43:11 MQT: stat/iFan02_02/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":475,"Rules":"on switch6#state do Fanspeed + endon"}
12:43:11 MQT: stat/iFan02_02/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":472,"Rules":"on switch7#state do POWER1 TOGGLE endon"}
12:43:12 MQT: stat/iFan02_02/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```
12:43:44 MQT: stat/iFan02_02/STATUS = {"Status":{"Module":44,"FriendlyName":["iFan02_02"],"Topic":"iFan02_02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,3,3,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
12:43:44 MQT: stat/iFan02_02/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:50:58","StartupUTC":"2020-07-05T06:22:46","Sleep":50,"CfgHolder":4617,"BootCount":193,"BCResetTime":"2020-06-02T09:16:54","SaveCount":1332,"SaveAddress":"FB000"}}
12:43:44 MQT: stat/iFan02_02/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"326/699"}}
12:43:44 MQT: stat/iFan02_02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mehta's",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028029","2805C8000100060000005A00000000000000","00008200","00000000"]}}
12:43:44 MQT: stat/iFan02_02/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:43:44 MQT: stat/iFan02_02/STATUS5 = {"StatusNET":{"Hostname":"iFan02_02","IPAddress":"192.168.31.212","Gateway":"192.168.31.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.31.250","Mac":"84:0D:8E:47:A8:9C","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
12:43:44 MQT: stat/iFan02_02/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.31.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_47A89C","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:43:44 MQT: stat/iFan02_02/STATUS7 = {"StatusTIM":{"UTC":"2020-07-05T07:13:44","Local":"2020-07-05T12:43:44","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+05:30","Sunrise":"09:24","Sunset":"01:25"}}
12:43:44 MQT: stat/iFan02_02/STATUS10 = {"StatusSNS":{"Time":"2020-07-05T12:43:44","Switch6":"ON","Switch7":"ON"}}
12:43:44 MQT: stat/iFan02_02/STATUS11 = {"StatusSTS":{"Time":"2020-07-05T12:43:44","Uptime":"0T00:50:58","UptimeSec":3058,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"Mehta's","BSSId":"78:11:DC:47:28:65","Channel":8,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:17"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set WifiConfig to 5
SetOption65 1
SetOption19 0
SwitchMode6 3
SwitchMode7 3

Reboot 3-4 times and WifiConfig would become 2

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The WifiConfig value should stay at 5 as set.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Rx and Tx pin are connected to capacitive touch switches(TT223), Vcc Gnd pulled from Pins 

**(Please, remember to close the issue when the problem has been addressed)**

rishabmehta7, 2020-07-29:
Closing the old ticket and reopened a new one
https://github.com/arendst/Tasmota/issues/8856

So i updated to latest dev

Reset 6

Changed the module to iFan02
05:22:30 MQT: tele/iFan02_03/INFO1 = {"Module":"Sonoff iFan02","Version":"8.4.0(202326c-tasmota)","FallbackTopic":"cmnd/DVES_47A6D4_fb/","GroupTopic":"cmnd/tasmotas/"}

Changed the Device name, friendly name, Hostname (Under configure Module and Wifi)

Changed the IP, DNS, NTP AP1 AP2 and WifiConfig using TasmoAdmin

Added Retain:Power using TasmoAdmin.

Will monitor now.

Ok so without the switches the config seems to have stayed put after reboot.
So the touch switches could be a problem?
I have similarly configured switches on sOnOff Basics which work perfectly fine.
#########
Just to clarify the switches were connected physically but not configured on the module and no rules were created.
#########
Spoke too soon, the immediate next reboot the device reset the wifi config to 2, no changes were made in any way.
#########
Every software reboot doesn't seem to reset the config, a power cycle resets the WifiConfig to 2.
#########
Removed them physically as well and redid everything
Still resets on a power failure.
Rishab

PrathikGopal, 2020-07-29:
I’m facing the exact same issue.

Jason2866, 2020-07-29:
Do you have tried `SetOption65`?

rishabmehta7, 2020-07-30:
SetOption65 1
SetOption36 0

The previous ticket marks everything that had been tried.

arendst, 2020-07-30:
I cannot reproduce but the only way wificonfig can change to 2 is when button1 is pressed for over six times. This can be disabled by ``SetOption1 1``. Hope this helps

thecode, 2020-07-30:
I have always had this problem (at least since 6.x) Devices are not user accessible so there is no way for someone to press the button.
I checked now and out of 22 tasmota devices only the 2 devices which are iFAN02 are showing WifiConfig 2.
It does not bother me but I noticed the issue so I would like to give the feedback that it is not something related to the latest release.
I think I checked in the past and when you press the WiFi button on the 2.4GHz RF remote.
I looked at the code and there is a path there to switch the WiFi mode.
look at:
https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_22_sonoff_ifan.ino#L168

arendst, 2020-07-30:
> I looked at the code and there is a path there to switch the WiFi mode.
look at:
https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_22_sonoff_ifan.ino#L168

That code is only valid for iFan03 which doesn't experience the issue related here.

thecode, 2020-07-30:
> That code is only valid for iFan03 which doesn't experience the issue related here.

Yes I noticed now that it is only valid for iFAN03, anyhow the important feedback is that this problem existed even before iFAN03 support was added.

I have tried to reproduce this with the remote without success, but the fact that the two iFAN02 switched to mode 2 should hint these devices have something take changes the WiFi mode. I have switched all devices more than once to mode 5 by sending the command to all of them.
I forgot to note that they are on 8.3.1 now

arendst, 2020-07-30:
Thx. Appreciated.

Let them try ``SO1 1`` and see if it makes a change.

rishabmehta7, 2020-07-30:
Let me add SO1 1 and get back.

rishabmehta7, 2020-07-30:
One thing to mention is you might need to cut power to the unit and not do a software reboot. I have noticed a software reboot doesnt change the WiFi Config.

Another thing i noticed is, when the device reboots(power failure) i see the SSID being published, connecting to it takes me to the WiFi Info page, I always see a 180sec countdown on reboot at the bottom. The 'iFan02 Light' stays on till the timer runs out or i click reboot. And then the previous states are retained.
The Light stay on due to the design of the MCU, that you have spoken about in the past.

The previous state only comes after a reboot of the device.

Rishab

rishabmehta7, 2020-07-31:
My 3 devices is inside an electrical box and hence the button is not accessible. But SO1 1 does seem to help. The device did continue to have WifiConfig at 4 and did not reset. I'll still request to keep the issue open for another day or so.

Cheers
Rishab

PrathikGopal, 2020-08-01:
> Thx. Appreciated.
> 
> Let them try `SO1 1` and see if it makes a change.

This seems to have done it, stays at wificonfig 4 now after powercuts.

ascillato2, 2020-08-01:
> SO1 1 does seem to help. The device did continue to have WifiConfig at 4 and did not reset.

Great. 

Thanks for reporting.

closed 2020-08-01 by ascillato2

rishabmehta7, 2020-08-03:
Should this be added do the iFan02 Docs so others can know?

Rishab

ascillato, 2020-08-03:
Yes, please, help us and make a PR to the docs. Thanks.

rishabmehta7, 2020-08-06:
Created a PR on Blakadders repo (https://github.com/blakadder/templates/pull/611)