arendst/Tasmota #9000: Version 8.1 documentation

hi-flyer, 2020-07-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I have all my smart devices working on Tasmota 8.1 and thank you for the excellent software.
According to your guidance "If it aint broke, do not upgrade" with which I agree. The problem is you are constantly upgrading documentation on the website (which is great) but i need a copy for 8.1 as I do not wish to upgrade my software. Is there somewhere I can download the Documentation for version 8.1? I cannot find it.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
PDF or Web download of Tasmota Version 8.1 documentation
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
PDF or Web download of the current version of Tasmota Version 8.4 Documentation
**Additional context**  
_Add any other context or screenshots about the feature request here._
none
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-07-30:
Not for 8.1. You can go back into time to 8.2 https://web.archive.org/web/20200416100814/https://tasmota.github.io/docs/

arendst, 2020-07-30:
Oh that's a nice feature!

sfromis, 2020-07-30:
If you can put up with a zip containing raw markdown files, my data hoard has a download from one month after the release of 8.2.

hi-flyer, 2020-07-30:
Thank you Jason2866, I am good with that.

closed 2020-07-30 by hi-flyer