arendst/Tasmota #9003: BW-SHP2 (Gosund SP1 v23 Module) button stopped working

melojms, 2020-07-31:
### PROBLEM DESCRIPTION
Physical on/off button stopped working after upgrading to Tasmota 8.4.0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Blitzwolf BW-SHP2 (Gosund SP1 v23)
- [x] Tasmota binary firmware version number used: 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
14:54:19 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
14:54:19 SRC: Backlog
14:54:19 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
14:54:19 MQT: bw1/stat/RESULT = {"NAME":"BlitzWolf SHP","GPIO":[57,255,56,255,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}
14:54:19 SRC: Backlog
14:54:19 CMD: Group 0, Index 1, Command "MODULE", Data ""
14:54:19 MQT: bw1/stat/RESULT = {"Module":{"55":"Gosund SP1 v23"}}
14:54:20 SRC: Backlog
14:54:20 CMD: Group 0, Index 1, Command "GPIO", Data "255"
14:54:20 MQT: bw1/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"158":"LedLinki"},"GPIO2":{"0":"None"},"GPIO3":{"17":"Button1"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"56":"Led1i"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
14:54:20 WIF: Checking connection...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

14:48:38 MQT: bw1/stat/STATUS = {"Status":{"Module":55,"DeviceName":"bw1","FriendlyName":["bw1"],"Topic":"bw1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:48:38 MQT: bw1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://http://thehackbox.org/tasmota/release/tasmota.bin:80/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:12","StartupUTC":"2020-07-31T13:42:26","Sleep":50,"CfgHolder":4617,"BootCount":98,"BCResetTime":"2020-05-29T10:07:57","SaveCount":77653,"SaveAddress":"F6000"}}
14:48:38 MQT: bw1/stat/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:05:28","Boot":4,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"440/699"}}
14:48:38 MQT: bw1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["R7000","R7000"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005AFF000000000000","00800200","00006000"]}}
14:48:38 MQT: bw1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
14:48:38 MQT: bw1/stat/STATUS5 = {"StatusNET":{"Hostname":"bw1-3471","IPAddress":"10.0.0.114","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"DC:4F:22:FC:4D:8F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:48:38 MQT: bw1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"x.x.x.x","MqttPort":xxxx,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FC4D8F","MqttUser":"xxxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:48:38 MQT: bw1/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-07-31T13:48:38","Local":"2020-07-31T14:48:38","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:23","Sunset":"20:29"}}
14:48:38 MQT: bw1/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:48:38 MQT: bw1/stat/STATUS10 = {"StatusSNS":{"Time":"2020-07-31T14:48:38","ENERGY":{"TotalStartTime":"2019-02-19T23:58:06","Total":225.511,"Yesterday":0.293,"Today":0.160,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
14:48:38 MQT: bw1/stat/STATUS11 = {"StatusSTS":{"Time":"2020-07-31T14:48:38","Uptime":"0T00:06:12","UptimeSec":372,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"R7000","BSSId":"E4:F4:C6:0F:D1:74","Channel":11,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

14:49:41 APP: Button1 multi-press 1
14:49:42 MQT: bw1/stat/BUTTON1T = {"TRIG":"SINGLE"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Upgrade to v8.4.0. Try to manually push the button to turn on/off the switch.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The switch should toggle as before.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-31:
Execute ``SO73 0`` et presto

melojms, 2020-07-31:
Thanks! It worked. Don't know why it happened. I have upgraded tasmota in other BW-SHP2, and didn't have this issue.

closed 2020-07-31 by melojms