arendst/Tasmota #9004: IRHVAC command should hold state

yoav1000, 2020-07-31:
**Have you looked for this feature in other issues and in the docs?**  yes

**Is your feature request related to a problem? Please describe.**  
Some of The AC vendors supported by IRremoteESP8266 library and tasmota do not support ON/OFF commands,
Instead they support toggle operations.
For example the WHIRLPOOL and AIRWELL protocols indicate power state and swing state change, not absolute state.
When using the IRHVAC command this translate to problematic scenario, most of the command happen with POWER=off, and in is turn result in hardcoded mode=off even if the mode has changed.

**Describe the solution you'd like**  
I think we should hold the state for those toggle commands in flash/ram so IRHVAC will represent the absolute state of the AC . I also think we should add a new command and option to query/set state and determine what will be the default state after power cycle.

why to hold the state in tasmota and not in Home Assistant for example? - I think the tasmota device itself is less prune to power cycle and can update the state even in cases where the wifi is down and the user use the AC remote. For some edge cases, commands to toggle the internal state will give a complete solution.

I also think we should store a state for each vendor,model pair to support multi device scenario or create a number of state memory slots and indicate which one we register to the specific vendor,model.

**Describe alternatives you've considered**  

**Additional context**  
Some examples:
ON capture
`{"IrReceived":{"Protocol":"WHIRLPOOL_AC","Bits":168,"Data":"0x0x830605A200008C0200000000002900010000080009","Repeat":0,"IRHVAC":{"Vendor":"WHIRLPOOL_AC","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":26,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`
OFF capture
`{"IrReceived":{"Protocol":"WHIRLPOOL_AC","Bits":168,"Data":"0x0x830605A200008C0100000000002A00010000080009","Repeat":0,"IRHVAC":{"Vendor":"WHIRLPOOL_AC","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":26,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`
Temp change (you can see the mode is off mistakenly)
`{"IrReceived":{"Protocol":"WHIRLPOOL_AC","Bits":168,"Data":"0x0x830601B200008C0400000000003B0002000008000A","Repeat":0,"IRHVAC":{"Vendor":"WHIRLPOOL_AC","Model":2,"Power":"Off","Mode":"Off","Celsius":"On","Temp":27,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
{"IrReceived":{"Protocol":"WHIRLPOOL_AC","Bits":168,"Data":"0x0x830601A200008C0400000000002B0002000008000A","Repeat":0,"IRHVAC":{"Vendor":"WHIRLPOOL_AC","Model":2,"Power":"Off","Mode":"Off","Celsius":"On","Temp":26,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-08-01:
As IRHVAC support in Tasmota is nothing more than sending IR codes to the HVAC you'll have to get a decent Home Automation solution which will do what you want.

It's not Tasmota's task to keep a score of IR controlled devices.

closed 2020-08-01 by ascillato2

yoav1000, 2020-08-05:
> 
> 
> As IRHVAC support in Tasmota is nothing more than sending IR codes to the HVAC you'll have to get a decent Home Automation solution which will do what you want.
> 
> It's not Tasmota's task to keep a score of IR controlled devices.

In order to create a good integration with tasmota we should remove this line to allow the integration to keep track on the AC mode.
https://github.com/arendst/Tasmota/blob/d321b6cf68334a278dac0acb2d3f3e0312619dd1/tasmota/xdrv_05_irremote_full.ino#L121

with the current state you can only detect mode changes on power toggle with those types of AC's