arendst/Tasmota #9005: Sonoff D1 ghost switching after upgrade to 8.3.1.7

epiniguin, 2020-07-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I use Tasmota on Sonoff D1 from March 2020. It worked without problem.
After upgrade to 8.3.1.7 from 8.0.x.x(I am not sure) Sonoff D1 dimmer switches on and off periodically.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff D1
- [X] Tasmota binary firmware version number used: 8.3.1.7
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
16:20:44 CMD: Backlog Template; Module; GPIO 255
16:20:44 SRC: WebConsole from 10.0.1.21
16:20:44 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
16:20:44 SRC: Backlog
16:20:44 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
16:20:44 MQT: stat/hall_light/RESULT = {"NAME":"Sonoff D1 wBut","GPIO":[255,148,255,149,255,255,0,0,0,56,0,0,0],"FLAG":0,"BASE":74}
16:20:45 SRC: Backlog
16:20:45 CMD: Group 0, Index 1, Command "MODULE", Data ""
16:20:45 MQT: stat/hall_light/RESULT = {"Module":{"0":"Sonoff D1 wBut"}}
16:20:45 SRC: Backlog
16:20:45 CMD: Group 0, Index 1, Command "GPIO", Data "255"
16:20:45 MQT: stat/hall_light/RESULT = {"GPIO0":{"92":"Button3n"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"10":"Switch2"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"9":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
16:21:59 CMD: Backlog Rule1; Rule2; Rule3
16:21:59 SRC: WebConsole from 10.0.1.21
16:21:59 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
16:21:59 SRC: Backlog
16:21:59 CMD: Group 0, Index 1, Command "RULE", Data ""
16:21:59 MQT: stat/hall_light/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":86,"Free":425,"Rules":"on switch1#state=2 do POWER TOGGLE endon on switch1#state=4 do DIMMER + endon  Rule1 1"}
16:22:00 SRC: Backlog
16:22:00 CMD: Group 0, Index 2, Command "RULE", Data ""
16:22:00 MQT: stat/hall_light/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":88,"Free":423,"Rules":"on switch2#state=2 do POWER1 TOGGLE endon on switch2#state=4 do DIMMER1 - endon  Rule2 1"}

```
- [X] Provide the output of this command: `Status 0`:
```
16:22:53 CMD: status 0
16:22:53 SRC: WebConsole from 10.0.1.21
16:22:53 CMD: Group 0, Index 1, Command "STATUS", Data "0"
16:22:53 MQT: stat/hall_light/STATUS = {"Status":{"Module":0,"DeviceName":"Hall light","FriendlyName":["Hall light"],"Topic":"%hostname%","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[12,12,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:22:53 MQT: stat/hall_light/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Power On","Uptime":"0T08:51:05","StartupUTC":"2020-07-31T06:31:48","Sleep":50,"CfgHolder":4617,"BootCount":93,"BCResetTime":"2020-03-08T17:07:59","SaveCount":1899,"SaveAddress":"F5000"}}
16:22:53 MQT: stat/hall_light/STATUS2 = {"StatusFWR":{"Version":"8.3.1.7(beb4b6b-tasmota)","BuildDateTime":"2020-07-28T10:01:04","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"368/699"}}
16:22:53 MQT: stat/hall_light/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["pine2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A800A","0A05C8000100060000005A00000000000000","00000200","00006000"]}}
16:22:53 MQT: stat/hall_light/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
16:22:53 MQT: stat/hall_light/STATUS5 = {"StatusNET":{"Hostname":"hall_light","IPAddress":"10.0.1.163","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"D8:F1:5B:DB:2B:D3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:22:53 MQT: stat/hall_light/STATUS6 = {"StatusMQT":{"MqttHost":"rbpi.lan","MqttPort":1883,"MqttClientMask":"hall_light","MqttClient":"hall_light","MqttUser":"tasmota","MqttCount":5,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:22:53 MQT: stat/hall_light/STATUS7 = {"StatusTIM":{"UTC":"2020-07-31T15:22:53","Local":"2020-07-31T16:22:53","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:23","Sunset":"20:29"}}
16:22:53 MQT: stat/hall_light/STATUS10 = {"StatusSNS":{"Time":"2020-07-31T16:22:53","Switch1":"OFF","Switch2":"OFF"}}
16:22:53 MQT: stat/hall_light/STATUS11 = {"StatusSTS":{"Time":"2020-07-31T16:22:53","Uptime":"0T08:51:05","UptimeSec":31865,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Dimmer":61,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"pine2","BSSId":"C8:BE:19:5F:9F:1A","Channel":6,"RSSI":42,"Signal":-79,"LinkCount":4,"Downtime":"0T00:07:24"}}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
16:49:11 MQT: tele/hall_light/STATE = {"Time":"2020-07-30T16:49:11","Uptime":"0T19:50:20","UptimeSec":71420,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"OFF","Dimmer":17,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"pine2","BSSId":"C8:BE:19:5F:9F:1A","Channel":6,"RSSI":42,"Signal":-79,"LinkCount":2,"Downtime":"0T00:00:04"}}
16:49:11 MQT: tele/hall_light/SENSOR = {"Time":"2020-07-30T16:49:11","Switch1":"OFF","Switch2":"OFF"}
[... 3 lines cut ...]
16:51:19 MQT: tele/hall_light/STATE = {"Time":"2020-07-30T16:51:19","Uptime":"0T19:52:28","UptimeSec":71548,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":4,"POWER":"ON","Dimmer":17,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"pine2","BSSId":"C8:BE:19:5F:9F:1A","Channel":6,"RSSI":42,"Signal":-79,"LinkCount":2,"Downtime":"0T00:00:04"}}
16:51:20 MQT: stat/hall_light/RESULT = {"POWER":"ON"}
16:51:20 MQT: stat/hall_light/POWER = ON
16:54:11 MQT: tele/hall_light/STATE = {"Time":"2020-07-30T16:54:11","Uptime":"0T19:55:20","UptimeSec":71720,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":25,"MqttCount":4,"POWER":"ON","Dimmer":17,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"pine2","BSSId":"C8:BE:19:5F:9F:1A","Channel":6,"RSSI":40,"Signal":-80,"LinkCount":2,"Downtime":"0T00:00:04"}}
16:54:11 MQT: tele/hall_light/SENSOR = {"Time":"2020-07-30T16:54:11","Switch1":"OFF","Switch2":"OFF"}
16:54:28 MQT: tele/hall_light/HASS_STATE = {"Version":"8.3.1.7(beb4b6b-tasmota)","BuildDateTime":"2020-07-28T10:01:04","Module or Template":"Sonoff D1 wBut","RestartReason":"Software/System restart","Uptime":"0T19:55:37","Hostname":"hall_light","IPAddress":"10.0.1.163","RSSI":"40","Signal (dBm)":"-80","WiFi LinkCount":2,"WiFi Downtime":"0T00:00:04","MqttCount":4,"LoadAvg":27}
```

### TO REPRODUCE
Power up Sonoff D1, turn it off and wait some time - 30 minutes, 1 hours, 3 hours etc.
Sonoff D1 will turn on.
Wait again.
Sonoff D1 will turn off.
........
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Sonoff D1 should not switch without user command.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-31:
Duplicate https://github.com/arendst/Tasmota/issues/7598.

closed 2020-07-31 by arendst

epiniguin, 2020-07-31:
It does not duplicate.
I wrote that from March 2020 to July 2020 with older firmware there is no any ghost switching.
This problem appeared after firmware upgrade.

arendst, 2020-07-31:
I know and it's not the firmware itselves. It could be the load of the firmware. Try a tasmota-lite version and see what happens.

epiniguin, 2020-08-03:
Tasmota-lite does not contain Sonoff D1.

epiniguin, 2020-08-03:
I have built tasmota-lite with Sonoff D1 support.
This did not help for ghost switching.

weidingerhp, 2020-09-21:
Could this be related to https://support.itead.cc/support/discussions/topics/11000030765?

epiniguin, 2020-09-21:
I have been using D1 from March 2020. This problem appeared only after upgrade to 8.3.1.7. So it is not only hardware problem, but it is software problem.
I do not have RF control and I can't solve this problem via RF repairing.
I unsoldered two capacitors near the antenna and there is not ghost switching after that.

tomexas2, 2020-09-21:
I have the same problem now with Sonoff D1. Looking for how to solve it...

weidingerhp, 2020-09-23:
Thanks for your reply @epiniguin.

Unfortunately i did re-appear here also. Its not deterministic and it also changes the brightness randomly.
(but only after hours. sometimes it runs half a day without any problems)

Unfortunately i need the Remote-Control since i do not have any other dedicated switch to turn the lights on or off.
(use this as a dimmer directly in the power-cord to a lamp)

samezrp, 2021-01-04:
Any update regarding this issue?
I did re-pairing remote few times and problem still persist. Switching ON is result of that command incoming randomly:
`20:49:17.332 CMD: Group 0, Index 1, Command "DIMMER", Data "1"`
`20:49:17.338 MQT: stat/tasmota_5A3B01/RESULT = {"POWER":"ON","Dimmer":1}`
`20:49:17.479 DMP: AA 55 01 04 00 0A 01 01 FF FF FF FF FF FF FF FF 09`
`20:49:17.645 CFG: Saved to flash at F4, Count 16, Bytes 4096`
visible in console with full debug. Tasmota 9.2.0.2
Any idea? Maybe any rule to ignore that command?

CoderFX, 2021-03-16:
Had the same problem and the solution is quite simple - just unpair and then pair the RF remote with D1.