arendst/Tasmota #9007: PZEM 004 unstable after upgrade from 6.5 to 8.4.0

javier6d2, 2020-08-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After updating OTA sonoff basic POW R2, the PZEM device has stopped working correctly. The values of voltage, current and others show correct values but after a few seconds they go to zero and then they show the correct value again. The ESP8266 dont reboot in this process. This does not bother me, the problem is that the today energy meter also restarts and does not accumulate the value of energy consumed. In my opinion I think it is a hardware problem, maybe something has gone wrong in the PZEM. When making OTA I did not have the 1K resistor soldered in PZEM, now I have put it on but the problem persists. I have verified welds and connections and everything is correct, as I said, the problem appeared after making the OTA, the device had been working well for 11 months. Any help is appreciated. PZEM 004 is V3. I have another V3 that has been broken for some time and 2 V1 that have never had problems.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Basic Module
- [X] Tasmota binary firmware version number used: 8.4 and 6.5
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: FlashESP8266
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
Serial Out: PZEMM0XX Tx (62)
Serial In: PZEMM0XX Rx (98)

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
09:45:48 CMD: Status 0
09:45:48 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:45:48 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:00","StartupUTC":"2020-08-01T08:43:48","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":25,"SaveAddress":"FB000"}}
09:45:48 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-11-17T13:49:16","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
09:45:48 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Tomato24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
09:45:48 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":565,"Free":436,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDEE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
09:45:48 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1870","IPAddress":"192.168.130.101","Gateway":"192.168.130.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"C4:4F:33:B3:67:4E","Webserver":2,"WifiConfig":4}}
09:45:48 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.130.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B3674E","MqttUser":"javalin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
09:45:48 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Aug 01 08:45:48 2020","Local":"Sat Aug 01 09:45:48 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"05:24","Sunset":"20:28"}}
09:45:48 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:45:48 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2020-08-01T09:45:48","ENERGY":{"TotalStartTime":"2020-08-01T00:00:00","Total":330.101,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":50,"Voltage":239,"Current":0.000}}}
09:45:48 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2020-08-01T09:45:48","Uptime":"0T00:02:00","UptimeSec":120,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Tomato24","BSSId":"B0:6E:BF:B2:7F:A0","Channel":1,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Video: https://gruporolear-my.sharepoint.com/:v:/g/personal/javier_elcuaz_rolearmais_pt/ESAvteI6X3NPoefCaNvoJgsB4CLB6mswABQABONqMujedA?e=erZFZq

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-08-01:
Self answering:

>  In my opinion I think it is a hardware problem, maybe something has gone wrong in the PZEM. I have downgade to 6.7.1 with FlashESP8266 and the problem persist.

closed 2020-08-01 by arendst

richard-scott, 2020-08-06:
I've got a couple of PZEM-004Tv3's on a Wemos D1 Mini and they run just fine on 8.4.0 :-)

serbasan, 2020-08-08:
Same problem here, with 8.3.1 it worked perfect. I have updated to 8.4 and the same behavior as described here.
I have lowered to 8.3.1 but now it works badly, everything is reset and I have no history.
It seems that something has happened in 8.4 and it is not fixed when downloading the version.