arendst/Tasmota #9008: Support for 3 step dim lamps

Pewibe, 2020-08-01:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
--> See first the attached product data sheet for explaining "3 step dim value".
1. After power outage and re-appear there is always 100% of light and not the last dim value.
2. There is no control to set a specific dim value.

**Describe the solution you'd like**  
Beside button/switch "on/off" a possibility to select dim value "high/medium/low".

**Describe alternatives you've considered**  
A 4-layer switch "on=high/medium/low/off".

**Additional context**  
Today for several 3-step dim lamps spread over the house I use a rule to set "medium" dim level (switch on = 100%, wait half a second, switch off, wait again half a sec and switch on again):
`Rule1 ON Clock#Timer=1 DO Backlog Power ON; Delay 5; Power TOGGLE; Delay 5; Power TOGGLE; ENDON` 

**(Please, remember to close the issue when the problem has been addressed)**

Pewibe, 2020-08-29:
@ascillato2 I address you direct because I'm wondering if the process of this issue is usual: I requested, you assigned, and the bot says finish. Nothing else?
I like to point out that I'm not talking about an exotic sensor that 3 to 5 people need. These bulbs are used in a lot of households. If you are unsure enter the search string "led 3 way bulb" and you receive over 1000 hits. I think this is important enough to analyse this issue and prepare Tasmota more broadly based.

ascillato, 2020-08-30:
Hi,

> I'm wondering if the process of this issue is usual

I'm sorry that your feature request haven't received any comment or any Pull Request. I totally agree with your point of view, but not all feature requests will find someone that will work on it.

I personally think that this is a good one, I like it but at this moment I'm very busy, so I can't take it, Sorry.

-----------

For now, you have 2 solutions.

1 - You can use `switchmodeX 11` see https://tasmota.github.io/docs/Buttons-and-Switches/#switchmode

2 - Adapt the dimming rule example (https://tasmota.github.io/docs/Rules/#control-a-dimmer-with-one-switch).
What I can think of for achiving your request is that a button (or momentary swicth) would control a VAR and that VAR produces the dimming steps like OFF -> LOW -> MED -> HIGH -> OFF

Assumming that switch1 is a momentary switch and that `switchmode1` is 0 (toggle)

```lua
rule1 1
rule1 on switch1#state do add1 1 endon on var1#state=0 do dimmer 0 endon on var1#state=1 do dimmer 30 endon on var1#state=2 do dimmer 60 endon on var1#state=3 do dimmer 100 endon on var1#state=4 do var1 0 endon
```

Hope this helps

Pewibe, 2020-08-30:
@ascillato Thanks concerning your warm words about my lonelyness of this issue :-) 

Regarding your 2 solutions I guess you didn't hit the problem or I have a lack in understanding. See problem description (item1 and 2) in my request.
1. Today I have my working rules (exept of power outage). With additional use of MEM I think I can extent my rules.
2. The problem is that a dim command has to be converted to switching sequences:
  a. 100% = switch bulb on
  b. 50% = switch bulb on-off-on
  c. 10% = switch bulb on-off-on-off-on
  d. 0% = switch bulb off

As last question: Can you send a message to the guy who is the author of the dim modul to be sure he reads my request? If he decides to be quiet I will give up.

ascillato, 2020-09-01:
Ah, ok. Sorry. I thought you were trying to "simulate" a 3 step dim lamp by the use of a button and a dimmable lamp.

So, your request is to provide by commands an option to do blinks on the relay output of a sonoff basic (for example), in order to use the bulbs that already had the option to dim itself by switching its own power ?

Ok, to be more generic, I have found that are bulbs of 2 steps, 3 steps, 4 steps (some with incremental dimming on each step and some others backwards) and also others that can switch from warm to cold light.

As this "dimming" is not controlled by Tasmota itself, and as there are several types of these bulbs, seems better to be controlled by rules adapted to each case. Also, seems not to be confortable to the eyes if for example you send a command to Tasmota to dim at 10%, and that bulb flickers 3 times to reach that dim step.

Anyway, if you want, you can use Rule EVENTs in order to make like a DIMMER command.

Pewibe, 2020-09-01:
@ascillato You got it - sorry that my original request was not clear enough to understand. And yes, you are right that there are some deviations like number of steps, percentage of dim at step, etc. And yes, the flickering is not comfortable but there is no alternative. The bulbs have a very good price compared to real dimmable bulbs, and need no extra hardware. That's why they are often used. I personally have around 10 pieces in the household. 
Your suggestion to use rules is clear - I even do this today. To solve my 1st problem (power outage) I think I can additionally use MEM. Any hint to solve the 2nd problem with rules, so I can send a MQTT command for a specific dim value?
Anyhow, Tasmota has the construct of a dimmer for a device. Such a capability should work for widely used devices, like e.g. a LED stripe. Because the step bulbs are used so often, I thought this hardware should be included to operate the dimmer construct.

stefanbode, 2020-09-09:
I took a look at these 3-step "dimmable" lights. They do not very cool. Just an inconvenient workaround to get some different light power. I also do not understand where the ESP should be placed in. Just as a relays and expose it as a dimmer. As @ascillato  mentioned the flickering through switching through the modes is "not very cool stuff". I assume in future true dimmalbe LED get as cheap and then there is no need for these devices anymore. TASMOTA can drive AC Dimmers now. With this on  the hardware side you can define a button emulating the 3-step but also have the ability on the TASMOTA side to set it to any dimmer value.

ascillato2, 2020-09-13:
> Any hint to solve the 2nd problem with rules, so I can send a MQTT command for a specific dim value?

May be using a MEM for storing the actual state of the bulb and a VAR for requesting the new dimming state, so with a rule, you can blink the difference between the two. 

Let me try a set of rules for that. Will post here later.

closed 2020-09-13 by ascillato2