arendst/Tasmota #9010: ZBBridge: rejoin fails for some devices

s-hadinger, 2020-08-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Repairing fails after coordinator was off for an extended period of time.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff ZBBridge
- [x] Tasmota binary firmware version number used: 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
16:56:54 MQT: stat/tasmota/ZBBridge/RESULT = {"NAME":"Sonoff ZBBridge","GPIO":[56,208,0,209,21,22,0,0,6,158,5,0,17],"FLAG":0,"BASE":18}
16:56:54 MQT: stat/tasmota/ZBBridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
16:56:55 MQT: stat/tasmota/ZBBridge/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"165":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"166":"Zigbee Rx"},"GPIO4":{"215":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"158":"LedLinki"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
16:57:27 MQT: stat/tasmota/ZBBridge/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":78,"Free":433,"Rules":"on zbstate#status==0 do backlog zblisten1 0; zblisten2 99; zblisten3 100 endon"}
16:57:27 MQT: stat/tasmota/ZBBridge/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
16:57:28 MQT: stat/tasmota/ZBBridge/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```
16:58:22 MQT: stat/tasmota/ZBBridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:04:44","StartupUTC":"2020-08-01T15:53:38","Sleep":50,"CfgHolder":4617,"BootCount":578,"BCResetTime":"2020-06-11T18:29:13","SaveCount":1327,"SaveAddress":"F7000"}}
16:58:22 MQT: stat/tasmota/ZBBridge/STATUS2 = {"StatusFWR":{"Version":"8.4.0.1(tasmota)","BuildDateTime":"2020-08-01T09:47:37","Boot":31,"Core":"2_7_3_2","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"424/699"}}
16:58:50 MQT: stat/tasmota/ZBBridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Tasmota",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00C08009","2805C8000100060000005A00000000000000","00000000","00046002"]}}
16:58:50 MQT: stat/tasmota/ZBBridge/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F0007C6","04500001","00000002","00000000","00000000","00020000"],"Drivers":"1,2,4,9,10,20,21,23,41","Sensors":""}}
16:58:50 MQT: stat/tasmota/ZBBridge/STATUS5 = {"StatusNET":{"Hostname":"tasmota/ZBBridge-0879","IPAddress":"192.168.1.92","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"B4:E6:2D:7F:C3:6F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:58:50 MQT: stat/tasmota/ZBBridge/STATUS6 = {"StatusMQT":{"MqttHost":"a3pa4ktnq87yfu-ats.iot.eu-central-1.amazonaws.com","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7FC36F","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:58:50 MQT: stat/tasmota/ZBBridge/STATUS7 = {"StatusTIM":{"UTC":"2020-08-01T15:58:50","Local":"2020-08-01T16:58:50","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:25","Sunset":"20:28"}}
16:58:50 MQT: stat/tasmota/ZBBridge/STATUS10 = {"StatusSNS":{"Time":"2020-08-01T16:58:50"}}
16:58:50 MQT: stat/tasmota/ZBBridge/STATUS11 = {"StatusSTS":{"Time":"2020-08-01T16:58:50","Uptime":"0T00:05:12","UptimeSec":312,"Vcc":3.504,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Tasmota","BSSId":"76:83:C2:90:D6:15","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
```

### TO REPRODUCE
_Steps to reproduce the behavior:

I paired on OSRAM mini-switch successfully. If I turn down the coordinator for a significant period (more than a day), the device loses the pairing and does an automatic rejoin. The rejoin is accepted by the coordinator, but the OSRAM device keeps sending rejoins requests for ever.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Using CC2530 the issue does not happen. There is something wrong with routing on ZBBridge maybe linked to Zigbee V3.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

This is a follow up from #8583 to have an issue dedicated to this problem.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-08-08:
Cannot reproduce anymore, it's probable that #9038 fixed this one too.

Closing.

closed 2020-08-08 by s-hadinger

bogorad, 2020-09-24:
I still have this problem. ZBBridge has three devices paired: 

- Xiaomi button
- Aqara motion sensor
- Aqara door sensor

Only the button survives extended (over 24hrs) power loss. 

I've got a self-compiled (+TLS). Maybe there's a more recent build?

Program Version: 8.5.0(tasmota)
Build Date & Time: 2020-09-09T10:30:23
Core/SDK Version: 2_7_4_1/2.2.2-dev(38a443e)

s-hadinger, 2020-09-25:
@bogorad The re-joining is part is managed by EZSP directly. Do you happen to have a Zigbee sniffer? I'm afraid that's the only way to know why the rejoining fails.

bogorad, 2020-09-25:
@s-hadinger I don't, but thank you. hopefully, with the latest build, it'll go away.