arendst/Tasmota #9012: Tasmota 8.4.0 loses the Thermostat settings

dexterbot80, 2020-08-02:
# PROBLEM DESCRIPTION

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
- [x] Self-compiled V8.4.0
- [x] IDE / Compiler used: Arduino IDE 1.8.12
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here: 
06:13:01 CMD: Backlog Template; Module; GPIO 255
06:13:01 MQT: stat/office_thermostat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
06:13:01 MQT: stat/office_thermostat/RESULT = {"Module":{"18":"Generic"}}
06:13:01 MQT: stat/office_thermostat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"4":"DS18x20"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
06:14:32 CMD: Backlog Rule1; Rule2; Rule3
06:14:33 MQT: stat/office_thermostat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
06:14:33 MQT: stat/office_thermostat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
06:14:33 MQT: stat/office_thermostat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
06:15:17 CMD: Status 0
06:15:17 MQT: stat/office_thermostat/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"office_thermostat","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:15:17 MQT: stat/office_thermostat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T01:07:18","StartupUTC":"2020-08-02T04:07:59","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-08-01T12:30:34","SaveCount":105,"SaveAddress":"FB000"}}
06:15:17 MQT: stat/office_thermostat/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-08-01T22:07:34","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"357/699"}}
06:15:17 MQT: stat/office_thermostat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["rdspro_IoT",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00008000","00006000"]}}
06:15:17 MQT: stat/office_thermostat/STATUS4 = {"StatusMEM":{"ProgramSize":658,"Free":280,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","00368001","000000CD","010013C0","C000F981","00004404"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,39","Sensors":"1,2,3,4,5,6"}}
06:15:17 MQT: stat/office_thermostat/STATUS5 = {"StatusNET":{"Hostname":"office_thermostat-1241","IPAddress":"192.168.1.203","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"50:02:91:69:84:D9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
06:15:17 MQT: stat/office_thermostat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6984D9","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
06:15:17 MQT: stat/office_thermostat/STATUS7 = {"StatusTIM":{"UTC":"2020-08-02T05:15:17","Local":"2020-08-02T06:15:17","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:25","Sunset":"20:26"}}
06:15:17 MQT: stat/office_thermostat/STATUS10 = {"StatusSNS":{"Time":"2020-08-02T06:15:17","DS18B20":{"Id":"00000AFBF3D7","Temperature":27.4},"TempUnit":"C"}}
06:15:17 MQT: stat/office_thermostat/STATUS11 = {"StatusSTS":{"Time":"2020-08-02T06:15:17","Uptime":"0T01:07:18","UptimeSec":4038,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"rdspro_IoT","BSSId":"F0:9F:C2:6A:72:27","Channel":13,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  **I connected the wemos D1 mini power supply and set it ThermostatModeSet 1  in the  console**
```
  Console output here:
00:00:00 Project tasmota Tasmota Version 8.4.0(tasmota)-2_7_1
00:00:00 RSL: stat/office_thermostat/RESULT = {"POWER":"OFF"}
00:00:00 RSL: stat/office_thermostat/POWER = OFF
00:00:00 WIF: Connecting to AP1 rdspro_IoT Channel 13 BSSId F0:9F:C2:6A:72:27 in mode 11N as office_thermostat-1241...
00:00:01 WIF: Connected
06:17:07 HTP: Web server active on office_thermostat-1241 with IP address 192.168.1.203
06:17:08 MQT: Attempting connection...
06:17:08 MQT: Connected
06:17:08 MQT: tele/office_thermostat/LWT = Online (retained)
06:17:08 MQT: cmnd/office_thermostat/POWER = 
06:17:08 MQT: tele/office_thermostat/INFO1 = {"Module":"Generic","Version":"8.4.0(tasmota)","FallbackTopic":"cmnd/DVES_6984D9_fb/","GroupTopic":"cmnd/tasmotas/"}
06:17:08 MQT: tele/office_thermostat/INFO2 = {"WebServerMode":"Admin","Hostname":"office_thermostat-1241","IPAddress":"192.168.1.203"}
06:17:08 MQT: tele/office_thermostat/INFO3 = {"RestartReason":"External System"}
06:17:08 MQT: stat/office_thermostat/RESULT = {"POWER":"OFF"}
06:17:08 MQT: stat/office_thermostat/POWER = OFF
06:17:12 MQT: tele/office_thermostat/STATE = {"Time":"2020-08-02T06:17:12","Uptime":"0T00:00:09","UptimeSec":9,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"rdspro_IoT","BSSId":"F0:9F:C2:6A:72:27","Channel":13,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}
[... 18 lines cut ...]
06:24:33 MQT: Attempting connection...
06:24:33 MQT: Connected
06:24:33 MQT: tele/office_thermostat/LWT = Online (retained)
06:24:33 MQT: cmnd/office_thermostat/POWER = 
06:24:33 MQT: tele/office_thermostat/INFO1 = {"Module":"Generic","Version":"8.4.0(tasmota)","FallbackTopic":"cmnd/DVES_6984D9_fb/","GroupTopic":"cmnd/tasmotas/"}
06:24:33 MQT: tele/office_thermostat/INFO2 = {"WebServerMode":"Admin","Hostname":"office_thermostat-1241","IPAddress":"192.168.1.203"}
06:24:33 MQT: tele/office_thermostat/INFO3 = {"RestartReason":"External System"}
06:24:33 MQT: stat/office_thermostat/RESULT = {"POWER":"OFF"}
06:24:33 MQT: stat/office_thermostat/POWER = OFF
06:24:37 MQT: tele/office_thermostat/STATE = {"Time":"2020-08-02T06:24:37","Uptime":"0T00:00:09","UptimeSec":9,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"rdspro_IoT","BSSId":"F0:9F:C2:6A:72:27","Channel":13,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}
06:24:37 MQT: tele/office_thermostat/SENSOR = {"Time":"2020-08-02T06:24:37","DS18B20":{"Id":"00000AFBF3D7","Temperature":27.6},"TempUnit":"C"}
06:24:39 CMD: ThermostatModeSet
06:24:39 MQT: stat/office_thermostat/RESULT = {"ThermostatModeSet":0}
```

### EXPECTED BEHAVIOUR
After reboot Thasmota device loses Thermostat settings

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-08-02:
Hi,

This is not a bug. The driver was programmed like that and is not saving its data in flash..

If you want to hardcode your values, you need to write them in your file tasmota/my_user_config.h OR in a rule for SYSTEM#BOOT

dexterbot80, 2020-08-02:
Thanks for your answer,
I use this Thermostat control options.
```
// -- Thermostat control ----------------------------
  #define USE_THERMOSTAT                                  // Add support for Thermostat
  #define THERMOSTAT_CONTROLLER_OUTPUTS         1         // Number of outputs to be controlled independently
  #define THERMOSTAT_SENSOR_NAME                "DS18B20" // Name of the local sensor to be used
  #define THERMOSTAT_RELAY_NUMBER               1         // Default output relay number for the first controller (+i for following ones)
  #define THERMOSTAT_SWITCH_NUMBER              1         // Default input switch number for the first controller (+i for following ones)
  #define THERMOSTAT_TIME_ALLOW_RAMPUP          300       // Default time after last target update to allow ramp-up controller phase in minutes
  #define THERMOSTAT_TIME_RAMPUP_MAX            960       // Default time maximum ramp-up controller duration in minutes
  #define THERMOSTAT_TIME_RAMPUP_CYCLE          30        // Default time ramp-up cycle in minutes
  #define THERMOSTAT_TIME_SENS_LOST             30        // Maximum time w/o sensor update to set it as lost in minutes
  #define THERMOSTAT_TEMP_SENS_NUMBER           1         // Default temperature sensor number
  #define THERMOSTAT_TIME_MANUAL_TO_AUTO        60        // Default time without input switch active to change from manual to automatic in minutes
  #define THERMOSTAT_TIME_RESET                 12000     // Default reset time of the PI controller in seconds
  #define THERMOSTAT_TIME_PI_CYCLE              30        // Default cycle time for the thermostat controller in minutes
  #define THERMOSTAT_TIME_MAX_ACTION            20        // Default maximum thermostat time per cycle in minutes
  #define THERMOSTAT_TIME_MIN_ACTION            4         // Default minimum thermostat time per cycle in minutes
  #define THERMOSTAT_TIME_MIN_TURNOFF_ACTION    3         // Default minimum turnoff time in minutes, below it the thermostat will be held on
  #define THERMOSTAT_PROP_BAND                  4         // Default proportional band of the PI controller in degrees celsius
  #define THERMOSTAT_TEMP_RESET_ANTI_WINDUP     8         // Default range where reset antiwindup is disabled, in tenths of degrees celsius
  #define THERMOSTAT_TEMP_HYSTERESIS            1         // Default range hysteresis for temperature PI controller, in tenths of degrees celsius
  #define THERMOSTAT_TEMP_FROST_PROTECT         40        // Default minimum temperature for frost protection, in tenths of degrees celsius
  #define THERMOSTAT_TEMP_RAMPUP_DELTA_IN       4         // Default minimum delta temperature to target to get into rampup mode, in tenths of degrees celsius
  #define THERMOSTAT_TEMP_RAMPUP_DELTA_OUT      2         // Default minimum delta temperature to target to get out of the rampup mode, in tenths of degrees celsius
  #define THERMOSTAT_TEMP_PI_RAMPUP_ACC_E       200       // Default accumulated error when switching from ramp-up controller to PI in hundreths of degrees celsius
  #define THERMOSTAT_TIME_OUTPUT_DELAY          180       // Default output delay between state change and real actuation event (f.i. valve open/closed)
  #define THERMOSTAT_TEMP_INIT                  180       // Default init target temperature for the thermostat controller
  #define THERMOSTAT_TIME_MAX_OUTPUT_INCONSIST  3         // Default maximum time where the input and the outpus shall differ (for diagnostic) in minutes
  #define THERMOSTAT_TIME_MAX_AUTOTUNE          21600     // Maximum time for the PI autotune function to complete in seconds
  #define THERMOSTAT_DUTYCYCLE_AUTOTUNE         35        // Default duty cycle (in % over PI cycle time) for the step response of the autotune PI function
  #define THERMOSTAT_PEAKNUMBER_AUTOTUNE        8         // Default number of peak temperatures (max or min) to be used for the autotune PI function
  #define THERMOSTAT_TEMP_BAND_NO_PEAK_DET      1         // Default temperature band in thenths of degrees celsius within no peak will be detected
  #define THERMOSTAT_TIME_STD_DEV_PEAK_DET_OK   10        // Default standard deviation in minutes of the oscillation periods within the peak detection is successful
```
I don't have to activate the line: #define USE_THERMOSTAT in my_user_config.h?

ascillato, 2020-08-02:
If you want to compile with the Thermostat driver included, you need to enable it by:

#define USE_THERMOSTAT in my_user_config.h

Or

#define USE_THERMOSTAT in user_config_override.h

Or

Activating the define in platformio.ini

dexterbot80, 2020-08-02:
Yes,
I use second option 
```
user_config_override.h
#ifndef USE_THERMOSTAT
#define USE_THERMOSTAT
#endif
```
```
support_features.ino
#ifdef USE_OPENTHERM
  feature6 |= 0x00000200;  // xsns_69_opentherm.ino
#endif
#ifdef USE_THERMOSTAT
  //feature6 |= 0x00000400;  // xdrv_39_heating.ino  **I commented this line**
#endif
```
```
tasmota.h
enum CommandSource { SRC_IGNORE, SRC_MQTT, SRC_RESTART, SRC_BUTTON, SRC_SWITCH, SRC_BACKLOG, SRC_SERIAL, SRC_WEBGUI, SRC_WEBCOMMAND, SRC_WEBCONSOLE, SRC_PULSETIMER,
                     SRC_TIMER, SRC_RULE, SRC_MAXPOWER, SRC_MAXENERGY, SRC_OVERTEMP, SRC_LIGHT, SRC_KNX, SRC_DISPLAY, SRC_WEMO, SRC_HUE, SRC_RETRY, SRC_REMOTE, SRC_SHUTTER,
                     //SRC_MAX }; **I deleted this line**
                     SRC_THERMOSTAT, SRC_MAX };
const char kCommandSource[] PROGMEM = 
                          "I|MQTT|Restart|Button|Switch|Backlog|Serial|WebGui|WebCommand|WebConsole|PulseTimer|"
                          //"Timer|Rule|MaxPower|MaxEnergy|Overtemp|Light|Knx|Display|Wemo|Hue|Retry|Remote|Shutter";  **I deleted this line**
                                 
                          "Timer|Rule|MaxPower|MaxEnergy|Overtemp|Light|Knx|Display|Wemo|Hue|Retry|Remote|Shutter|Thermostat";

const uint8_t kDefaultRfCode[9] PROGMEM = { 0x21, 0x16, 0x01, 0x0E, 0x03, 0x48, 0x2E, 0x1A, 0x00 };
```
What am I doing wrong here?

arendst, 2020-08-02:
A LOT!!!!

Users are supposed to only change user_config_override.h

You'll need to copy the wanted parameters from my_userconfig.h to user_config_override.h and change them there. No need to change any other files as it will break your future upgrades.

dexterbot80, 2020-08-02:
> A LOT!!!!
> 
> Users are supposed to only change user_config_override.h
> 
> You'll need to copy the wanted parameters from my_userconfig.h to user_config_override.h and change them there. No need to change any other files as it will break your future upgrades.
This is exactly what I did at the beginning, but the thermostat settings do not remain saved after the restart.
A pre-compiled code would help me - for me your source code is quite complicated teacher :)
I just want to be able to use the thermostat function, which I find extremely useful.
Thank you for your patience with me

curzon01, 2020-08-03:
your origin way enabling the module using `#define USE_THERMOSTAT`in user_config_override.h was the correct one

To permanent change a setting like `ThermostatModeSet` in Tasmota which is not stored in flash by Tasmota itself there is no need to change source code.

Use one of the alternative methode instead:
- set a rule which executes during boot, e.g. `Rule1 ON System#Boot DO ThermostatModeSet 1 ENDON`, then enable the rule with `Backlog Rule1 1;Rule1 4`
- or (without rules) set a retained MQTT msg which setup the Tasmota command, .e.g a retainted MQTT topic `cmnd/office_thermostat/ThermostatModeSet` with payload `1` will setup `ThermostatModeSet 1` each time Tasmota connects to your MQTT broker

closed 2020-08-03 by ascillato2