arendst/Tasmota #9016: AS3935 lost on restarts

rowanalex123, 2020-08-02:
### PROBLEM DESCRIPTION
AS3935 is detected on I2C on cold starts but goes missing if the device is restarted

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): node mcu
- [ ] Tasmota binary firmware version number used: 8.4.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
19:28:32 CMD: Backlog Template; Module; GPIO 255
19:28:32 RSL: stat/tasmota_FC5E57/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:28:32 RSL: stat/tasmota_FC5E57/RESULT = {"Module":{"0":"Generic"}}
19:28:33 RSL: stat/tasmota_FC5E57/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:30:42 CMD: Status 0
19:30:42 RSL: stat/tasmota_FC5E57/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota_FC5E57","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:30:42 RSL: stat/tasmota_FC5E57/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:01:09","StartupUTC":"2020-08-02T18:29:33","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2020-08-02T18:15:12","SaveCount":86,"SaveAddress":"F6000"}}
19:30:42 RSL: stat/tasmota_FC5E57/STATUS2 = {"StatusFWR":{"Version":"8.3.0(sensors)","BuildDateTime":"2020-05-14T16:18:52","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"325/699"}}
19:30:42 RSL: stat/tasmota_FC5E57/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["abcd",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000"]}}
19:30:42 RSL: stat/tasmota_FC5E57/STATUS4 = {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":2,"Features":["00000809","8FDAA787","0415A3A4","B7FFBFCD","01DA9BC4","64367CC7","00000072"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67"}}
19:30:42 RSL: stat/tasmota_FC5E57/STATUS5 = {"StatusNET":{"Hostname":"espgeiger","IPAddress":"192.168.1.118","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:FC:5E:57","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:30:42 RSL: stat/tasmota_FC5E57/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FC5E57","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:30:42 RSL: stat/tasmota_FC5E57/STATUS7 = {"StatusTIM":{"UTC":"2020-08-02T18:30:42","Local":"2020-08-02T19:30:42","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:25","Sunset":"20:26"}}
19:30:42 RSL: stat/tasmota_FC5E57/STATUS10 = {"StatusSNS":{"Time":"2020-08-02T19:30:42","AS3935":{"Event":0,"Distance":0,"Energy":0}}}
19:30:42 RSL: stat/tasmota_FC5E57/STATUS11 = {"StatusSTS":{"Time":"2020-08-02T19:30:42","Uptime":"0T00:01:09","UptimeSec":69,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Wifi":{"AP":1,"SSId":"abcd","BSSId":"D4:6E:0E:36:CA:C8","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

00:00:00 CFG: Loaded from flash at F6, Count 86
00:00:00 Project tasmota Tasmota Version 8.3.0(sensors)-2_7_1
00:00:00 WIF: Connecting to AP1 abcd Channel 11 BSSId D4:6E:0E:36:CA:C8 in mode 11N as espgeiger...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on espgeiger with IP address 192.168.1.118
19:31:36 RSL: tele/tasmota_FC5E57/INFO1 = {"Module":"Generic","Version":"8.3.0(sensors)","FallbackTopic":"cmnd/DVES_FC5E57_fb/","GroupTopic":"cmnd/tasmotas/"}
19:31:36 RSL: tele/tasmota_FC5E57/INFO2 = {"WebServerMode":"Admin","Hostname":"espgeiger","IPAddress":"192.168.1.118"}
19:31:36 RSL: tele/tasmota_FC5E57/INFO3 = {"RestartReason":"Software/System restart"}
19:31:40 RSL: tele/tasmota_FC5E57/STATE = {"Time":"2020-08-02T19:31:40","Uptime":"0T00:00:09","UptimeSec":9,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"abcd","BSSId":"D4:6E:0E:36:CA:C8","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:31:46 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3456, room 16
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
00:00:00 CFG: Loaded from flash at F5, Count 87
00:00:00 Project tasmota Tasmota Version 8.3.0(sensors)-2_7_1
00:00:00 WIF: Connecting to AP1 abcd Channel 11 BSSId D4:6E:0E:36:CA:C8 in mode 11N as espgeiger...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on espgeiger with IP address 192.168.1.118
19:31:51 RSL: tele/tasmota_FC5E57/INFO1 = {"Module":"Generic","Version":"8.3.0(sensors)","FallbackTopic":"cmnd/DVES_FC5E57_fb/","GroupTopic":"cmnd/tasmotas/"}
19:31:51 RSL: tele/tasmota_FC5E57/INFO2 = {"WebServerMode":"Admin","Hostname":"espgeiger","IPAddress":"192.168.1.118"}
19:31:51 RSL: tele/tasmota_FC5E57/INFO3 = {"RestartReason":"Software/System restart"}
19:31:55 RSL: tele/tasmota_FC5E57/STATE = {"Time":"2020-08-02T19:31:55","Uptime":"0T00:00:09","UptimeSec":9,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"abcd","BSSId":"D4:6E:0E:36:CA:C8","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
```

### TO REPRODUCE
Add the DFRobot AS3935 board and click on the restart button on the UI, click Ok to confirm

### EXPECTED BEHAVIOUR
AS3935 is detected on restart and appears on the UI. Instead there are no devices detected.

### ADDITIONAL CONTEXT
 If I add a BME680, it comes back without issues after restarts.

**(Please, remember to close the issue when the problem has been addressed)**

rowanalex123, 2020-08-02:
Noticed this in the companis implementation, looks like there's a need to wait for 2ms. See https://github.com/DFRobot/DFRobot_AS3935/blob/master/RaspberryPi/Python/DFRobot_AS3935_Lib.py#L113

ascillato2, 2020-08-02:
Hi,

Can you update to latest Tasmota and try again please?

Your template shows **_Tasmota binary firmware version number used: 8.4.0_**
But your status 0 shows (`"Version":"8.3.0(sensors)"`)

Please update and try again. Thanks.

ascillato, 2020-08-04:
@rowanalex123 

Hi, any news on this? Were you able to try with latest Tasmota?

rowanalex123, 2020-08-04:
@ascillato Yes I tried initially on 8.4.0 and then downgraded to 8.3.0 to see if this was a regression bug. AS3935 disapears after a "restart 1" command on both versions.

joba-1, 2020-08-05:
an app restart probably does not reset the sensor. Maybe it does not like to be setup twice?
Then toggling bit 0 (power) of addr 0x00 and waiting a bit at the start of AS3539Setup() might help.

device111, 2020-08-06:
> 
> 
> an app restart probably does not reset the sensor. Maybe it does not like to be setup twice?
> Then toggling bit 0 (power) of addr 0x00 and waiting a bit at the start of AS3539Setup() might help.

this might be not helped, but the sensor is missing and we don't can send commands.

@rowanalex123 :
do you have additional sensors on the bus?
what does I2C scan indicate after restart?
give please a status0 after a fail restart.

thanks

arendst, 2020-08-06:
Sorry, I can't help.

I don't understand the device and have no clue how to get rid of the abundance of mqtt messages and gui texts like "noise detected","interrupt with no event","disturber detected". I guess it has something to do with the capacitor data to be entered but  my device has a (standard it seems) sticker with the standard "MA5532-AE" label. No clue what 8-bit value (deducted from the code) I should use here.

But at least I can say the device always pops back up after a restart ;-) so you likely have a power issue as @device111 suggested.

device111, 2020-08-06:
yes, the device had a lot of messages and a lot of things to adjust.
The device itself is verry complex and you have to study the datasheet.
You can watch  the youtube Video for a better understanding the settings and the messages. (21 minutes!)

[](url) https://www.youtube.com/watch?v=fTEEDfWbGEU  

otherwise you can go to
[](url) https://github.com/arendst/Tasmota/pull/8130
or contact  @ludwich66. Ludwich has a lot of Experience with the device.

@arendst 
The sticker with the "tuningcap" Value is apparently stuck on the outside of the packaging. If the sticker is not there, you can calibrate the sensor with "AS3935calibrate". The device should now calibrated.

The funny thing is, I never had a sensor...my order and my money fell victim to the COVID crisis and the sensor never arrived. 
All the tests were made by ludwich66. :-)

rowanalex123, 2020-08-07:
I don't have any other device on the i2c bus. If I add a bme680, it comes up after restart correctly. I'm away from my computer for the next one week, will try setting toggling bit 0 (power) of addr 0x00 and post here. I'm using the df robot breakout board of this sensor.

device111, 2020-08-08:
@rowanalex123 
can you try the bin here? It reset the device to there defaults every restart and then load the settings from eeprom. 

[tasmota.bin.gz](https://github.com/arendst/Tasmota/files/5045445/tasmota.bin.gz)

alfista2600, 2020-08-09:
FWIW, the Sparkfun version of this sensor says it is not supported on I2C, but they left that code there anyway. SPI is how they recommend interfacing. I have one but haven't submitted a request to have it supported over SPI.

device111, 2020-08-09:
thats definitiv not true. the Sparkfun Sensor have full I2c support and even with a 2nd address.

[image]

alfista2600, 2020-08-09:
I was not clear in my post. They left the circuitry not code. 

See notes here may only apply to V2 of the board

https://learn.sparkfun.com/tutorials/sparkfun-as3935-lightning-detector-hookup-guide-v20?_ga=2.115263175.1580721257.1597002705-930883413.1597002705

device111, 2020-08-09:
ah, ok. :-)
they don't give code support for I2c, but the sensor can be used with I2c.

alfista2600, 2020-08-09:
Yes, but it may be why you are having inconsistent/intermittent issues:

"SparkFun does not actively support I2C on this product due to the inconsistency of its' implementation by its' manufacturer AMS and their lack of active support of the product."

device111, 2020-08-10:
ok, do you know what errors can occur? they don't write about the detailed issues.

rowanalex123, 2020-08-13:
@device111  thanks for the build.

There's a description of some issues with the device here esp. in the comments, in case you've not come across it before - http://fll-freak.com/blog/?p=554.

I tried with a different NodeMCU board and the AS3935 seems to be coming back after restarts now. I dont know how the board influences the I2C device scan though. I'll keep testing and post here if I encounter the issue with this new board.

Could you please share the code changes you made? I'd like to add them back and retest if I face an issue again.

device111, 2020-08-13:
thanks for the link. So I didn't know the article before.

Several problems are described:

1. Problem with register 0x00:
The problem should not actually occur with Tasmota, since all registers are read with the `I2cRead8()` function and the function does always a "Repeated Start".
But, in your case this could eventually the Issue. The The chip is identified using the 0x00 address. So, we need more investigation, what extactly the Problem is. (So we can read the Chip always after Cold Boot)

2. Problems with the board / antenna.
Unfortunately there is nothing we can do with the Code.

3. Problems with the IRQ pin. This behavior cannot be suppressed even by SPI communication. This is mostly noticeable in Tasmota with "IRQ with no event".

ok, some questions:
1. you have no Problems with the new Board?
2. Do you have tested both Versions of Tasmota with the new board? (Normal and my modified)
3. What does I2Cscan says if the AS3935 was not detected?

at he moment a make a refactor of the driver and if its ready, i do a Pull-Request with the changes i made.

alfista2600, 2020-08-17:
Not sure if it is helpful but here is some code that runs on an ESP8266 and connects to the sensor over SPI. It uses a REST API to make the data available. I haven't had any lightning in a while so not sure if it is working or not.

https://github.com/shred/kaminari

device111, 2020-08-18:
there is no reason to switch to SPI at the moment.

Jason2866, 2020-08-23:
All sonoff devices do NOT have a good stable power supply.
For such a high sensitive sensor a sonoff device is not a good choice

device111, 2020-08-30:
@mar565 
what else you have to consider:
I don't know your Board but you will need to wire up all of the associated pins for the different boards.

For I2C Use:
- Always connect CS and MISO to GND
- If "SI" is present, you need to connect to VCC

device111, 2020-09-03:
new version. go to [9221](https://github.com/arendst/Tasmota/pull/9221)

device111, 2020-09-03:
@ascillato2 
can be closed.

closed 2020-09-03 by ascillato2

philpetnz, 2020-11-11:
> 
> 
> I was not clear in my post. They left the circuitry not code.
> 
> See notes here may only apply to V2 of the board
> 
> https://learn.sparkfun.com/tutorials/sparkfun-as3935-lightning-detector-hookup-guide-v20?_ga=2.115263175.1580721257.1597002705-930883413.1597002705

Not sure if it's relevant, but it worked for me...
Most* dev. boards have pull-downs on I2C address pins (A0 & A1)
Page 24, in Data Sheet states in bold
"The combination a0 = 0 (low) and a1 =0 (low) is explicitly not allowed for I²C communication."
I pulled both A0 & A1 HIGH & now everything is working reliable. [ESP-07 & CJMCPU-AS3935]

[image]

device111, 2020-11-11:
Ok, thanks!
If you want, Feel free to enter the changes in the Tasmota Docs.

hadahop, 2021-09-22:
@rowanalex123 
Hi,
I also have a **DFRobot AS3935** board and can't get it connected to a D1Mini to work.

[image]

**How did you do that ?**

The DFRobot AS3935 board is recognized, but no lightning bolts are displayed. 

```log
11:06:06.001 -> 00:00:00.001 HDW: ESP8266EX
11:06:06.035 -> 00:00:00.050 CFG: Loaded from flash at FA, Count 10
11:06:06.035 -> 00:00:00.055 QPC: Count 1
11:06:06.070 -> 00:00:00.065 Project tasmota - Tasmota Version 9.5.0.8(sensors)-2_7_4_9(2021-09-21T12:38:43)
11:06:06.246 -> 00:00:00.250 I2C: AS3935 found at 0x3
11:06:08.794 -> 00:00:02.006 HTP: Web server active on tasmota-4EFD7B-7547 with IP address 10.1.1.118
11:06:09.820 -> 10:06:09.030 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"9.5.0.8(sensors)","FallbackTopic":"cmnd/DVES_4EFD7B_fb/","GroupTopic":"cmnd/tasmotas/"}}
11:06:09.820 -> 10:06:09.034 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-4EFD7B-7547","IPAddress":"10.1.1.118"}}
11:06:09.854 -> 10:06:09.045 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
11:06:13.035 -> 10:06:12.217 QPC: Reset
11:06:13.932 -> 10:06:13.138 RSL: STATE = {"Time":"2021-09-22T10:06:13","Uptime":"0T00:00:09","UptimeSec":9,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Microwelle","BSSId":"78:8A:20:2A:70:8D","Channel":8,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
11:06:13.966 -> 10:06:13.160 RSL: SENSOR = {"Time":"2021-09-22T10:06:13","AS3935":{"Event":0,"Distance":0,"Energy":0,"Stage":7}}
11:11:13.980 -> 10:11:13.170 RSL: STATE = {"Time":"2021-09-22T10:11:13","Uptime":"0T00:05:09","UptimeSec":309,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Microwelle","BSSId":"78:8A:20:2A:70:8D","Channel":8,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
11:11:13.980 -> 10:11:13.192 RSL: SENSOR = {"Time":"2021-09-22T10:11:13","AS3935":{"Event":0,"Distance":0,"Energy":0,"Stage":7}}
11:16:04.958 -> 10:16:04.170 APP: Serial logging disabled

```

**Do I have to change anything on the DFRobot AS3935 board?** 
[image]

Current Settings:
[image]