arendst/Tasmota #9017: MQTT - Send new value instantly instead of periodically

Winteriver, 2020-08-02:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
Realted to the problem: 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-08-02:
Hi,

A teleperiod is a telemetry message (is a status message) and is not intended for your stated use case.

What you need is to use **_rules_** to check how much fluid have flown. Those are checked by default every 200msec, but you can lower that time with `SetOption34`

Please, check the docs for rules and check the rules cookbook for examples.

If you need further assistance with this, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-02 by ascillato2

nord3214, 2020-08-27:
@Winteriver 
I had the same issue or atleast i wanted to determine the time when the info gets send. I have found out that if you just use the command:

`teleperiod now` 

instead of a number behind it sends the data instantly. I am Using io Broker and on my dashboard it just works fine.
Hope that helps.

abc2006, 2022-10-14:
@nord3214 in which version does this work? I always get the answer 
19:45:59.581 MQT: stat/tasmota_56F37A/RESULT = {"TelePeriod":10}

But i will check for rules, too. thx

sfromis, 2022-10-14:
Well, just issuing the TelePeriod command (even with an invalid useless argument like `now`) will result in the current TelePeriod value being printed, **and** the TelePeriod messages being published to MQTT (which should very much be visible in the console).

abc2006, 2022-10-14:
Hi, thank you for this quick answer. As i am not that used to Tasmota yet, i am looking forward to decrease the MQTT-Update-Period of my power-meter-reader to 2, max 3 Seconds. Which solution do you recommend, if TelePeriod is *not* the right way to do that? Should i actually do this with rules?

sfromis, 2022-10-14:
When it is energy monitoring, do not try getting more granular data by a low reporting interval, use [PowerDelta](https://tasmota.github.io/docs/Commands/#powerdelta) instead. This feature is for generating extra reporting when the consumption level changes, which works well for most devices. As a starting point, I like something like `PowerDelta 110` to get reporting when 10W up/down, but that obviously depends on what you are monitoring.