arendst/Tasmota #9020: ESP32 Deepsleep

helmar74, 2020-08-03:
Hi,

I'm just playing around with Tasmota on ESP32.
I figured out, that Deepsleep isn't working for ESP32.

Does someone know, if it is planned to add deepsleep to ESP32 Firmware?

Thanks a lot

**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
DEEPSLEEP on ESP32 not working

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Command "DEEPSLEEPTIME" also works on ESP32

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
N.A.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-08-07:
Easier than expected. Let me do a small smoke test and I will commit the new sources.

stefanbode, 2020-08-07:
Does someone has an idea why the driver is not compiled in tasmota32? Same with tasmota ESP8266 works.

ERROR:

13:04:40 CMD: deepsleeptime
13:04:40 MQT: hm/status/tasmota_058EC8/RESULT = {"**Command":"Unknown"**}

```
00:00:08 MQT: hm/status/tasmota_058EC8/INFO1 = {"Module":"ESP32-DevKit","Version":"8.3.1.7(tasmota)","FallbackTopic":"cmnd/DVES_058EC8_fb/","GroupTopic":"hm/setting/tasmotas/"}
00:00:08 MQT: hm/status/tasmota_058EC8/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_058EC8-3784","IPAddress":"192.168.178.106"}
00:00:08 MQT: hm/status/tasmota_058EC8/INFO3 = {"RestartReason":"RTC Watch dog reset digital core and rtc module"}
00:00:12 MQT: hm/status/tasmota_058EC8/STATE = {"Time":"1970-01-01T00:00:12","Uptime":"0T00:00:14","UptimeSec":14,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BodeNet_Keller","BSSId":"38:10:D5:14:E6:70","Channel":1,"RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:08"}}
13:04:40 CMD: deepsleeptime
13:04:40 MQT: hm/status/tasmota_058EC8/RESULT = {"Command":"Unknown"}

13:08:34 MQT: hm/status/tasmota_058EC8/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_058EC8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
13:08:34 MQT: hm/status/tasmota_058EC8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.178.35/firmware.bin","RestartReason":"RTC Watch dog reset digital core and rtc module","Uptime":"0T00:04:18","StartupUTC":"2020-08-07T11:04:16","Sleep":50,"CfgHolder":4627,"BootCount":8,"BCResetTime":"2020-07-03T12:16:49","SaveCount":24}}
13:08:34 MQT: hm/status/tasmota_058EC8/STATUS2 = {"StatusFWR":{"Version":"8.3.1.7(tasmota)","BuildDateTime":"2020-08-07T12:54:59","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":160,"Hardware":"ESP32","CR":"396/699"}}
13:08:34 MQT: hm/status/tasmota_058EC8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.178.96","LogPort":514,"SSId":["BodeNet_Keller","SchreiberWLAN"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000080E9","2805C8000100060000005A00000000000000","00000000","00006000"]}}
13:08:34 MQT: hm/status/tasmota_058EC8/STATUS4 = {"StatusMEM":{"ProgramSize":1037,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F904687","00140001","000084C5","050013D1","40000381","00004000"],"Drivers":"1,2,3,4,7,8,9,10,20,21,24,26,27,28","Sensors":"1,2,3,5,6,9,12,29,45"}}
13:08:34 MQT: hm/status/tasmota_058EC8/STATUS5 = {"StatusNET":{"Hostname":"tasmota_058EC8-3784","IPAddress":"192.168.178.106","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"24:0A:C4:05:8E:C8","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
13:08:34 MQT: hm/status/tasmota_058EC8/STATUS6 = {"StatusMQT":{"MqttHost":"elocm.synology.me","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_058EC8","MqttUser":"sonoff","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:08:34 MQT: hm/status/tasmota_058EC8/STATUS7 = {"StatusTIM":{"UTC":"2020-08-07T11:08:34","Local":"2020-08-07T13:08:34","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:32","Sunset":"21:18"}}
13:08:34 MQT: hm/status/tasmota_058EC8/STATUS10 = {"StatusSNS":{"Time":"2020-08-07T13:08:34"}}
13:08:34 MQT: hm/status/tasmota_058EC8/STATUS11 = {"StatusSTS":{"Time":"2020-08-07T13:08:34","Uptime":"0T00:04:18","UptimeSec":258,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"BodeNet_Keller","BSSId":"38:10:D5:14:E6:70","Channel":1,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
13:04:57 MQT: Attempting connection...
13:04:57 MQT: Connected
13:04:57 MQT: hm/status/tasmota_058EC8/LWT = Online (retained)
13:04:57 MQT: hm/setting/tasmota_058EC8/POWER =

arendst, 2020-08-07:
Yes. Look in template_esp32 and remove undefine at the beginning.

closed 2020-08-10 by ascillato2