arendst/Tasmota #9024: ZBBridge: can NOT connect from Home Assistant ZHA

jsiemek, 2020-08-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonof ZBBridge
- [ ] Tasmota binary firmware version number used: 8.4.0 tasmota-zbbridge.bin
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```21:42:45 SRC: WebConsole from 192.168.2.101
21:42:45 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
21:42:46 SRC: Backlog
21:42:46 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
21:42:46 RSL: stat/ZHABridge/RESULT = {"NAME":"Sonoff ZHABridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18}
21:42:46 SRC: Backlog
21:42:46 CMD: Group 0, Index 1, Command "MODULE", Data ""
21:42:46 RSL: stat/ZHABridge/RESULT = {"Module":{"0":"Sonoff ZHABridge"}}
21:42:46 SRC: Backlog
21:42:46 CMD: Group 0, Index 1, Command "GPIO", Data "255"
21:42:46 RSL: stat/ZHABridge/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"208":"TCP Tx"},"GPIO2":{"0":"None"},"GPIO3":{"209":"TCP Rx"},"GPIO4":{"59":"Led4i"},"GPIO5":{"58":"Led3i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}

  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
21:44:39 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
21:44:40 SRC: Backlog
21:44:40 CMD: Group 0, Index 1, Command "RULE", Data ""
21:44:40 RSL: stat/ZHABridge/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":37,"Free":474,"Rules":"on system#boot do TCPStart 8888 endon"}
21:44:40 SRC: Backlog
21:44:40 CMD: Group 0, Index 2, Command "RULE", Data ""
21:44:40 RSL: stat/ZHABridge/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
21:44:40 SRC: Backlog
21:44:40 CMD: Group 0, Index 3, Command "RULE", Data ""
21:44:40 RSL: stat/ZHABridge/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:46:28 RSL: stat/ZHABridge/STATUS = {"Status":{"Module":0,"DeviceName":"ZHABridge","FriendlyName":["ZHABridge"],"Topic":"ZHABridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:46:28 RSL: stat/ZHABridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:23:53","StartupUTC":"2020-08-03T20:22:35","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-08-03T18:01:47","SaveCount":26,"SaveAddress":"F4000"}}
21:46:28 RSL: stat/ZHABridge/STATUS2 = {"StatusFWR":{"Version":"8.4.0(zbbridge)","BuildDateTime":"2020-07-29T12:13:05","Boot":7,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"388/699"}}
21:46:28 RSL: stat/ZHABridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"192.168.0.15","LogPort":50514,"SSId":["SiemekG",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00008000","00006000"]}}
21:46:28 RSL: stat/ZHABridge/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":27,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F100786","04500001","00000002","00000000","00000000","00020000"],"Drivers":"1,2,4,7,9,10,20,21,23,41","Sensors":""}}
21:46:28 RSL: stat/ZHABridge/STATUS5 = {"StatusNET":{"Hostname":"ZHABridge-0103","IPAddress":"192.168.10.80","Gateway":"192.168.0.1","Subnetmask":"255.255.0.0","DNSServer":"192.168.0.15","Mac":"98:F4:AB:E2:60:67","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:46:28 RSL: stat/ZHABridge/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E26067","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:46:28 RSL: stat/ZHABridge/STATUS7 = {"StatusTIM":{"UTC":"2020-08-03T20:46:28","Local":"2020-08-03T21:46:28","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:27","Sunset":"20:25"}}
21:46:28 RSL: stat/ZHABridge/STATUS10 = {"StatusSNS":{"Time":"2020-08-03T21:46:28"}}
21:46:28 RSL: stat/ZHABridge/STATUS11 = {"StatusSTS":{"Time":"2020-08-03T21:46:28","Uptime":"0T00:23:53","UptimeSec":1433,"Vcc":3.458,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SiemekG","BSSId":"78:8A:20:54:15:5C","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

00:00:00 CFG: Loaded from flash at F8, Count 30
00:00:00 CFG: CR 388/699, Busy 0
00:00:00 SNS: Hardware Serial
00:00:00 SRC: Restart
00:00:00 Project tasmota ZHABridge Version 8.4.0(zbbridge)-2_7_2_1
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 SiemekG Channel 11 BSSId 78:8A:20:54:15:5C in mode 11N as ZHABridge-0103...
00:00:01 WIF: Connected
21:50:46 NTP: UTC 2020-08-03T20:50:45, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
21:50:46 HTP: Web server active on ZHABridge-0103 with IP address 192.168.10.80
21:50:47 RSL: tele/ZHABridge/INFO1 = {"Module":"Sonoff ZHABridge","Version":"8.4.0(zbbridge)","FallbackTopic":"cmnd/DVES_E26067_fb/","GroupTopic":"cmnd/tasmotas/"}
21:50:47 RSL: tele/ZHABridge/INFO2 = {"WebServerMode":"Admin","Hostname":"ZHABridge-0103","IPAddress":"192.168.10.80"}
21:50:47 RSL: tele/ZHABridge/INFO3 = {"RestartReason":"Software/System restart"}
21:50:47 RUL: SYSTEM#BOOT performs "TCPStart 8888"
21:50:47 SRC: Rule
21:50:47 CMD: Group 0, Index 1, Command "TCPSTART", Data "8888"
21:50:47 TCP: Starting TCP server on port 8888
21:50:47 RSL: stat/ZHABridge/RESULT = {"TCPStart":"Done"}
21:50:51 RSL: tele/ZHABridge/STATE = {"Time":"2020-08-03T21:50:51","Uptime":"0T00:00:09","UptimeSec":9,"Vcc":3.458,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SiemekG","BSSId":"78:8A:20:54:15:5C","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:50:52 APP: Boot Count 17
21:50:52 CFG: Saved to flash at F7, Count 31, Bytes 4096
21:51:44 HTP: Main Menu
21:51:59 TCP: to MCU/1: 1AC038BC7E
21:51:59 HTP: Main Menu
21:52:05 HTP: Main Menu
21:52:06 HTP: Main Menu
21:52:29 HTP: Console
21:52:39 TCP: to MCU/1: 1AC038BC7E

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Trying to connect to ZBBridge from Home Assistant ZHA via
socket://ipaddress:8888, speed 115200
I got:" Unable to connect to ZHA device."
in Tasmota console only: "TCP: to MCU/1: 1AC038BC7E"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Got connected.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-08-04:
The EFR32  does not respond to the reboot request. Please reach out of n Discord

closed 2020-08-04 by ascillato2

ryrzy, 2020-09-12:
hello, 
the problem is related to this problem so i don't open a new one (sorry if i'm wrong).

I will connect my Sonoff ZbBridge to HomeAssistant with ZHA integration (HA 0.114.4, Tasmota 8.5.0 (and tested on dev 8.5.0.1)).

When i try to enable port with command TCPStart 8888 and when i have selected module 75 (Sonoff ZbBridge) i have got error
```
07:05:22 CMD: tcpstart 8888
07:05:22 RSL: stat/tasmota_DC04A5/RESULT = {"Command":"Error"}
```
always when module is set to 75, tcpstart doesn't work.

jacekk015, 2020-09-15:
> hello,
> the problem is related to this problem so i don't open a new one (sorry if i'm wrong).
>
> I will connect my Sonoff ZbBridge to HomeAssistant with ZHA integration (HA 0.114.4, Tasmota 8.5.0 (and [...]

Paste this to Zbbridge console
`backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"Sonoff ZHABridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0`

ryrzy, 2020-09-16:
@jacekk015 i know that it works with selected generic module (18), but then is no devices list on tasmota web page ;) i think that tcpstart should work on module 75 (Sonoff ZbBridge) :)

jacekk015, 2020-09-16:
Reason could be because:
https://zigbee.blakadder.com/Sonoff_ZBBridge.html
**Support for this device is experimental!!!**
Otherwise I can confirm that with module 75 TCPStart gives error.

ryrzy, 2020-09-16:
i know... thats why i'm reporting this o.O why are you yelling at me?

jacekk015, 2020-09-16:
It was just the quote from the page.
Pozdrawiam

s-hadinger, 2020-09-16:
@ryrzy It's all explained in details in Digiblur's blog. If you use Module 75, then GPIO13/15 are configured for Zigbee2Tasmota, therefore unavailable for TCPStart.

TCPStart needs GPIOs configured to Serial TX/RX.

Quoting Digiblur's blog:
```
backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"Sonoff ZHABridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0
```

ryrzy, 2020-09-16:
Thank you for explanation  :)