arendst/Tasmota #9026: System Boot Timer Rule failes if can't connect to MQTT server

chrisphotonic, 2020-08-04:
The rule below works perfectly if I can connect to MQTT server, if the MQTT server is offline on device bootup then the rule never gets called.

console # PowerOnState 0
console # rule 1
console # rule on system#boot do ruletimer1 120 endon on Rules#Timer=1 do backlog power1 on; endon
console # rule1 1

Currently my power company has a lot of on/off power events that could damage my networking equipment.  I'm using this to power up my networking equipment if the power can stay on for two minutes. This little rule saves my equipment from the spikes and causes my home automation devices to startup in the correct order. I'd like to be able to use MQTT too, but I've disabled it for now to add this safeguard.

chrisphotonic, 2020-08-04:
I'm using Tasmota 8.4.0.

meingraham, 2020-08-04:
Unless your device can set its time, timers won't function, **_even_** just a duration timer (i.e., not dependent on time of day). So, it's not the lack of a connection to the MQTT broker that causes the rule not to work but rather the fact that your device cannot connect to the network (which you haven't turn on) in order to set its time. To make this work without Wi-Fi available, you'll have to add either an RTC module or a GPS module so your device can retrieve and set the time.

ascillato, 2020-08-04:
The `system#boot` trigger requires all what you have found in order to be executed. This is explained in the docs. 

But for your use case you can use instead the trigger `power1#boot` that will be executed right after your Tasmota is energized without checking if wifi or mqtt are available.

Also, you can use the command `ruletimer` to make that wait before power on. `Ruletimer` command does not need NTP, WIFI, or MQTT in order to work. 

So, all together for your use case:

```lua
Poweronstate 0 
Rule1 1
Rule1 on power1# boot do ruletimer1 120 endon on rules#timer=1 do power1 1 endon
```

ascillato2, 2020-08-04:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-04 by ascillato2

meingraham, 2020-08-05:
> `Ruletimer` command does not need NTP

I stand corrected and happy to be. I thought **_any_** timer function required the device time to be set.

chrisphotonic, 2020-08-05:
`Poweronstate 0 
Rule1 1
Rule1 on power1# boot do ruletimer1 120 endon on rules#timer=1 do power1 1 endon`

I wasn't able to get that one working but 

`Pulsetime1 220
Poweronstate 5
`
Worked without a problem.

Tasmota is amazing. You could probably create a system#boot rules patch if the MQTT server isn't available it triggers but I've got something that works now. 

Thanks again guys!

chrisphotonic, 2020-08-05:
Looks like it has the added effect of keeping on whatever its plugged into every 220 seconds. Unlike the rule which would be more specific. Turns out that's beneficial effect for me in this case.