arendst/Tasmota #9030: Inconsistant output on tele/tasmota/RESULT

chriswalker01, 2020-08-05:
### PROBLEM DESCRIPTION

I've put Tasmota on my kettle and am monitoring the state and temperature of it in Openhab.

The temperature is included in tele/tasmotaKettle/RESULT as DpType2Id5.

The issue I have is that the RESULT topic doesn't always include DpType2Id5 in its data.

Below are two lines that I received when I subscribed to tele/tasmota/RESULT

```
tele/tasmotaKettle/RESULT {"TuyaReceived":{"Data":"55AA0007001C010100010005020004000000206501000100660400010067040001008E","Cmnd":7,"CmndData":"01010001000502000400000020650100010066040001006704000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"},"DpType2Id5":32,"5":{"DpId":5,"DpIdType":2,"DpIdData":"00000020"},"DpType1Id101":0,"101":{"DpId":101,"DpIdType":1,"DpIdData":"00"},"DpType4Id102":0,"102":{"DpId":102,"DpIdType":4,"DpIdData":"00"},"DpType4Id103":0,"103":{"DpId":103,"DpIdType":4,"DpIdData":"00"}}}

tele/tasmotaKettle/RESULT {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
```

It all works ok, but Openhab logs an error when the shorter of the two lines is received because it doesn't include DpType2Id5.

Is there a way to split these into separate topics or can I make the shorter one also include DpType2Id5?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
 03:38:32 CMD: status 0
03:38:32 MQT: stat/tasmotaKettle/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmotaKettle","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:38:32 MQT: stat/tasmotaKettle/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T01:43:31","StartupUTC":"2020-08-05T00:55:01","Sleep":50,"CfgHolder":4617,"BootCount":1559,"BCResetTime":"2020-03-11T11:31:30","SaveCount":3124,"SaveAddress":"F7000"}}
03:38:32 MQT: stat/tasmotaKettle/STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(tasmota)","BuildDateTime":"2020-03-09T12:11:17","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"365/699"}}
03:38:32 MQT: stat/tasmotaKettle/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXX",""],"TelePeriod":100,"Resolution":"558180C0","SetOption":["0000A009","2805C8000100060000005A00000000000000","00010000","00000000"]}}
03:38:32 MQT: stat/tasmotaKettle/STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":428,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
03:38:32 MQT: stat/tasmotaKettle/STATUS5 = {"StatusNET":{"Hostname":"tasmotaKettle-2926","IPAddress":"192.168.10.7","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"84:F3:EB:EC:0B:6E","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
03:38:32 MQT: stat/tasmotaKettle/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.254","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EC0B6E","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
03:38:32 MQT: stat/tasmotaKettle/STATUS7 = {"StatusTIM":{"UTC":"2020-08-05T02:38:32","Local":"2020-08-05T03:38:32","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:29","Sunset":"20:22"}}
03:38:32 MQT: stat/tasmotaKettle/STATUS10 = {"StatusSNS":{"Time":"2020-08-05T03:38:32"}}
03:38:32 MQT: stat/tasmotaKettle/STATUS11 = {"StatusSTS":{"Time":"2020-08-05T03:38:32","Uptime":"0T01:43:31","UptimeSec":6211,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXX","BSSId":"C4:EA:1D:B9:0C:15","Channel":11,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:06"}}}
03:38:35 MQT: tele/tasmotaKettle/RESULT = {"TuyaReceived":{"Data":"55AA0007001C0101000100050200040000001E6501000100660400010067040001008C","Cmnd":7,"CmndData":"0101000100050200040000001E650100010066040001006704000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"},"DpType2Id5":30,"5":{"DpId":5,"DpIdType":2,"DpIdData":"0000001E"},"DpType1Id101":0,"101":{"DpId":101,"DpIdType":1,"DpIdData":"00"},"DpType4Id102":0,"102":{"DpId":102,"DpIdType":4,"DpIdData":"00"},"DpType4Id103":0,"103":{"DpId":103,"DpIdType":4,"DpIdData":"00"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

joba-1, 2020-08-05:
in general there are even more possible tele result messages. I think the only guarantee you get is that it contains valid json and you should make openhab ignore other messages.

schrej-zz, 2020-08-05:
I think, the problem arises, when you use the old MQTT1 Binding in OpenHab. Try to move to MQTT2 binding.

Jason2866, 2020-08-05:
It is not a problem of the MQTT1 Binding. Json parsing has to be done more specific. Ask for help/support in OH community.
I had this too and solved all with JSON handling in OH. It can be tricky

closed 2020-08-05 by ascillato2