arendst/Tasmota #9035: BME280 reports dew point null when humidity is 100%

MaceMoneta, 2020-08-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

BME280 reports dew point null when humidity is 100%

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer / OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
07:26:24 CMD: Backlog Template; Module; GPIO 255
07:26:24 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
07:26:24 RSL: RESULT = {"Module":{"18":"Generic"}}
07:26:24 RSL: RESULT = {"GPIO0":{"6":"I2C SDA"},"GPIO1":{"0":"None"},"GPIO2":{"5":"I2C SCL"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
07:27:39 CMD: Status 0
07:27:39 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"hvac","FriendlyName":["hvac"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:27:39 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T22:50:06","StartupUTC":"2020-08-04T12:37:33","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2020-04-17T23:11:11","SaveCount":44,"SaveAddress":"F9000"}}
07:27:39 RSL: STATUS2 = {"StatusFWR":{"Version":"8.4.0(sensors)","BuildDateTime":"2020-07-29T12:08:32","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"341/699"}}
07:27:39 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":3,"LogHost":"135.82.8.141","LogPort":514,"SSId":["moneta",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008101","2805C8000100060000005AFF000000000000","00000040","00006000"]}}
07:27:39 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":20,"ProgramFlashSize":1024,"FlashSize":8192,"FlashChipId":"1740EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCD","01DA9BC4","64367CC7","00084052"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
07:27:39 RSL: STATUS5 = {"StatusNET":{"Hostname":"hvac","IPAddress":"135.82.8.78","Gateway":"135.82.8.240","Subnetmask":"255.255.255.0","DNSServer":"135.82.8.1","Mac":"A0:20:A6:07:04:7D","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
07:27:39 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-08-05T11:27:39","Local":"2020-08-05T07:27:39","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":99,"Sunrise":"05:58","Sunset":"20:06"}}
07:27:39 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-08-05T07:27:39","BME280":{"Temperature":59.5,"Humidity":88.9,"DewPoint":56.2,"Pressure":1012.8},"PressureUnit":"hPa","TempUnit":"F"}}
07:27:39 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-08-05T07:27:39","Uptime":"0T22:50:06","UptimeSec":82206,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"moneta","BSSId":"78:8A:20:21:81:20","Channel":6,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Raise humidity to 100%

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I think the dew point should be equal to the temperature when the humidity is 100%

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MaceMoneta, 2020-08-05:
"A relative humidity of 100% indicates the dew point is equal to the current temperature and that the air is maximally saturated with water."
https://en.wikipedia.org/wiki/Dew_point

Staars, 2020-08-05:
This line will fail for a humidity of 100 (resulting in x=1):
 https://github.com/arendst/Tasmota/blob/934ca43463e33799ca023453a7b78757f09f7394/tasmota/support_float.ino#L119

TrDA-hab, 2020-08-05:
For me this means the sensor is broken and needs a replacement soon. After reset, the sensor works normally for a while. Maximum outdoor life 6-9 months. The sensor works indoors for 10-12 months.

This information applies to Chinese sensors with Aliexpress.

[image]

MaceMoneta, 2020-08-05:
The sensor isn't broken. The dew point is a calculated value, based on the temperature and relative humidity (both of which are returning correct values). It's a simple calculation error.

TrDA-hab, 2020-08-05:
**For information only!**
1. Humidity physically cannot be 100%, maximum 99.99%.
2. If your sensor shows 100%, then:
  - the dew point can be calculated correctly.
  - your sensor will be broken soon.
  - reset helps several times.
3. You need to buy a new sensor and make a replacement.

MaceMoneta, 2020-08-05:
Since the spec sheet indicates the range for relative humidity to be "0...100%", I disagree.

https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280-DS002.pdf

MaceMoneta, 2020-08-05:
Note the spec sheet sample code:

[image]

TrDA-hab, 2020-08-05:
It's fine.
Yesterday I replaced two of the six sensors again, and again I have no problem with 100% humidity. And I know 100% that in 6-9 months I will have to change the sensors again. This is my experience, sorry, but Tasmota has no error, this is the quality problem of the sensors.

joba-1, 2020-08-05:
@TrDA-hab have you even looked?
Or are you so blinded by prejudices about china that you can’t see it?
AFAICS @MaceMoneta has analysed it perfectly:

Humidity 100 leads to log(1) which leads to division by zero in the cited line instead of returning zero. Clearly wrong and completely independent from the sensor.

TrDA-hab, 2020-08-05:
@joba-1 
> @TrDA-hab have you even looked?
Not. I described my practice. Sorry if I disappointed you.
But you have to understand 100% humidity does not exist, it is absurd. Good luck.

joba-1, 2020-08-05:
> But you have to understand 100% humidity does not exist, it is absurd.

Me and wikipedia disagree on that. It is not absurd, relative humidity of 100% even has its own name: "dew point". The german (Bosch) sensor seems to be fine :)

closed 2020-08-06 by arendst

TrDA-hab, 2020-09-08:
I have been looking for a solution to the problem for a very long time and have tried many options. The problem with BME280 "freezing", more details here:
https://github.com/arendst/Tasmota/issues/9035
1. A very large I2C line (1.1m + 1.8m) == unstable power supply (starvation) simply disables the BME280.
I wrote how I solved it here:
https://discordapp.com/channels/479389167382691863/526796261714821151/728640555541856327
2. Very high current at the start of MH-Z19V ~ 0.15A. I added a 1000uF 10V aluminum capacitor in parallel to the power supply.
3. Very high current at start SDS011 ~ 0.2A. I added a 1000uF 10V aluminum capacitor in parallel to the power supply.
This is how the connection diagram looks now - everything works stably for 2 months.
[image]
[image]

MaceMoneta, 2020-09-08:
In the case of 100% humidity, the BME280 was not freezing, and everything is stable (for over a year). This bug report simply covers an error in the calculation.

TrDA-hab, 2020-09-08:
Perhaps you are right - this is my calculation error.
I described the steps that helped me solve the problem. And as I emphasized earlier, this is not Tasmota's fault.

PR #9039 (2020-08-06): Fix ln(x) via taylor series for special case x = 1
I think this trivial change fixes #9035 

@MaceMoneta, could you please test?

## Checklist:
  - [ x] The pull request is done against the latest dev branch
  - [ x] Only relevant files were touched
  - [x ] Only one feature/fix was added per PR.
  - [ ] The code change is tested and works on Tasmota core ESP8266 V.2.7.3.2
  - [ ] The code change is tested and works on core ESP32 V.1.12.2
  - [x ] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change m [... cut]