arendst/Tasmota #9040: ESP8266 Wemos D1 mini with Dimmer it restarts.

pabloeorsi, 2020-08-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am using an ESP8266 Wemos board for home automation. I have connected some relays and a Dimmer, with Triac and zero crossing. It seems to be working fine, but randomly when I modify the dimmer value it generates an error and restarts.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used: ESP8266 Wemos D1 mini
- [ ] Tasmota binary firmware version number used: Tasmota v8.4.0 George
  - [ ] Pre-compiled
- [ ] Flashing tools used: tasmotizer-1.1a
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

20:28:30 MQT: stat/RT_07001/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:28:31 MQT: stat/RT_07001/RESULT = {"Module":{"18":"Generic"}}
20:28:31 MQT: stat/RT_07001/RESULT = {"GPIO0":{"19":"Button3"},"GPIO1":{"29":"Relay1i"},"GPIO2":{"20":"Button4"},"GPIO3":{"30":"Relay2i"},"GPIO4":{"18":"Button2"},"GPIO5":{"31":"Relay3i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"37":"PWM1"},"GPIO13":{"17":"Button1"},"GPIO14":{"42":"Counter1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
20:29:29 MQT: stat/RT_07001/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":183,"Free":328,"Rules":"on system#boot do Backlog Timezone -3; Latitude -38.735109; Longitude -62.277783; TelePeriod 3600; CounterDebounceLow 0; SetOption99 1; SetOption15; LedTable 0; PwmFrequency 100 endon"}
20:29:30 MQT: stat/RT_07001/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":76,"Free":435,"Rules":"on system#boot do Backlog SetOption73 1; ButtonTopic 0; SetOption32 10 endon"}
20:29:30 MQT: stat/RT_07001/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":220,"Free":291,"Rules":"on button1#state=10 do power4 2 endon on button1#state=11 do dimmer 40 endon on button1#state=3 do dimmer 100 endon on button2#state do power1 2 endon on button3#state do power2 2 endon on button4#state do power3 2 endon"}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

20:30:00 MQT: stat/RT_07001/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"RT_07001","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"bt_07001","SwitchMode":[1,1,1,1,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:30:00 MQT: stat/RT_07001/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:02:43","StartupUTC":"2020-08-05T23:27:17","Sleep":50,"CfgHolder":4617,"BootCount":76,"BCResetTime":"2020-08-02T00:16:14","SaveCount":809,"SaveAddress":"FA000"}}
20:30:00 MQT: stat/RT_07001/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:05:28","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"346/699"}}
20:30:00 MQT: stat/RT_07001/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hassio",""],"TelePeriod":3600,"Resolution":"558180C0","SetOption":["00008009","0A05C8000100060000005A00000000000000","00800000","00026000"]}}
20:30:00 MQT: stat/RT_07001/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:30:00 MQT: stat/RT_07001/STATUS5 = {"StatusNET":{"Hostname":"RT_07001-1451","IPAddress":"192.168.1.4","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"4C:11:AE:0D:E5:AB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:30:00 MQT: stat/RT_07001/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0DE5AB","MqttUser":"pabloeorsi","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:30:00 MQT: stat/RT_07001/STATUS7 = {"StatusTIM":{"UTC":"2020-08-05T23:30:00","Local":"2020-08-05T20:30:00","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"-03:00","Sunrise":"08:05","Sunset":"18:22"}}
20:30:00 MQT: stat/RT_07001/STATUS10 = {"StatusSNS":{"Time":"2020-08-05T20:30:00","COUNTER":{"C1":302059}}}
20:30:00 MQT: stat/RT_07001/STATUS11 = {"StatusSTS":{"Time":"2020-08-05T20:30:00","Uptime":"0T00:02:43","UptimeSec":163,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Dimmer":40,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"hassio","BSSId":"EC:08:6B:B4:9E:0A","Channel":3,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

{"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["40203cc8","00000000","400043f0"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40100986","4024620a","4010068d","40101160","401005cc","4000050c","400043e6","400044ab","402024bf","40101d37","4024607d","4024609c","4000444e","40223f0f","40106b1b","4000410f","40004a3c","40106dac","40202534","40244662","40202584","4021a3c7","40230218","401015e8","402323ec","402305d3","40247728","401010c5"]}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
The error occurs when using it, usually when I modify the dimmer value. It is random.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Maybe I have some wrong settings.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I use Home Assistant with MQTT.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-08-06:
Hi,

"Exception":0 sometimes means not enough power for CPU.

Try a better power source for your Wemos D1 Mini

pabloeorsi, 2020-08-06:
> Hi,
> 
> "Exception":0 sometimes means not enough power for CPU.
> 
> Try a better power source for your Wemos D1 Mini

Hi ascillato,

Thanks, I will try a better font. I am going to try with an exclusive feeding for the ESP.
I will comment on the results.

Regards!

pabloeorsi, 2020-08-06:
Hi!

Put an exclusive font for ESP and check everything was fine, however the error still happens, and always happens when I modify the PWM value. It does not always happen, it is random, but sometimes the error is generated and everything is restarted.

I get the following:

00:00:00 CFG: Loaded from flash at F5, Count 39
00:00:00 Project tasmota Tasmota Version 8.4.0(tasmota)-2_7_2_1
00:00:00 WIF: Connecting to AP1 hassio Channel 3 BSSId EC:08:6B:B4:9E:0A in mode 11N as RT_07001-1451...
00:00:01 WIF: Connected
20:01:17 HTP: Web server active on RT_07001-1451 with IP address 192.168.1.4
20:01:18 MQT: Attempting connection...
20:01:18 MQT: Connected
20:01:18 MQT: tele/RT_07001/LWT = Online (retained)
20:01:18 MQT: cmnd/RT_07001/POWER = 
20:01:18 MQT: tele/RT_07001/INFO1 = {"Module":"Generic","Version":"8.4.0(tasmota)","FallbackTopic":"cmnd/DVES_0DE5AB_fb/","GroupTopic":"cmnd/tasmotas/"}
20:01:18 MQT: tele/RT_07001/INFO2 = {"WebServerMode":"Admin","Hostname":"RT_07001-1451","IPAddress":"192.168.1.4"}
20:01:18 MQT: tele/RT_07001/INFO3 = {"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["40203cc8","00000000","400043d5"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40100986","401015e8","4010068d","40247610","401005cc","4000050c","400043e4","400044ab","402024bf","401005cc","4000050c","40230db6","4000444e","40106b1b","4000410f","40004a3c","40106dac","40202534","40244662","40202584","4021a3c7","40230218","401015e8","402323ec","402305d3","40247728","401010c5"]}}

Thanks for the help of everything! I love this community and Tasmota!

stefanbode, 2020-08-11:
Hi, I see you're the one using my AC-Dimmer extension with zero crossing. I haven't seen that post but working on the same issue. As far as I can see I have found the issue. My device is now in endurance test. If I'm able to have it 48h stable I will commit the change. If you can't wait feel free to copy the source from https://github.com/stefanbode/Sonoff-Tasmota/blob/master/tasmota/xsns_01_counter.ino Just replace only this file.

ascillato2, 2020-08-11:
Closing in favour of #9041

closed 2020-08-11 by ascillato2