arendst/Tasmota #9041: Hardware Watchdog bites for AC-Dimmer after ~126.000 sec

stefanbode, 2020-08-06:
### PROBLEM DESCRIPTION
Without any additional info the device reboots through Hardware watchdog. How can the issue further logged to get more info where the issue has its root cause. I will try to make the fix.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _Wemos Mini
- [x ] Tasmota binary firmware version number used: _8.3.1.7____
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
08:43:13 MQT: hm/status/ESP_1EF0C0/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:43:13 MQT: hm/status/ESP_1EF0C0/RESULT = {"Module":{"0":"Generic"}}
08:43:13 MQT: hm/status/ESP_1EF0C0/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"42":"Counter1"},"GPIO13":{"37":"PWM1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"ESP_1EF0C0","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.178.35/firmware.bin","RestartReason":"Hardware Watchdog","Uptime":"0T04:29:16","StartupUTC":"2020-08-06T02:10:28","Sleep":50,"CfgHolder":4627,"BootCount":4236,"BCResetTime":"2020-07-24T08:53:24","SaveCount":12710,"SaveAddress":"1F5000"}}
STATUS2 = {"StatusFWR":{"Version":"8.3.1.7(tasmota)","BuildDateTime":"2020-07-24T09:13:04","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"391/699"}}
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":4,"LogHost":"192.168.178.96","LogPort":514,"SSId":["BodeNet_Keller","SchreiberWLAN"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000080E9","2805C8000100060000005A00000000000000","00000000","00026000"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":1436,"Heap":21,"ProgramFlashSize":2048,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9046C7","04368001","000084C5","050013C0","C000FB81","00004004"],"Drivers":"1,2,3,4,6,7,8,9,10,16,18,19,20,21,22,24,26,27,28,29,30,35,37","Sensors":"1,2,3,4,5,6,9,12,45"}}
STATUS5 = {"StatusNET":{"Hostname":"ESP_1EF0C0-4288","IPAddress":"192.168.178.127","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"60:01:94:1E:F0:C0","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
STATUS6 = {"StatusMQT":{"MqttHost":"elocm.synology.me","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1EF0C0","MqttUser":"ESP_1EF0C0","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
STATUS7 = {"StatusTIM":{"UTC":"2020-08-06T06:39:45","Local":"2020-08-06T08:39:45","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:31","Sunset":"21:20"}}
STATUS10 = {"StatusSNS":{"Time":"2020-08-06T08:39:45","COUNTER":{"C1":115410948}}}
STATUS11 = {"StatusSTS":{"Time":"2020-08-06T08:39:45","Uptime":"0T04:29:17","UptimeSec":16157,"Heap":21,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":15,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"BodeNet_Keller","BSSId":"7C:FF:4D:3D:A3:5B","Channel":5,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Let the dimmer run on 20% for more than one day.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
No Watchdog bite

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-08-06:
How can I log the Web Console Window to a file. THis might give some insight. SysLog is not an option because I get a reboot if I place a Log output in the interrupt handler with syslog. Only serial and surprise weblog is working.

arendst, 2020-08-06:
Watchdogs tend to report NOTHING simply because a timer expired which resets the device. It could report Software or Hardware watchdog which allows for some control of timeout timer.

I'm afraid it's one of those SDK issues related to wifi in combination with other busy items like PWM.

You might want to try the latest core version available from tasmota-stage or experiement with another SDK version also configurable for the tasmota-stage version.

arendst, 2020-08-06:
Just FYI this triggers a watchdog too:
```
  volatile uint32_t dummy = 0;
  while (1) {
    dummy++;
  }
```
because there is no yield() the watchdog timer is never reset.

stefanbode, 2020-08-06:
yes, I'm aware of the example. The dimmer also has no while clause that might cause the issue. Currently the only special is the call of the startWaveformClockCycles(Pin(GPIO_PWM1, index), high, low, 0, -1, 0, true); after the PWM was set OFF a few microseconds before. as mentioned it can work for quite a long time e.g. 120k of execution. I just have captured a new one after 10k. I do not see any "special" data during the crash. I will try the SDK update.

arendst, 2020-08-06:
But is it a software WDT or Hardware WDT?

stefanbode, 2020-08-06:
As mentioned. For sure Hardware Watchdog. Anyhow I have one idea test now.  17k seconds already passed

stefanbode, 2020-08-07:
another Hardware WDT at 11:30pm. data from the zero-cross detection is completely normal. Build with:
platform                  = espressif8266@2.5.1
platform_packages         = framework-arduinoespressif8266 @ https://github.com/tasmota/Arduino/releases/download/2.7.2.1/esp8266-2.7.2.1.zip

arendst, 2020-08-07:
From the core ``reference.rst`` documentation:

> Long-running (>1ms) tasks in interrupts will cause instabilty or crashes.
  WiFi and other portions of the core can become unstable if interrupts
  are blocked by a long-running interrupt.  If you have much to do, you can
  set a volatile global flag that your main ``loop()`` can check each pass
  or use a scheduled function (which will be called outside of the interrupt
  context when it is safe) to do long-running work.

So as the counter interrupt uses ``delayMicroseconds(dimm_time);`` I think the WDT is an accident bound to happen...

arendst, 2020-08-07:
For debugging purposes you might want to try ti add function ``WdtDisable()`` introduced with the latest commit. You might want to execute it when ``SO99 1`` is set like in ``CounterInit()``. It's definitly not a state you want to use in production but it might show a hang where it used to do a restart due to the hang.

stefanbode, 2020-08-07:
All good hints. First I will try to overcome the delaymicroseconds because this can be up to 10ms before hooking into the sinus curve.

stefanbode, 2020-08-07:
A hang looks like a good reason to try my lately purchased debug hardware for the esp8266.

stefanbode, 2020-08-10:
the hardware reset is 100% linked to the delaymicroseconds() in the interrupt. I need to get this removed out of the interrupt.

stefanbode, 2020-08-11:
Ok, I removed the PWM control from the interrupt and only do the calculation of the time drift in the interrupt. The device is in testing mode. If >48h survived I will create a pull request.

closed 2020-08-12 by stefanbode