arendst/Tasmota #9044: Tasmota + HM10 + MiFlora, wrong moisture and fertility

phoxy84, 2020-08-06:
### PROBLEM DESCRIPTION
I have a ESP flashed with Tasmota, works great, nice job!

I bought a HM10 module, flashed with v708 that works like a charm. I have 3 flora's en 2 Xiaomi temperature sensors connected with it. I can read the sensors in Home Assistant

The issue I have is that the reported moisture is incorrect, see screenshot, can this issue be solved?

The reported moisture in Home assistant is:
1078689792 %
1078460416 %
1078296576 %

When I read the data, it should be:
55%
48%
43%

The basic number is 1076887552 and every degree stands for 32768 (2^15).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

### REQUESTED INFORMATION
- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): ESP32 with HM10 (v708)
- [ x] Tasmota binary firmware version number used: 8.4.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
15:53:57 CMD: Backlog Template; Module; GPIO 255
15:53:58 MQT: stat/HM10/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:53:58 MQT: stat/HM10/RESULT = {"Module":{"0":"Generic"}}
15:53:58 MQT: stat/HM10/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"194":"HM10 RX"},"GPIO3":{"0":"None"},"GPIO4":{"195":"HM10 TX"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:54:50 CMD: Status 0
15:54:50 MQT: stat/HM10/STATUS = {"Status":{"Module":0,"DeviceName":"HM10","FriendlyName":["HM10"],"Topic":"HM10","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:54:50 MQT: stat/HM10/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software Watchdog","Uptime":"0T02:41:44","StartupUTC":"2020-08-06T11:13:06","Sleep":50,"CfgHolder":4617,"BootCount":2320,"BCResetTime":"2020-05-20T13:37:47","SaveCount":2395,"SaveAddress":"F8000"}}
15:54:50 MQT: stat/HM10/STATUS2 = {"StatusFWR":{"Version":"8.4.0(sensors)","BuildDateTime":"2020-07-29T12:08:32","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"359/699"}}
15:54:50 MQT: stat/HM10/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VFNL-C9D1E0_ub",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000"]}}
15:54:50 MQT: stat/HM10/STATUS4 = {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCD","01DA9BC4","64367CC7","00084052"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
15:54:50 MQT: stat/HM10/STATUS5 = {"StatusNET":{"Hostname":"HM10-6674","IPAddress":"10.0.0.55","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"EC:FA:BC:5E:DA:12","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:54:50 MQT: stat/HM10/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.15","MqttPort":1883,"MqttClientMask":"HM10","MqttClient":"HM10","MqttUser":"hassio","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:54:50 MQT: stat/HM10/STATUS7 = {"StatusTIM":{"UTC":"2020-08-06T13:54:50","Local":"2020-08-06T15:54:50","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:31","Sunset":"21:20"}}
15:54:50 MQT: stat/HM10/STATUS10 = {"StatusSNS":{"Time":"2020-08-06T15:54:50","Flora-61b3d5":{"Temperature":28.3,"Illuminance":1873,"Moisture":1077968896,"Fertility":1082372096,"Battery":30},"Flora-660899":{"Temperature":27.1,"Illuminance":725,"Moisture":1078689792,"Fertility":1082583040,"Battery":95},"Flora-6a48b3":{"Temperature":26.2,"Illuminance":68,"Moisture":1078525952,"Fertility":1082417152,"Battery":100},"MJ_HT_V1-d8a69d":{"Temperature":26.5,"Humidity":40.8,"DewPoint":12.1,"Battery":18},"MJ_HT_V1-d8b304":{"Temperature":26.9,"Humidity":38.7,"DewPoint":11.6,"Battery":91},"TempUnit":"C"}}
15:54:50 MQT: stat/HM10/STATUS11 = {"StatusSTS":{"Time":"2020-08-06T15:54:50","Uptime":"0T02:41:44","UptimeSec":9704,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":1,"SSId":"VFNL-C9D1E0_ub","BSSId":"80:2A:A8:94:69:1E","Channel":11,"RSSI":100,"Signal":-18,"LinkCount":1,"Downtime":"0T00:00:03"}}}
15:54:50 MQT: stat/HM10/STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["4024897a","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40228fb8","4023b152","402148a2","402368cd","4024d578","40101091"]}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Nothing

### EXPECTED BEHAVIOUR
Show the correct sensor output value

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2020-08-06:
Looks like a driver bug. I will take a look.

Staars, 2020-08-07:
https://github.com/arendst/Tasmota/pull/9049

There were remnants of old code with now incorrect format specifiers. In my test this works now.

Please check, if the driver works for as before, because this is a long planned feature upgrade too.

phoxy84, 2020-08-08:
How can I help to test?

closed 2020-08-10 by ascillato2