arendst/Tasmota #9047: Discrete DimmerLevel<x> command

erikleo, 2020-08-07:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I would like to be able to control the dimmer level without turning on the light power, and keep the default functionality of the Dimmer<x> command turning on a light

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I propose the creation of a discrete DimmerLevel<x> command where calling this command would set the dimLevel and leave the power state unchanged.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Without this function I would have to require users who want to implement scheduled presets to run 3 commands:

SetOption20 1
Dimmer<x>
Setoption20 0

I find this solution overly complicated and adds the burden of having to manage what state the SetOption20 is in.

**Additional context**  
_Add any other context or screenshots about the feature request here._
I find that users are accustomed to the Dimmer<x> command turning on the light and believe this is the proper functionality.  However, there is a strong use case for changing the dimmer level without changing the power state as many users like to automatically dim the level at night.  It also is nice to visually be able to see/query the current dim levels at night verify that the default state has changed (even though they may not be using the light)  Therefore I believe the DimmerLevel<x> would be a very useful addition to the Tasmota firmware.

**(Please, remember to close the issue when the problem has been addressed)**

erikleo, 2020-08-07:
An alternative solution could also be to add a second optional parameter to the Dimmer command that would be the SetOption20 value to be used only for that command. So the comnmand
cmnd=Dimmer 50 1
would set the light to 50% and leave the power as is regardless of the value of Option20 (and not change the Option20 value)

I think this solution is actually better because it would be easier/cleaner to extend the same functionality for other commands like setting the color.

s-hadinger, 2020-08-08:
There are many commands that would need to be changed too: `HsbColor`, `Channel`, `Color`...

For now you will need to live with `SetOption20 1` that you can group into a single command with `Backlog SetOption20 1; Dimmer <x>; Setoption20 0`

closed 2020-08-10 by ascillato2