arendst/Tasmota #9051: Zigbee Sonoff ZBBridge todo/wishlist

s-hadinger, 2020-08-08:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
This is a general Feature Request for short and mid-term features for Sonoff ZBBridge and Zigbee support.

**Describe the solution you'd like**  
List below

**Describe alternatives you've considered**  
See below

**Additional context**  
This is a follow-up for #8583 and #9009

**(Please, remember to close the issue when the problem has been addressed)**  

Here is a list of short-term and mid-term features for the Sonoff ZBBridge. I don't guarantee that they will be all implemented.

### Short term:
- Fix device `0x0000` showing up in the WebGui and list of devices
- Set Extended Timeout for better reachability of sleepy devices (polling)
- Make the Green LED brighter
- Improve user feedback on EFR32 flashing

### Mid Term:
- What should we do with the button on Sonoff ZBBridge?
- Use the I2C EEPROM to save device configuration, and make back-ups
- Auto-bind devices to the coordinator and usual clusters (zigbee2mqtt is already doing it)
- Auto-configure attribute reporting on usual attributes (zigbee2mqtt is already doing it)
- Include TLS/AWS IoT in the tasmota-zbbridge (because we have more than enough space for it)
- Automatically transform IAS (Intruder Alarm System) cluster to a more user-friendly type: ex: `Occupancy` instead of `ZoneStatus`
- Add option for some device to not be queried for state change (could create a network storm)

### Longer Term:
- Support for Green Power = battery-less devices that use the press on a button create enough energy to send a message (I'm not sure what EZSP is capable of doing)

### Features we will not support
Because they would waste a lot of flash space and are niche features.
- TouchLink
- Zigbee device OTA upgrade

MattWestb, 2020-08-09:
Here we go !!

First my comments for Short terms:

If fixing the device 0x0000 I can not longer using “ZbName 1 IKEA_BILLY_EZSP” for getting a nice list on the main menu. (better taking the Coordinator away as you have proposal).
For the LEDs and the EFR32 flash I think its possible doing more user friendly thing. More later under My Mid Terms.

My Short term suggestions:
It looks the pairing of Ikea bulbs is not complete paired (still slowly dimming until time out)  and EZSP have al done and looks OK.
Is looks like some parameters missed that is needed for the bulb going from pairing mode to normal paired mode. Perhaps only cosmetic.

Battery status for Ikea devices, Both new and old motion sensor and the On/Off Switch don’t getting any battery status and very likely for the other devices in the “FAMELY” (“E1766” open/close remote, “E1524/E1810” remote control,  “ICTC-G-1“ wireless dimmer and “E1744” SYMFONISK sound controller).

Saving “ZB Listen” for new created groups that being executed after the NCP is starting so not need running commands for getting messages from groups after restart.

Comments on Mid Term:
Using the EEPROM as backup of configuration its great  but do it as no braking thing if not being presents like on Elelabs Zigbee adapters and so on.

Auto binding to default group and also for “remotes” to the default group then the device don’t have doing it self (the old Ikea MS is making one grope and the new don’t) and if there do then configuring “ZB Listen” for the group (or sniffing groups and auto configuring “ZB Listen”). 
Attributes reporting and / or pulling status after restart of the NCP so getting a view of the system status and also then the system is running (periodical?).

The “Occupancy” thing. Don’t mix the “motion sensors” that sending light command “on with time off” an the real “Occupancy sensors” that sending “ True / false”. It have being more or less a ware in the deCONZ git then light steering is used and alarm and opposite and is miss implanted in the system (3 minutes auto reset of status then the (LL) sensor sending “on with time off 60 seconds) and the devs is only contra productive.

IAS I have not looking at but shod being my aria from my professional experience, but for the moment I living it for other to explore.

My Mid Term suggestions:
I have seeing that the Tasmota detecting that the NCP is coming on line (normally VCC was disconnected after ESP flashing and configuring for not interfering with the serial flashing) but only reporting it and do not trying to stating it.  I think its also pretty easy detecting the bootloader status and if the bootloader is in downloading mode then the NCP is of line so way not using that ?
Then the EZSP its not started its good showing that status on the status LED. And also trying start / restart the NCP if not stopped by user wold being god (perhaps watchdog?) 

LED / bottom / EZSP status.
The LED can having different pattern for different modes like EZSP in bootloader mode (Slowly long on short off), Flash download in progress (fast blinking), NCP fatal error / offline (SOS =  - - - -- -- -- - - -), pairing is enabled and so on.
The bottom can also have mullite purposes like 2 seconds = paring mode for 2 minutes, 1 fast = ending pairing mode (if is in pairing mode), 5 seconds toggle EZSP bootloader mode and NCP mode,  3 fast = sending 1 to bootloader then being in bootloader mode (Starting XModem server for download) and so on.

Its possible making very nice and good combinations that the user can using if making it good.
Look wat functions and feedback the user normally needs and making priority from that.

My Longer Term suggestions:
Try holding the system clean and not making one mess of it.
Don’t close doors then not necessarily from the tactical point.
The first priority is to working on the Sonoff bridge and then its possible leaving the door open for other hardware like Elelabs and other module by making no braking things for them then its possible.
For doing that you need thinking more global and trying keeping perspective and then diving in the details with the global perspective in mind. 

One very good example of doing things wrong in the past is deCONZ. 
They was starting as very good light steering system (LL) and then trying converting it to one universal ZB3 HA system and have messing things up.
One new bad example: deCONZ was more or less implanting the new “Opple switches” as HA and was making it more or less unusable for lights (Its ZB3 and can do both as Z2M have done). That is for my braking thing then its rear finding dissent light switches that also is working offline, and its gets many “HA” switches on the market that working well).
IAS and GP sounds great in the long run and opening many doors for the future!!
For IAS and Lights (x LL) profiles in ZB3 is autonomy A and O so try not braking that.
Many users have not experience coming home in late night in the winter then its dark outside and the light is not working at all because of corrupted database (deCONZ / HA) or SD-card is broken. I have getting both but have building all autonomy for not being left in the dark.

I have 3 large things but they need more time for writing so they is coming later and you is not liking 2 of them.

Sorry for not having time for doing more nice text formatting.

MattWestb, 2020-08-10:
Then building larger systems it can being good implanting node identifying funktion command in the NCP (node cluster 0x0003 = Identify) so its being easier finding the right bulb in the mesh.
My suggested priority is between short and medium.

duczz, 2020-08-11:
SONOFF SNZB-02 - ZigBee temperature and humidity sensor not integrated?

[image]

`17:51:42 CMD: ZbPermitJoin 1
17:51:42 SRC: WebConsole from 192.168.178.21
17:51:42 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "1"
17:51:42 ZIG: ZbEZSPSend 22003C
17:51:42 ZIG: ZbEZSPSend 3600FCFF00003600000000000000021E0103023C01
17:51:42 MQT: stat/tasmota_05DF17/RESULT = {"ZbPermitJoin":"Done"}
17:51:42 ZIG: {"ZbEZSPReceived":"220000"}
17:51:42 MQT: tele/tasmota_05DF17/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
17:51:42 ZIG: NAK, resending packet 0
17:51:42 ZIG: {"ZbEZSPReceived":"3600008F"}
17:51:42 ZIG: {"ZbEZSPReceived":"450005000036000000000000008FFF000000FFFF03023C01"}
17:51:42 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)
17:51:43 ZIG: {"ZbEZSPReceived":"3F0006FCFF000036000000000000008F010000"}
17:51:44 ZIG: {"ZbEZSPReceived":"2300000111745195D51C004B120004"}
17:51:44 ZIG: {"ZbEZSPReceived":"240011745195D51C004B120001000000"}
17:51:45 MQT: tele/tasmota_05DF17/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x00124B001CD59551","ShortAddr":"0x7411","ParentNetwork":"0x0000","Status":1,"Decision":0}}
17:51:45 ZIG: {"ZbEZSPReceived":"450004000013000000000000000084BD1174FFFF0C0011745195D51C004B12008002"}
17:51:45 MQT: tele/tasmota_05DF17/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x00124B001CD59551","ShortAddr":"0x7411","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
17:51:45 ZIG: ZbEZSPSend 340000117400000500000040010000030103031174
17:51:45 ZIG: {"ZbEZSPReceived":"34000092"}
17:51:46 ZIG: {"ZbEZSPReceived":"3F0006FDFF0000130000000000000000000000"}
17:51:46 ZIG: {"ZbEZSPReceived":"3F000011740000050000004001000092010000"}
17:51:46 ZIG: {"ZbEZSPReceived":"450000000005800000000100000184BD1174FFFF0603001174010102"}
17:51:46 MQT: tele/tasmota_05DF17/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
17:51:46 ZIG: ZbEZSPSend 34000011740401000001014001000001010700010004000500
17:51:46 ZIG: {"ZbEZSPReceived":"34000093"}
17:51:47 ZIG: {"ZbEZSPReceived":"3F000011740401000001014001000093010000"}
17:51:47 ZIG: {"ZbEZSPReceived":"450000040100000101000100000288BE1174FFFF1818010104000042076557654C696E6B05000042045448303102"}
17:51:47 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0x7411","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":55,"securityuse":0,"seqnumber":2,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"04000042076557654C696E6B050000420454483031"}}
17:51:47 ZIG: ZbZCLRawReceived: {"0x7411":{"0000/0004":"eWeLink","0000/0005":"TH01"}}
17:51:47 MQT: tele/tasmota_05DF17/SENSOR = {"ZbReceived":{"0x7411":{"Device":"0x7411","Manufacturer":"eWeLink","ModelId":"TH01","Endpoint":1,"LinkQuality":55}}}
17:51:49 ZIG: ZbFlashStore 012011745195D51C004B120001010000FFFF54483031006557654C696E6B0000FF
17:51:49 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 33 bytes)`

s-hadinger, 2020-08-11:
I don't have a `SONOFF SNZB-02`, I guess you need to both Bind and ask for Attribute Reporting.

Try:
```
ZbBind {"Devlce":"0x7411","Cluster":"Temperature"}
ZbBind {"Devlce":"0x7411","Cluster":"Humidity"}
```

Then:
```
ZbSend {"Devlce":"0x7411","Config":{"Temperature":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}
ZbSend {"Devlce":"0x7411","Config":{"Humidity":{"MinInterval":60,"MaxInterval":600,"ReportableChange":5}}}
```

duczz, 2020-08-11:
Tahnks for your fast reply, following error on the first command: ZbBind":"Unknown source device"

s-hadinger, 2020-08-11:
Sorry for the typo. It's `"Device"`

duczz, 2020-08-11:
`18:34:58 CMD: .ZbBind {"Device":"0x7411","Cluster":"Temperature"}
18:34:58 SRC: WebConsole from 192.168.178.21
18:35:02 CMD: ZbBind {"Device":"0x7411","Cluster":"Humidity"}
18:35:02 SRC: WebConsole from 192.168.178.21
18:35:02 CMD: Group 0, Index 1, Command "ZBBIND", Data "{"Device":"0x7411","Cluster":"Humidity"}"
18:35:02 ZIG: ZbEZSPSend 3400001174000021000000400100000D01160D5195D51C004B120001050403AFA342FEFF23A46001
18:35:02 MQT: stat/tasmota_05DF17/RESULT = {"ZbBind":"Done"}
18:35:02 ZIG: {"ZbEZSPReceived":"340000A2"}
18:35:07 CMD: ZbSend {"Device":"0x7411","Config":{"Temperature":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}
18:35:07 SRC: WebConsole from 192.168.178.21
18:35:07 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x7411","Config":{"Temperature":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}"
18:35:07 ZIG: guessing endpoint 1
18:35:07 ZIG: ZbEZSPSend 34000011740401020401014001000005010D100506000000293C0058026400
18:35:07 MQT: stat/tasmota_05DF17/RESULT = {"ZbSend":"Done"}
18:35:07 ZIG: {"ZbEZSPReceived":"340000A3"}
18:35:10 ZIG: {"ZbEZSPReceived":"8000421174"}
18:35:10 MQT: tele/tasmota_05DF17/RESULT = {"ZbRouteError":{"ShortAddr":"0x7411","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
18:35:12 CMD: ZbSend {"Device":"0x7411","Config":{"Humidity":{"MinInterval":60,"MaxInterval":600,"ReportableChange":5}}}
18:35:12 SRC: WebConsole from 192.168.178.21
18:35:12 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x7411","Config":{"Humidity":{"MinInterval":60,"MaxInterval":600,"ReportableChange":5}}}"
18:35:12 ZIG: guessing endpoint 1
18:35:12 ZIG: ZbEZSPSend 34000011740401050401014001000006010D100606000000213C005802F401
18:35:12 MQT: stat/tasmota_05DF17/RESULT = {"ZbSend":"Done"}
18:35:12 ZIG: {"ZbEZSPReceived":"340000A4"}
18:35:15 ZIG: {"ZbEZSPReceived":"8000421174"}
18:35:15 MQT: tele/tasmota_05DF17/RESULT = {"ZbRouteError":{"ShortAddr":"0x7411","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}`

MattWestb, 2020-08-11:
I dont being good in reading logs but the last ``` 18:35:15 MQT: tele/tasmota_05DF17/RESULT = {"ZbRouteError {"ShortAddr":"0x7411","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}} ```
is normally then its end device that is sleeping (like my Ikea motion sensors). 
Try waking the device up before you is sending the command so its not is in sleep mode and can receive it.

duczz, 2020-08-11:
That's it, Thanks, no Errors now.
But no Informations 
[image]

MattWestb, 2020-08-11:
You getting the "extended" infos in the console or if you have configured MQTT.
Like this for Aqara weather: ``` 18:00:48 RSL: SENSOR = {"ZbReceived":{"0xD991":{"Device":"0xD991","Name":"LUMI_THP","Temperature":27.84,"Humidity":59.47,"PressureUnit":"hPa","Pressure":996,"PressureScale":-1,"PressureScaledValue":9960,"Endpoint":1,"LinkQuality":150}}} ```

You can naming the device with ``` ZbName  <Device>, Name```

duczz, 2020-08-11:
i have nothing like this in the console and on MQTT are only the Bridge states

MattWestb, 2020-08-11:
Then its best waiting until s-hadinger coming back after his dinnering.

s-hadinger, 2020-08-11:
I need to look at the logs for the 4 commands to make sure they were taken into account. Use `weblog 3` and paste the logs. Press the button to wake up the device before each command. 
Thx

duczz, 2020-08-11:
`20:27:46 CMD: weblog 3
20:27:46 SRC: WebConsole from 192.168.178.21
20:27:46 CMD: Group 0, Index 1, Command "WEBLOG", Data "3"
20:27:46 MQT: stat/tasmota_05DF17/RESULT = {"WebLog":3}
20:27:52 ZIG: {"ZbEZSPReceived":"3F0000117404010504010140000000B0016600"}
20:27:58 CMD: ZbBind {"Device":"0x7411","Cluster":"Temperature"}
20:27:58 SRC: WebConsole from 192.168.178.21
20:27:58 CMD: Group 0, Index 1, Command "ZBBIND", Data "{"Device":"0x7411","Cluster":"Temperature"}"
20:27:58 ZIG: ZbEZSPSend 340000117400002100000040010000140116145195D51C004B120001020403AFA342FEFF23A46001
20:27:58 MQT: stat/tasmota_05DF17/RESULT = {"ZbBind":"Done"}
20:27:58 ZIG: {"ZbEZSPReceived":"340000B1"}
20:28:03 CMD: ZbBind {"Device":"0x7411","Cluster":"Humidity"}
20:28:03 SRC: WebConsole from 192.168.178.21
20:28:03 CMD: Group 0, Index 1, Command "ZBBIND", Data "{"Device":"0x7411","Cluster":"Humidity"}"
20:28:03 ZIG: ZbEZSPSend 340000117400002100000040010000150116155195D51C004B120001050403AFA342FEFF23A46001
20:28:03 MQT: stat/tasmota_05DF17/RESULT = {"ZbBind":"Done"}
20:28:03 ZIG: {"ZbEZSPReceived":"340000B2"}
20:28:08 CMD: ZbSend {"Device":"0x7411","Config":{"Temperature":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}
20:28:08 SRC: WebConsole from 192.168.178.21
20:28:08 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x7411","Config":{"Temperature":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}"
20:28:08 ZIG: guessing endpoint 1
20:28:08 ZIG: ZbEZSPSend 3400001174040102040101400100000D010D100D06000000293C0058026400
20:28:08 MQT: stat/tasmota_05DF17/RESULT = {"ZbSend":"Done"}
20:28:08 ZIG: {"ZbEZSPReceived":"340000B3"}
20:28:13 CMD: ZbSend {"Device":"0x7411","Config":{"Humidity":{"MinInterval":60,"MaxInterval":600,"ReportableChange":5}}}
20:28:13 SRC: WebConsole from 192.168.178.21
20:28:13 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x7411","Config":{"Humidity":{"MinInterval":60,"MaxInterval":600,"ReportableChange":5}}}"
20:28:13 ZIG: guessing endpoint 1
20:28:13 ZIG: ZbEZSPSend 3400001174040105040101400100000E010D100E06000000213C005802F401
20:28:13 MQT: stat/tasmota_05DF17/RESULT = {"ZbSend":"Done"}
20:28:13 ZIG: {"ZbEZSPReceived":"340000B4"}`

s-hadinger, 2020-08-11:
Your device didn't receive the commands. You should have received acknowledgement for the commands. I suppose it was sleeping

duczz, 2020-08-11:
Dont Know, i pressed the button before each command

techman83, 2020-08-12:
Thanks for the instructions @s-hadinger, took a long press (until the LED flashed) for the commands to be sent to my already paired SNZB-02. Guessing it goes into a config mode of sorts after initial pairing.

s-hadinger, 2020-08-13:
@MattWestb I actually added the `Identify` cluster because it was very easy :)  #9080

MattWestb, 2020-08-13:
I was thinking that and i'm not disappointed :-)))
Thanks very much !!
Now our firmware maker @grobasoz can installing his 30 IKEA lights (upside down as normal for the Aussie from our point of view) and identifying them easy. 

I cant testing for the moment then I is looking on the bellows ZB3 connecting problems.

Keep coding  hard ;-)

duczz, 2020-08-13:
Is this for Batterie not possible? He dont knows "battery"

ZbBind {"Device":"0x7411","Cluster":"battery"}

Then:

ZbSend {"Device":"0x7411","Config":{"battery":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}

s-hadinger, 2020-08-13:
It's `BatteryVoltage`

duczz, 2020-08-13:
Thanks works, but Battery will not send via MQTT

s-hadinger, 2020-08-13:
You can also try `BatteryPercentage`

duczz, 2020-08-13:
On Tasmota it works, but he dont send it to my MQTT Broker

[image]
[image]

tric111, 2020-08-15:
Currently, when using the Philips outdoor motion sensor SML002 in the same way as described in https://github.com/arendst/Tasmota/issues/8192, the internal red LED flashes when motion is detected. Apparently this is a "fault code" that occurs because the sensor does not receive a response to the occupancy message: https://github.com/Koenkk/zigbee2mqtt/issues/897

duczz, 2020-08-18:
any news for my issue?

s-hadinger, 2020-08-18:
@duczz You stated earlier that it worked on Tasmota. There's not much more I can do.

Btw this Github issue is not meant to solve individual issues, please connect to Discord or open a specific Github issue.

MattWestb, 2020-08-18:
For your 2 points in the to do list:
+ Automatically transform IAS (Intruder Alarm System) cluster to a more user-friendly type: ex: Occupancy instead of ZoneStatus
+ Customize the timeout for Occupancy sensor #8202

The newes document i have finding on the team "Occupancy".

[ZigBee Lighting & Occupancy
Device Specification
Version 1.0
ZigBee Document 15-0014-05
February 24th, 2016](https://zigbeealliance.org/wp-content/uploads/2019/11/docs-15-0014-05-0plo-Lighting-OccupancyDevice-Specification-V1.0.pdf)

I don't knowing if its older the the latest ZCL that you have getting, if then its the newer ZCL that is the master and if not this is the converting document for Light Link to Zigbee PRO = ZB3.

The #8202 is it one light controlling device, HA sensor or one IAS device ?
Keep the device types apart and don't messing it up mixing the functions between the types and making all confusing for the users (typical users is complaining about Ikeas motion sensor is not reporting light level and occupancy false then its not one light/occupancy sensor its one "On/off sensor" for lights).

I don't have reading the IAS but the name "Occupancy " is reserved for lights components in Zigbee and "Zone" or "Area" is the real terminology then working with intruder alarm systems.

The interesting thing is the definition of "26 On/off sensor" In that category is both Ikeas Motion sensors and also is you have one light sensor that is turning the light on then its dark.

The "13 Occupancy sensor" is (normally) not direkt controlling lights and not sending any commands only possible reporting attribute "Occupancy" true / false.

For my the most interesting is the "25 Control bridge" that can being used as one universal transmitter of commands to lights in one Zigbee networks as one router.

Also attribute reporting of status (on/off Light level and so on) is mandatory for all types of lights and plugs but not for controlling devices. 
So its make sense that the Ikeas ZB3 lights and plugs devices is reporting status and not the controllers. And then the coordinator is listen on the group commands its being sended to the groups from the controllers its getting all status from the devices in the zigbee mesh.
So in ZB3 lights is reporting there status and not being pulled for not flooding the zigbee mesh.
And the controlling devices do not reporting attribute or being pulled for there status (only "operating" status like battery level is normally reported).
I wonder how Philips have getting there ZB3 certificate for there bulbs (The HUE bridge is still LL certified as Ikea) .
I have not reading the HA and IAS part of the ZB3 / ZCL so i don't knowing if its reporting attributes or pulling for getting status of HA and IAS devices in ZB3.

Hedda, 2020-08-19:
An ESP32 single and dual-core version of tasmota-zbbridge firmware would be on my own long-term wish-list, like example: 
* @grobasoz wrote in https://github.com/zigpy/bellows/issues/243 that he's working on an ESP32 + EFR32 based Zigbee bridge
* @mtx512 also wrote in https://github.com/zigpy/bellows/issues/243 that he is as well working on an ESP32 + EFR32 bridge
* DIY Zigbee gateway/bridge based on ex ESP32-DevKitC https://www.espressif.com/en/products/devkits/esp32-devkitc/
* Wyze/Lockin lock gateway has an ESP32 with 8MB flash and EFR32MG21A020F1024 https://lockinsmarthome.com/pages/lockE

ESP32-DevKitC would probably make a good reference board https://www.espressif.com/en/products/devkits/esp32-devkitc/ or?

Zeroconf + mDNS as requested in https://github.com/arendst/Tasmota/issues/9081 (also on my wish-list) are for example best suited for ESP32 as more resource demanding?

Hedda, 2020-08-19:
> * What should we do with the button on Sonoff ZBBridge?

I would like to suggest using the button on the Sonoff ZBBridge to put the bridge into pairing-mode for pairing Zigbee devices.

Scenario; Another reason why Sonoff ZBBridge being a USB-powered WiFi is great is because you can plug it into a USB battery/power-pack which makes it portable so that you can bring it to different rooms in your house when pairing permanently installed mains-powered Zigbee devices like example in-wall mounted switches.

Very concept convenient easy when migrating from another platform and you have lots of devices installed around your house.

This could also make it possible to initially pair new devices without a computer, though you will still need to configure them later.

MattWestb, 2020-08-19:
@Hedda I agree 100% on the bottom but i think feedback on the LED then pairing mode is enabled and ended world being good. I think its possible putting more (but not too much) functions for the user in the bottom perhaps EZSP in flash / NCP mode toggle and start xmodem download.

Hedda, 2020-08-19:
@s-hadinger Can you pick when the button on Sonoff ZBBridge is pressed down to count seconds until it released?

If so then the button could have many functions and if combined with the LED for feedback as suggested it would make it easier for the users to know if they have successfully entered the command for the function and Sonoff ZBBridge is doing what you want.

Example suggestions if possible to detect how long the button is hold down for:

- Hold the button for 3-seconds to make it enter Zigbee pairing mode for 15-seconds= Fast pulse LED rapidly for 15-seconds.
  - Then have LED flash/blink slowly 3-times to indicate a successful paring of a Zigbee device
- Hold the button for 10-seconds to make EFR32 enter xmodem for 60-seconds= Slow pulse the LED rapidly for 60-seconds.
  - Then have LED flash/blink slowly 5-times to indicate a successful flashing of the EFR32 application firmware

Otherwise, you could do something like: 

- Press the button 3-times to make it enter Zigbee pairing mode for 10-seconds= Fast pulse LED rapidly for 15-seconds.
  - Then have LED flash/blink slowly 3-times to indicate a successful paring of a Zigbee device
- Press the button 5 times to make EFR32 enter xmodem for 60-seconds= Slow pulse the LED rapidly for 60-seconds.
  - Then have LED flash/blink slowly 5-times to indicate a successful flashing of the EFR32 application firmware

s-hadinger, 2020-08-19:
@Hedda This will need to be done via Rules. Keep in mind that the button needs a thin tool to press it. It should be used as emergency button, not as a pairing button.

s-hadinger, 2020-08-19:
@MattWestb Thanks for the document. It is more aimed at lights manufacturer but gave some good information. I am adding some of the extended ZCL attributes. However IAS is not part of this document, and some PIR don't follow the "Occupancy" profile. I will need to find something to emulate it.

kugelkopf123, 2020-08-19:
Maybe an integration into the scripting function would also be a good option that offers more possibilities than rules.

MattWestb, 2020-08-19:
@s-hadinger Yes i think it is one supplemental to the ZCL for ZB3. The ZCL "original" is based of the HA and suld being in the "normal" ZCL.
The Light Link was not implemented in the original ZCL then it was to much things interfere with the HA profile and its still problem with cluster compatibility between old LL and new ZB3 LO (silabs EZSP is still using cluster numbers that is not compatible with ZCL).
I don't knowing if IAS is part of the original ZCL or if Zigbee alliance have making one separate paper for the IAS.
And in the end your problem with #8202 is that its not one real ZB3 (Zigbee PRO variante with HA i think its more true) as normal with the not new Xaomi devices.
I think its best is to making one separate solution for this device and not destroying your clean implantation of the ZCL.

How is it going on your hardware front ? All equipment well burned and the supplier is not sending you new ones and you dont having trust in soldering iron ?

I'm waiting for you coming back on track and until i doing some deeeeeep diving in the (re)pairing problems of TB3 devices in the new belloes is having (first time pairing working well (no tokens is saved in the NCP for devices) but the rejoining / repairing one known device the TC-Link key is not updated in the NCP token = not actirusated devises).

Hedda, 2020-08-19:
> * Include TLS/AWS IoT in the tasmota-zbbridge (because we have more than enough space for it)

Maybe add support "Google Smart Home platform" as an alternative or as an option to use nstead of AWS IoT to the wish-list?

Google Smart Home platform lets users control their connected devices through the Google Home app and Google Assistant:

https://developers.google.com/assistant/smarthome/overview

Google's "Smart Home" platform is based on "Actions on Google":

https://developers.google.com/assistant/smarthome/develop/create

I think that very much of interest then would probably be Google's new "Local Home SDK" for local control, meaning fast response:

https://developers.google.com/assistant/smarthome/concepts/local

https://codelabs.developers.google.com/codelabs/smarthome-local/#0

https://developers.google.com/assistant/smarthome/reference/local

https://github.com/actions-on-google/smart-home-local

The Local Home SDK concept allows users to use smart actions fulfilment over local-WiFi/LAN, without a roundtrip to the cloud.

roger-, 2020-08-19:
> Set Extended Timeout for better reachability of sleepy devices (polling)

I've noticed that my Aqara (Xiaomi) sensors will pair and work correctly for a few minutes and then no longer update (unless I press the physical button). Is this a timeout issue? I've heard that Xiaomi devices are non-standard.

MattWestb, 2020-08-19:
@roger- I  have the Aqara door & window contact sensor (MCCGQ11LM) and Aqara temperature, humidity and pressure sensor (WSDCGQ11LM). Its can taking little before they start reporting battery status but its always doing that. The MCCGQ11LM is not sending much then i not detecting changing status of the magneetcontact. The WSDCGQ11LM is reporting the attribute in the log more times in the hour (have not clocking the intervall) and i have seen both device status in the the MQTT broker. 
I do testing for zigpy / belloes for the moment so don't having the Tasmota Zigbee Bridge setup active and cant providing data from logs at the moment.
Witch Aqara sensors do you have problem with ? Model and version is helping getting more information how they is working and can answer your question.

MattWestb, 2020-08-19:
Then the "hidden" bottom is not good for everyday use its better using it for halting the EZSP starting the telnet server (if its not started) and rebooting the EZSP in bootloader mode and giving one ``` 1 ``` then the consolle then its sending one ``` BL > ```  so keeping the bootloader in xmodem upload mode until rebooting the box or long pressing the button > reboot EZSP.
If its easy using the LED for for status of the process status.
One question wat GPIO is the button connected ? Only wondering if it's interfering with the booting of the ESP.

s-hadinger, 2020-08-19:
@MattWestb The button is connected to GPIO16. Currently in Tasmota it is declared as `Button1` and triggers Power On/off which has zero effect. You can easily set a Rule to trigger whatever you want from the button.

My ZBBridge is back online, and I have 2 more that just arrived in the mailbox. Back on track!

MattWestb, 2020-08-19:
Thanks for answer.
I was thinking you was needing going to Ik....   ;-) 
Take little care then J-tagging so not writing bad things at wrong location then tasting.
The good thing is that you now can testing the original firmware updating before you "customizing" it.
I was looking for IAS specifications and its looks its in integrated in the ZCL v7 (or v6).
No more Ruhetag until ZB3 is finished and then the R23 (Zigbee 4.0 ?) is in the pipeline ;-)))

roger-, 2020-08-19:
@MattWestb 

I also have a WSDCGQ11LM (connected via tasmota-zb to ZHA/home assistant). 

The problem isn't just infrequent updates but my devices become "unknown" after rebooting home assistant (even though the "last seen" field gets updated).

MattWestb, 2020-08-20:
@roger- Good that you is providing little more information about your setup.
Then using AHs ZHA for EZSP you is using the tasmota Zige bridge as serial over TCP.
So your problem is in HA / ZHA and you have doing right opening one issue in there git.
I think you is running the main branch of ZHA and its have getting support for the version 8 protocol framing but have not all futures implanted. In the ground is working in ZHA but they having problem with Zigbee 3 devices then they rejoining /  repairing. Then (nearly) all Xaomi devices is not ZB3 they are working there. The ZHA supports for newer versions of EZSP is in alpha stage and need time to getting stabili.

unscel, 2020-08-20:
Hi, hope this is correct place to drop the message. I use mqttthing with zigbee2tasmota and try to link xiaomi contact sensor (MCCGQ01LM) via sonoff zigbee bridge. So far it is working fine except an unexpected mqtt message from the sensor via ZbBridge triggers false positive "door closed" status. Below is expected the message I receive and my config to parse Json; tele/zigbee_bridge/A9A9/SENSOR = {"ZbReceived":{"home_theatre_door":{"Device":"0xA9A9","Name":"home_theatre_door","Power":0,"Endpoint":1,"LinkQuality":110}}}
and the topic I use to extract "1" door closed and "0" door open
"getContactSensorState": {
                    "topic": "tele/zigbee_bridge/A9A9/SENSOR",
                    "apply": "return !(JSON.parse(message).ZbReceived.home_theatre_door.Power);"
this works fine until below regular unexpected message received from the sensor which triggers "offValue": "1" (Door closed)
tele/zigbee_bridge/A9A9/SENSOR = {"ZbReceived":{"home_theatre_door":{"Device":"0xA9A9","Name":"home_theatre_door","0000/FF02":"null","0000/E521":605268001,"DateCode":1,"StackVersion":"null","ZCLVersion":23,"Endpoint":1,"LinkQuality":110}}}

 Is it my json parse template or the regularly trigered unexpected message that creates false positives. I tried the same with zigbee2mqtt, messages I receive seem more detailed (battery status+voltage) and organized which I can deal with. What do I miss here? Any feedback appreciated.

MattWestb, 2020-08-20:
@unscel Your first received  message ``` {"ZbReceived":{"home_theatre_door":{"Device":"0xA9A9","Name":"home_theatre_door","Power":0,"Endpoint":1,"LinkQuality":110}}}```
Its reported then the contacts status have being changed.
The second message ```  {"ZbReceived":{"home_theatre_door":{"Device":"0xA9A9","Name":"home_theatre_door","0000/FF02":"null","0000/E521":605268001,"DateCode":1,"StackVersion":"null","ZCLVersion":23,"Endpoint":1,"LinkQuality":110}}} ``` its one status report that the device is sending on regular basis. Its dont have the ``` "Power":0 ``` in it then the status of the contact is not changed. Your extraction of messages is not correct made and triggering on status messages from the device that not having the correct syntax ("Powe":).
Taking one more lock how your "extraction" is working and make its only triggering on messages with"Power" in and you is not getting any "false Power" tiggings.

unscel, 2020-08-20:
> @unscel Your first received message ` {"ZbReceived":{"home_theatre_door":{"Device":"0xA9A9","Name":"home_theatre_door","Power":0,"Endpoint":1,"LinkQuality":110}}}`
> Its reported then the contacts [...]

Thank you Matt, I feel there is something wrong with the extraction as it is even triggered by a message with no "Power" keyword involved. Just to confirm if I got you correctly, should the extraction be like - "apply": "return !(JSON.parse(message).ZbReceived.home_theatre_door."Power");"
-or any tool (website) I myself can check if extraction (json(parse) returns expected output?
Thank you again, appreciated

MattWestb, 2020-08-20:
I don't being one software more the hardware but i think you need one "if" some ware so its excluding all messages without your desired "Power". Use google and you finding it.

Hedda, 2020-08-24:
> there is no standard protocol to put a password an a pure TCP connection,

OK, is it instead technically possible to set a username and password in the EmberZNet application firmware on the ERF32 chip?

That is, would it be possible to build a new EmberZNet application firmware in which you could set a username and password?

Thinking that encryption is not needed and just username and password should be enough since not it is not over the internet.

>  password protection is mostly futile unless you encrypt the traffic - which will not happen either because of limitations of the ESP8266.

OK, proper secure connection use server-side TLS/SSL encryption which is bad for ESP8266 because of low RAM + slow CPU.

Do you think that a DYI Zigbee-bridge with ESP32 instead of ESP8266 is powerful enough to encrypt a 115200 baud connection?

Not asking you to implement any of those yourself, just curious so asking if you think that it is technically even possible in theory.

MattWestb, 2020-08-24:
@Hedda You can doing one porting of the tasmota for the EZSP Secure EZSP Protocol Version 2 and you is getting good security from NCP to host. The version 1 Its well documented in "AN1125: Creating and Using a Secure EZSP Host-to-NCP Interface" and other slabs dokuments is being updated with the version 2 and the EZSP 6.8.x.x.

s-hadinger, 2020-08-25:
Agreed, the issue is to implement the secure protocol on the Home Automation software, not in Tasmota. SEZSP does not make sense if it's only about securing the communication between ESP8266 and EFR32. It needs to be end-to-end.

Please revert this request to ZHA. This is off-topic for this GitHub issue.

s-hadinger, 2020-08-25:
@tric111 thanks for reporting the issue with Philips outdoor motion sensor SML002.

I have implemented sending the default response required by this device. It will be available as soon as #9163 gets merged.

MattWestb, 2020-08-26:
Request of implanting battery status from IKEA controller devices.
IKEA controller devices is working well is my experience with the new and old motion sensor and the On/Off dimmer switch (E1743). Have 5 bottom remotes and wireless dimmer but its in produktion net for the moment so cant easy testing.
All devices is missing the new nice future of showing battery status in WEB UI ;-(
On/Off status is not needed / wished from controlling devices then its by ZB3 standard light that suld reporting that attribute to the coordinator but is possible "sniffing" groupe commands but its not 100% then the coordinator can sending unicast and steering devices 2 and then its not in sync.
If you need help with logs from some controlling devices i can doing it for you if you dont finding the reporting attributes online.

The off status and still light level is OK by TB3 standard and is the level the lights have then  sending one on command so don't destroying that behaviour.

By the way is it possible getting one blitz for main powered device in the battery position for indicating its / was "online" ?

[image]
Nice work !!!!!

MattWestb, 2020-08-26:
Request of encoding group address in group messages for easy to configuring device groupes to bounded controller devices. If its possible in dec and hex (42/0x2a).

littlebilly, 2020-08-26:
I paired an Ikea Bulb and a Tradfri Remote Control Model E1810 with my ZbBridge!
Pairing was ok as far as I got.
Problem: Pushing any of the remote buttons do not show an activity on Tasmota Log.
Only the green LED of the ZbBridge shows that commands were send to the Bridge.

`ZbStatus2
18:39:46 MQT: stat/zigbee_05697B/RESULT = {"ZbStatus2":[{"Device":"0x4CB8","Name":"IKEA-Bulb","IEEEAddr":"0x14B457FFFE4EBCB4","ModelId":"TRADFRI bulb E27 WS opal 1000lm","Manufacturer":"IKEA of Sweden","Endpoints":["0x01","0xF2"]},{"Device":"0x24E2","Name":"IKEA-Remote","IEEEAddr":"0x000D6FFFFE052760","ModelId":"TRADFRI remote control","Manufacturer":"IKEA of Sweden","Endpoints":["0x01"]}]}`

What is wrong? Sorry I am a Zigbee Beginner.
Thanks in advance.

MattWestb, 2020-08-26:
@littlebilly To getting the binding of your remote have its not so easy to getting from the gateway but it can being that is configured then paring to groupe 0. My On/Off dimmer switch is doing that and also the new motion sensor but not the old model.
Try adding grupe 0 to the light and testing if its working.
Groupe 0 its the "default" group for not configured device and it not good to using then its being one mess if you having more devices but for beginning its OK.
I have putting one request for getting that from the gateway then its seeing it being broadcasted from one controller device but it was 6 hours ago.
You can also deleting the binding in the remote and making one new to a new groupe but its not so easy to getting right.

s-hadinger, 2020-08-26:
@MattWestb I didn't understand your request for group addresses. Can you please elaborate?

@littlebilly Indeed it is best to bind the remote to a group, not necessarily group 0 since you may want different groups.

See the example for IKEA remote and bulb here: https://tasmota.github.io/docs/Zigbee/#zigbee-binding

MattWestb, 2020-08-26:
@s-hadinger The IKEAs old motion sensor(LL) is making one LL grupe (bonded) and need "sniffing" and calculating it after pairing for adding lights to the groupe.
The new version and the E1743 (ZB3) don't making groupe and sending the command to the default groupe (0).
I don't have trying to binding them to other groups but i think its first need being unbounded before possible binding then to one new groupe.

CKerswell, 2020-08-26:
Hi, thanks for the great work so far on this!

My main request would be to be able to assign a Domoticz device ID to each Zigbee device and send/receive commands/status on the domoticz/in and domoticz/out MQTT topics.

littlebilly, 2020-08-27:
@s-hadinger @MattWestb 
Thanks alot for your quick response.
When binding the remote to a group like:
`ZbBind {"Device":"IKEA-Remote","ToGroup":100,"Endpoint":1,"Cluster":6}
08:05:08 MQT: stat/zigbee_05697B/RESULT = {"ZbBind":"Done"}
08:05:08 ZIG: {"ZbEZSPReceived":"450000000021800000400100005AFFE2E224FFFF021F0002"}
08:05:08 MQT: tele/zigbee_05697B/RESULT = {"ZbBind":{"Device":"0x24E2","Name":"IKEA-Remote","Status":0,"StatusMessage":"SUCCESS"}}`
I got this. 
What I'm missing as described in Zigbee documentation is:
**Receiving Commands from the remote**

> If you pair devices such as switches or remotes, you will also receive commands from those devices.
> When a command is received, attributes are published both in their low-level and high-level formats (if known).

What is going wrong?

[165 more replies]

closed 2020-11-22 by s-hadinger