arendst/Tasmota #9052: RF sensor extension

toastbrot69, 2020-08-08:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
None of the RF decoders can handle NRZ or other “special” signals
- the rc_switch (xdrv_17_rcswitch.ino)
- xsns_37_rfsensor.ino
- Portisch FW 

I mean NRZ signals like this:

```
 ______        ______        ________________                  ________________        ______
       |______|      |______|                |________________|                |______|      |
|-1000-|-1000-|-1000-|-1000-|------2000------|------2000------|------2000------|-1000-|-1000-|
|------1------|------1------|--------0-------|--------0-------|--------0-------|------1------|
```
I also could not decode/monitor this type of signal correctly (1st pulse duration is the bit)
```
 _____            ______________
      |__________|              |__________|
|-550-|---1000---|-----1400-----|---1000---|
|-------1--------|------------0------------|
```

**Describe the solution you'd like**  
Portisch and rc_switch are external projects, so i would prefer an “native-tasmota” solution 

A more flexible (and not busy-waiting) framework where one can easily add new decoders in xsns_37_rfsensor would be nice

**Describe alternatives you've considered**  
updating Portisch with new codecs would prevent OTA (Open the bridge, switch key off, flash,...)

**Additional context**  
All these modulations are from temperature/humidity/etc rf sensors sending on 433 or 868 MHz I have around the house. At this time I’m using an old Raspi with a DVB stick and rtl_433 to decode these, but I think that’s a little over the top for this, also the power consumption much more than a esp.
A cheap receiver attached to an already installed Tasmota light/shutter switch or a sonoff rf bridge could easily do the job.

My private fork with a new sensor in xsns_37 is already working “in production” at my place.

Sorry if this was the wrong place to post this.
Thanks for your attention!

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-08-08:
The chip that decodes the signal is not the esp8266 where Tasmota is, so, your new decoding MUST go to Portisch Firmware.

toastbrot69, 2020-08-08:
That’s true for the sonoff rf bridge (if you don’t wire one extra cable), but not for other devices like a d1 or else where you attach a 5€ receiver.

Jason2866, 2020-08-09:
There is nothing against to add new protocols to the xsns_37_rfsensor.ino driver without breaking changes. Theo suggested this https://github.com/arendst/Tasmota/pull/9036#issuecomment-669916050
Adding a new hardware as receiver (cc1101) will have no great chance to be merged since all
Rf433 and rf868 codes can be decoded with receivers which do not need a driver

toastbrot69, 2020-08-10:
Ok, then i will try to refactory the xsns_37_rfsensor.ino, but i suggest that will be quite a change...

ascillato2, 2020-08-10:
> Ok, then i will try to refactory the xsns_37_rfsensor.ino

Great. Waiting for the PR. Thanks.

closed 2020-08-10 by ascillato2

toastbrot69, 2020-08-11:
Anybody knows a good and cheap receiver except CC1101 (needs driver) or Aurel RX-AM8SF (out of production) for 868MHz?

fmarzocca, 2021-08-08:
> Anybody knows a good and cheap receiver except CC1101 (needs driver) or Aurel RX-AM8SF (out of production) for 868MHz?

I am interested in this too... Did you find anything?