arendst/Tasmota #9053: Status 11 command does not trigger rule for more than 4 relays configured

borky, 2020-08-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Rule StatusSTS#POWER[x] is not triggered when Status 11 command is used and more than 4 relay are configured.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff T1 3CH clone_____
- [x] Tasmota binary firmware version number used: 8.4.0.1_____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Tasmota Compiler_____
- [x] Flashing tools used: Web GUI_____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"Sonoff T1 6CH","GPIO":[17,0,0,7,23,22,18,19,21,56,24,0,0],"FLAG":0,"BASE":30}
{"Module":{"30":"Sonoff T1 3CH"}}
{"GPIO0":{"17":"Button1"},"GPIO1":{"24":"Relay4"},"GPIO2":{"25":"Relay5"},"GPIO3":{"7":"WS2812"},"GPIO4":{"23":"Relay3"},"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"19":"Button3"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":402,"Free":109,"Rules":"on power1#state do var1 %value% endon  on var1#state=1 do backlog led1 %var5%;power1 1 endon on var1#state=0 do led1 %var4% endon ON Event#Pwr1 DO power1 1 ENDON on power3#state do if (var3==0) power4 %value% else delay endon on power4#state do backlog status 11;delay endon  ON Event#Pwr3 DO power4 %value% ENDON ON StatusSTS#POWER4=OFF do led3 %var6% endon ON StatusSTS#POWER4=ON do led3 %var7% endon"}
{"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":246,"Free":265,"Rules":"on power2#state do var2 %value% endon on var2#state=1 do led2 %var9% endon on var2#state=0 do led2 %var8% endon ON mqtt#connected DO Subscribe PwrLedLiving, stat/sonoff_led_rafturi_living/POWER1 ENDON ON Event#PwrLedLiving DO power2 %value% ENDON"}
{"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":567,"Free":201,"Rules":"on system#boot do backlog var4 0000A0;var5 A00000;var6 000080;var7 800000;var8 000030;var9 300000;power1 0;var1 0;var2 0 endon on System#Boot do backlog status 11;delay endon ON mqtt#connected DO backlog Subscribe Pwr1, cmnd/sonoff_living/POWER1;Subscribe Pwr3, cmnd/sonoff_living/POWER3 ENDON on event#connected=0 do backlog buttontopic 0;var3 0 endon on event#connected=1 do if (var1==2) buttontopic 0 else buttontopic sonoff_living;var3 1 endif endon on event#online do backlog event connected=1;RuleTimer8 %v..."}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:45:59 CMD: Status 0
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS = {"Status":{"Module":30,"DeviceName":"sonoff_intrerupator_living","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","Tasmota5","Tasmota6"],"Topic":"sonoff_intrerupator_living","ButtonTopic":"sonoff_living","Power":4,"PowerOnState":3,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:14:00","StartupUTC":"2020-08-08T16:31:59","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2020-08-08T15:36:50","SaveCount":120,"SaveAddress":"F9000"}}
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS2 = {"StatusFWR":{"Version":"8.4.0.1(tasmota)","BuildDateTime":"2020-08-08T16:10:27","Boot":31,"Core":"2_7_3_2","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"455/699"}}
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000A00060000005A00000000000000","00000000","00006000"]}}
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F80C582","04260001","000000C1","00000000","C000F981","00000004"],"Drivers":"1,2,4,6,8,9,10,18,19,22,24,26,27,29,30,35,37","Sensors":"1,2,4,5,6"}}
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS5 = {"StatusNET":{"Hostname":"sonoff_intrerupator_living-4435","IPAddress":"xxx","Gateway":"xxx","Subnetmask":"255.255.255.0","DNSServer":"xxx","Mac":"xxx","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS6 = {"StatusMQT":{"MqttHost":"xxx","MqttPort":xxx,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9E7153","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS7 = {"StatusTIM":{"UTC":"2020-08-08T16:45:59","Local":"2020-08-08T17:45:59","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:34","Sunset":"20:17"}}
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS10 = {"StatusSNS":{"Time":"2020-08-08T17:45:59"}}
17:45:59 MQT: stat/sonoff_intrerupator_living/STATUS11 = {"StatusSTS":{"Time":"2020-08-08T17:45:59","Uptime":"0T00:14:00","UptimeSec":840,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":1,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:05"}}}
17:46:00 MQT: stat/sonoff_intrerupator_living/RESULT = {"Event":"Done"}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure 5 relays and one WS2812(6 POWERs).
Define rule: ON StatusSTS#POWER4 do var1 1 endon
Send command Status 11

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Rule should be triggered, but it is not.
It works as expected with 4 relays, but not with 5 or more.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I don't have anything connected to some of the relays i configured, i am using those as virtual devices to control the configured WS2812 leds.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-08-08:
Hi,

Your rules have an error.

Please delete the double spaces they have between ENDON and ON, and try again. This error is between several rules.

borky, 2020-08-08:
I deleted, still the same. The rules i pasted work fine when i have maximum 5 POWER controls(4 relays and 1 WS2182).
The trigger for the command Status 11 stops working when i add one more Power control(Relay).

arendst, 2020-08-11:
This is currently as designed. The problem is that the status 11 message is becoming too long.

The internal JSON parser reserves an amount of RAM (currently 1024 bytes buffer) to decode the status 11 message. As your message contains many JSON values it's running out of memory and returns invalid JSON.

As a workaround you might want to compile a version with an increased amount of buffer. In that case just change line 500 in xdrv_10_rules.ino from
```
  StaticJsonBuffer<1024> jsonBuf;
```
into 
```
  StaticJsonBuffer<1280> jsonBuf;
```
Pls report back if that solves your issue.

arendst, 2020-08-11:
I'll reinvestigate moving from ArduinoJason 5 to ArduinoJson 6. The latter now supports StringDeduplication which uses less memory as it only stores one "POWER" string with pointers to the other occurences.

The previous hold back of unavailable StaticJsonBuffer alternative has also been solved in 6.

If the overall memory footprint is less or slightly larger I'll move to 6 which will need so global tasmota rewrites...

arendst, 2020-08-11:
Alas. it still isn't possible to dynamically resize a DynamicJsonBuffer according to the conversion handbook:

> DynamicJsonDocument has a fixed capacity that you must specify to the constructor. Unlike the DynamicJsonBuffer, DynamicJsonDocument doesn’t automatically expand.

Considering the answer from Benoit (https://github.com/bblanchon/ArduinoJson/issues/880#issuecomment-450886158) I doubt we will ever get rid of ArduinoJson 5...

A search for "DynamicJsonBuffer" reveals many uses in xdrv_05_irremote_full.ino, xdrv_10_scripter.ino, xdrv_23_zigbee_x.ino, xdrv_39_thermostat.ino and xdrv_4-_telegram.ino

I can fix the latter but all others are written by supporting programmers like @s-hadinger , @gemu2015 and @arijav. If they could only decide on what max DynamicJsonBuffer they needed... (Dynamic here means heap location, Static is stack location and is bound to about max 1k)

s-hadinger, 2020-08-11:
In several places, it's almost impossible to know in advance the size of the JSON buffer. Allocating a random number like 1KB or 2KB is not a good compromise and it is a ticking bomb. Most of the time the JSON buffers are very small but sometimes they can get bigger - and I can't know precisely in advance.

I would on the contrary change `StaticJsonBuffer<1024> jsonBuf;` into `DynamicJsonBuffer`.

arijav, 2020-08-11:
I did not get the full discussion, does this mean that
the DynamicJsonBuffer is in reality not dynamic at all and we need to fix a
length to fit us all?

s-hadinger, 2020-08-11:
In ArduinoJSON 5 (used in Tasmota) it is dynamic unless you choose the stats one. In ArduinoJSON 6 there is no dynamic option anymore.

effelle, 2020-08-11:
`DynamicJsonBuffer` is for AJ5 and is what we are using now but was replaced by `DynamicJsonDocument` on AJ6 and if we migrate to it we wil face issue with allocation because isn't able to expand itself automatically. 
I tend to agree with @s-hadinger, using it instead of the `StaticJsonBuffer` could lead to unpredictable effects (and btw I'm still using a static buffer on Discovery).

borky, 2020-08-11:
> This is currently as designed. The problem is that the status 11 message is becoming too long.
>
> The internal JSON parser reserves an amount of RAM (currently 1024 bytes buffer) to decode the status [...]

@arendst  This change fixed the issue i had. Thanks!

arendst, 2020-08-12:
How about implementing a kind of dynamic DynamicJsonDocument jsonBuf(capacity) by calculating the capacity needed before the call to DynamicJsonDocument is perfomed.

As an example in rules I can safely use:
```
  const size_t capacity = 128 + (event.length() * 2);  // Offset of 128 to cater for short JSON
  DynamicJsonDocument root(capacity);
  DeserializationError error = deserializeJson(root, event);
  if (error) {
    AddLog_P2(LOG_LEVEL_DEBUG, PSTR("RUL: Json too long (%d)"), event.length());
    return false;                                      // No valid JSON data
  }
```
where ``event`` is any JSON message from short to very long. Testing some JSON messages in the AJ6 assistant leads to the rough formula ``128 + (event.length() * 2)``.

Also for JsonSerialization a rough JSON string length can be calculated by the programmer based on the amount of field pairs and their know value names.

s-hadinger, 2020-08-12:
That's not so much de serialization that concerns me. It's more serialization with unknown size at the beginning. This is typically with Zigbee. You don't know in advance how many fields there are. Which means there will be an overshooting of buffers here. 

Do you Know the migration path from 5 to 6? ArduinoJSON is a delicate thing that can easily drive to a crash.

arendst, 2020-08-12:
> Do you Know the migration path from 5 to 6?

The only info available is the guide from Benoit and the issue I noted before of not supporting Dynamic expansion. I did a rough implemntation on a side branch and after some major refactoring it works.

How about deviating from the ArduinoJson serialization and do it the way I do/did by doing it yourselves using a string. This solves the unknown fields issue.

arendst, 2020-08-12:
OK. I'll stop further investigation in ArduinoJson v6. It's just no good compared to AJ5.

closed 2020-08-14 by arendst