arendst/Tasmota #9054: timer for shuttermode

5chufti, 2020-08-08:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
IMHO the current timer options don't make sense for shutters. 
I can't imagine anyone wanting to switch his actuator hard on/off for up/down at certain time for given periode to get his blinds to a wanted shutterposition.

**Describe the solution you'd like**  
I would consider it more userfriendly like: at day/time XX goto shutterposition YY. 

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
maybe I just missed an option and only need a push to the correct wiki/docs paragraph

**(Please, remember to close the issue when the problem has been addressed)**  

thank you for the great work so far

stefanbode, 2020-08-10:
The Timer module only supports relays and only supports ON/OFF. your target command should be something like "shutterposition1 60" to set the shutter to 60%. I would add an additional relay on a free GPIO and then create a rule that send the shutter to the wanted position when the relay goes ON. Enhancing the TIMER would be difficult. Same with dimmers. You cannot use the timer to seta dimmer to a specific value.

ascillato, 2020-08-10:
Timers can trigger rules at the time you want. So, you can make a rule with the behaviour you need

closed 2020-08-10 by ascillato2

5chufti, 2020-08-11:
> Enhancing the TIMER would be difficult.

but would make lots of sense for shutters and dimmers; so better ignore a request for a good solution in favour of a clumsy workaround?