arendst/Tasmota #9060: IRHVAC command should always return true mode state

yoav1000, 2020-08-09:
**Have you looked for this feature in other issues and in the docs?**  yes

**Is your feature request related to a problem? Please describe.**  
Some of The AC vendors supported by IRremoteESP8266 library and tasmota do not support ON/OFF commands,
Instead they support toggle operations.
For example the WHIRLPOOL and AIRWELL protocols indicate power state and swing state change, not absolute state.

Tasmota always return mode=Off if Power=Off, preventing automation tracking mode state.
https://github.com/arendst/Tasmota/blob/d321b6cf68334a278dac0acb2d3f3e0312619dd1/tasmota/xdrv_05_irremote_full.ino#L121

**Describe the solution you'd like**  
Delete this if state or add configuration for this behavior.

**Describe alternatives you've considered**  
See #9004

**Additional context**  
Some examples:
ON capture
`{"IrReceived":{"Protocol":"WHIRLPOOL_AC","Bits":168,"Data":"0x0x830605A200008C0200000000002900010000080009","Repeat":0,"IRHVAC":{"Vendor":"WHIRLPOOL_AC","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":26,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`
OFF capture
`{"IrReceived":{"Protocol":"WHIRLPOOL_AC","Bits":168,"Data":"0x0x830605A200008C0100000000002A00010000080009","Repeat":0,"IRHVAC":{"Vendor":"WHIRLPOOL_AC","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":26,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`
Temp change (you can see the mode is off mistakenly)
`{"IrReceived":{"Protocol":"WHIRLPOOL_AC","Bits":168,"Data":"0x0x830601B200008C0400000000003B0002000008000A","Repeat":0,"IRHVAC":{"Vendor":"WHIRLPOOL_AC","Model":2,"Power":"Off","Mode":"Off","Celsius":"On","Temp":27,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
{"IrReceived":{"Protocol":"WHIRLPOOL_AC","Bits":168,"Data":"0x0x830601A200008C0400000000002B0002000008000A","Repeat":0,"IRHVAC":{"Vendor":"WHIRLPOOL_AC","Model":2,"Power":"Off","Mode":"Off","Celsius":"On","Temp":26,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-08-09:
The mode is `Off` because Power is `Off` which makes sense if you consider the Power value.

I understand that capturing a Temperature change should trigger Power to `On`, not to `Off`. Is it possible to change the temperature without turning on the device?

In such case you need to push the request to IRRemoteESP8266 lib.

yoav1000, 2020-08-09:
The main problem is that some of the AC vendors do not use absolute Power states and use the Power value as a toggle.
You can see from my example above that the OFF an ON capture are equal (the remote send the same code), When the user change temperature the remote doesn't know if the AC is On or OFF, in the case the AC is Off, its still receive the value but doesn't do anything.

Because of the current code behavior, you cant know if the AC changed Mode from Cool to Heat until the user switch Off the AC

I also think that Off and On should be replaced by Toggle and None, but that not really important.

See this Answer from the ir lib: https://github.com/crankyoldgit/IRremoteESP8266/issues/1228#issuecomment-666720638

yoav1000, 2020-08-09:
I think Tasmota need to go in one of two ways regarding AC:
1. Reflect the **true** state of the AC - for some of the ac vendors require keeping internal state.
2. Pass the **true** state of the remote - today we change the mode from the result preventing any automation from keeping track of changes

ascillato2, 2020-08-10:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-10 by ascillato2

ayavilevich, 2020-09-20:
Hi @yoav1000 

I had the same issue with an Airwell AC.
I believe the better way is for tasmota to implement IRremote properly and handle state. Currently they just pass "null" for "prev".
IRremote has an implementation for managing state for toggle properties. It has some issues but it is good enough. Whatever is missing will be fixed (https://github.com/crankyoldgit/IRremoteESP8266/issues/1275).

You are welcome to try my fork. If it works maybe they will accept the change in the main repository.
https://github.com/ayavilevich/Tasmota