arendst/Tasmota #9062: Support LSC GU10 CCT

deku-m, 2020-08-09:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
I just bought a LSC GU10 from Action. Didnt look but there are new types available as this is normal CCT, Only white or dim.
Looks like this isn't supported by Tasmota. Tried to flash but smartconfig keeps sending packets.

**Describe the solution you'd like**  
Maybe if possible support for this type?

**Describe alternatives you've considered**  
It is possible to use them with Tuya. But rather have it without cloud function.
As tasmota firmware is awesome.

**Additional context**  
https://www.action.com/globalassets/cmsarticleimages/76/02/3000267_8712879146223-111.png?preset=mediaSliderImageLargeHD

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-08-10:
How do you know Tasmota is not working? You havent flashed Tasmota on it or did i missunderstood you? Tasmota is in no way related to Tuya Convert.

sfromis, 2020-08-10:
If Tuya-convert fails, the device may have newer non-hacked firmware, or not use an ESP8266.

deku-m, 2020-08-10:
> 
> 
> How do you know Tasmota is not working? You havent flashed Tasmota on it or did i missunderstood you? Tasmota is in no way related to Tuya Convert.

You are right i mean tuya convert is failing as soon as the light isnt blinking again and solid it fails and keeps sending packets.
I will adjust call.

deku-m, 2020-08-10:
> 
> 
> If Tuya-convert fails, the device may have newer non-hacked firmware, or not use an ESP8266.

Thats also my thoughts on this but i dont know.

deku-m, 2020-08-10:
Ask for support on tuya-convert.

closed 2020-08-16 by ascillato2