arendst/Tasmota #9063: Enter MQTT setting in the initial phase

Hydrosys4, 2020-08-10:
**Have you looked for this feature in other issues and in the docs?**  
YES
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
AS FROM THE DOCUMENTATION:
MQTT setting is possible only after the ESP is connected to a wifi network, so to set the MQTT it is required to know the IP address of the device which is not straightforward (need to access the router inyterface). 
If MQTT can be set when the ESP is in wifi access point mode, this will greatly simplify its configuration reducing unnecessary steps.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Add MQTT setting when the ESP is in WiFi access point mode (initial config)

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
NO

**Additional context**  
_Add any other context or screenshots about the feature request here._
Add MQTT setting in the initial setting when in AP mode.

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-08-10:
What is the reason/justification? After all, if you are trying to avoid having to know the device IP address in order to have MQTT set up, what about all the other configuration settings that you'll need to set up on the device. Then the question becomes, where do you stop in terms of initial setup? This quickly snowballs...

Setting up MQTT on the device only requires a connection to the device. Typically you access the device by knowing its IP address and using the web UI. However, you can also use the serial interface used to flash the device after the device boots into Tasmota. So, all configuration can be done via the serial interface... including inspecting what the device's IP address is after it connects to Wi-Fi.

What I want to distinguish is that you do NOT need the device IP as part of the MQTT setup parameters. You only need to know the network details of the MQTT broker... which you should already know prior to setting up the Tasmota device.

sfromis, 2020-08-10:
If you configure a host name in the initial config form, you can use that host name to connect to the device, without having to know the IP address.

sfromis, 2020-08-10:
While I've not tested this, I think that you'd be able to issue http commands to do stuff in the AP mode. While there is no web interface for this, this may not stop you from 192.168.4.1/cm?cmnd=xxxx

Jason2866, 2020-08-10:
Using Tasmotizer you even dont have to connect to the device to configure wifi.
Set up all your needed settings there.
The wifi ap mode is for setting up the one and only setting to bring the device in wlan network.
There will be no change made.

Hydrosys4, 2020-08-11:
Is this closed because duplicated? cannot find other issues related to this.
At least give me the opportunity to reply....

Well, I've created an irrigation/automation system based on the Raspberry PI. (https://hydrosysblog.wordpress.com/)
I'm adding MQTT support and Tasmota is a very easy to use tool. 
The goal is to make the configuration as easy as possible, and accessing the router to find the IP is definitely not an user friendly step.

I asked if possible to add only the MQTT parameters because the configuration could be done in this way:
1) When in AP mode initial configuration, add the wifi credentials, MQTT broker and topic.
2) The device connects to the wifi and the broker
3) using the MQTT command I can retrieve the IP address of the device
4) with the IP address, I can access the GUI I set all the other parameters.

No need to search the IP address.

Hope you can reconsider my proposal.

meingraham, 2020-08-11:
> using the MQTT command I can retrieve the IP address of the device

What "MQTT Command" are you referring to? Where do you invoke that command?

stefanbode, 2020-08-11:
It already worked if you compile the mqtt-setting into the firmware with a user config-override. I assume it is a small deal to make the mqtt config accessible during ap-mode. That might solve the issue already

meingraham, 2020-08-11:
@stefanbode & @Hydrosys4 

I updated my previous post to suggest using the Console during the AP connection. I haven't tried this, but it may also be possible to navigate to any of the Tasmota menus, e.g., http://192.168.4.1/mq and save (the device will reboot). But since Wi-Fi is still not configured, it should come back into AP mode and then connecting to 192.168.4.1 allow entering the Wi-Fi credentials. This would require no modifications to Tasmota (e.g., adding MQTT entry in a new menu page with both Wi-Fi and MQTT).

Jason2866, 2020-08-11:
There will be no change made. There are several easy ways for a seamless mass deployment.

Hydrosys4, 2020-08-12:
@ meingraham
Thank you, it worked as described.
1) use the http://192.168.4.1/mq to configure the MQTT first, then save (restart)
2) It restarts still in AP mode (the SSID is changed according to the topic), so it is possible to configure the wifi using the http://192.168.4.1

Thanks for the hint and for the time.
Great project!

meingraham, 2020-08-12:
Answered

closed 2020-08-12 by meingraham