arendst/Tasmota #9067: Web GUI Toggle1 does not show correct status when connected to "NO" Circuit of Relay

NsinghP, 2020-08-10:
Flashed NodeMCU2866-ESP-12E Module (ESP Chip Id 13312240,Flash Chip Id 0x1640EF, Flash Size 4096kB) with Tasmota Program Version 8.4.0(sensors), Core/SDK Version 2_7_2_1/2.2.2-dev(38a443e). I am using Module type (Sonoff Basic) - Generic (0) with following parameters: D6 GPIO12 Relay1: Relay1 (21), D5 GPIO14 Sensor: Switch1 (9). I have used jumper cables to connect a single channel relay Module's "IN" Pin to NodeMCU at Pin "D6 GPIO12" and TTP223 Module's "I/O" Pin to Pin "D5 GPIO14". A light bulb working on 220v is connected to the Relay's "NC" port. The Setup is working great. I can turn the light on and off both from the TTP223 touch sensor and Tasmota's Web GUI "Toggle1" button. This setup also remembers the last state if there is power outage. But there is a huge saftey issue with this setup because if there is a problem with 5v AC to DC power supply and it fails, the light bulb will stay constantly on because it is connected to "NC" circuit of the Relay. I would really like to use the "NO" circuit of the Relay for this setup, but when I connect the light bulb to "NO" port of Relay then Tasmota's Web GUI "Toggle1" button information is not correct and it shows "OFF" when the Light is ON and "ON" when light is OFF. How do I get Tasmota's Web GUI to show correct status? Can a new (Sonoff Basic)-Generic Module be made to enable correct display of status when using "NO" port of Relay? All help will be appreciated.

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
![ON when OFF-NO Circuit](https://user-images.githubusercontent.com/69470381/89810100-273bee00-db5a
[image]
-11ea-9c3c-0c653054de4e.jpeg)
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-08-10:
Hi, please change your gpio definition from relayX to relayXi and your issue will be solved. Thanks

NsinghP, 2020-08-10:
> Hi, please change your gpio definition from relayX to relayXi and your issue will be solved. Thanks

Thanks...That worked

closed 2020-08-10 by ascillato2