arendst/Tasmota #9068: Windmeter for ESP32

Ingenieur89, 2020-08-10:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-08-11:
Without enabling define USE_WINDMETER you won't see the GPIO (on either ESP8266 or ESP32). So pls check if you enabled this define for ESP32 as it is default not defined in ``my_user_config.h``.

closed 2020-08-11 by ascillato2

ionultd, 2021-03-20:
I have just build Tasmota32 9.3.1 for sensors and have #define USE_WINDMETER in my user_config_override.h. When entering in the console Sensor68 parameter settings I gat an error as in [Image1](https://i.imgur.com/akbDyGM.png)