arendst/Tasmota #9071: Rolling Shutter - Homekit "opening" or "closing" continuous message are showed

GerSant, 2020-08-11:
### PROBLEM DESCRIPTION
When I control my rolling shutter from the physical buttons of the Tuya device, example, open...wait a couple of seconds and press stop, Homekit (through mqttthing and Homebridge) shows "opening...." all the time, even when the rolling shutter has stop

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Avatto SYS-CS01 (Tuya)
- [X] Tasmota binary firmware version number used: 8.3.1
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Atom
- [ ] Flashing tools used: _____
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

22:20:13 MQT: stat/avatto-cd27/RESULT = {"NAME":"Avatto SYS-CS01","GPIO":[56,158,23,18,22,11,0,0,0,21,57,0,17],"FLAG":0,"BASE":18}
22:20:13 MQT: stat/avatto-cd27/RESULT = {"Module":{"0":"Avatto SYS-CS01"}}
22:20:13 MQT: stat/avatto-cd27/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"158":"LedLinki"},"GPIO2":{"23":"Relay3"},"GPIO3":{"18":"Button2"},"GPIO4":{"22":"Relay2"},"GPIO5":{"11":"Switch3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"21":"Relay1"},"GPIO14":{"57":"Led2i"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

22:20:48 MQT: stat/avatto-cd27/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":57,"Free":454,"Rules":"on Power3#State=1 do backlog ShutterStop1; Power3 0 endon"}
22:20:48 MQT: stat/avatto-cd27/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
22:20:48 MQT: stat/avatto-cd27/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

22:21:18 MQT: stat/avatto-cd27/STATUS = {"Status":{"Module":0,"DeviceName":"Avatto-CD27","FriendlyName":["Avatto-CD27","Tasmota2","Tasmota3"],"Topic":"avatto-cd27","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,4,4,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:21:18 MQT: stat/avatto-cd27/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"10T21:58:31","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"1970-01-01T00:00:00","SaveCount":152,"SaveAddress":"F4000"}}
22:21:18 MQT: stat/avatto-cd27/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-06-26T08:05:59","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"414/699"}}
22:21:18 MQT: stat/avatto-cd27/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HomeLAN-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000000","00000000"]}}
22:21:18 MQT: stat/avatto-cd27/STATUS4 = {"StatusMEM":{"ProgramSize":477,"Free":524,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000809","8E886583","04260001","000000C1","010000C0","C000F981","00000024"],"Drivers":"1,2,4,6,8,10,18,19,22,24,26,27,29,30,35,37","Sensors":"1,2,4,5,6"}}
22:21:18 MQT: stat/avatto-cd27/STATUS5 = {"StatusNET":{"Hostname":"avatto-cd27-3367","IPAddress":"192.168.0.221","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"CC:50:E3:E7:CD:27","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:21:18 MQT: stat/avatto-cd27/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.253","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E7CD27","MqttUser":"DVES_USER","MqttCount":14,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:21:18 MQT: stat/avatto-cd27/STATUS7 = {"StatusTIM":{"UTC":"1970-01-11T22:21:18","Local":"1970-01-11T22:21:18","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00"}}
22:21:18 MQT: stat/avatto-cd27/STATUS10 = {"StatusSNS":{"Time":"1970-01-11T22:21:18","Switch3":"OFF","Shutter1":{"Position":46,"Direction":0,"Target":47}}}
22:21:18 MQT: stat/avatto-cd27/STATUS11 = {"StatusSTS":{"Time":"1970-01-11T22:21:18","Uptime":"10T21:58:31","UptimeSec":943111,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":14,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Wifi":{"AP":1,"SSId":"HomeLAN-IoT","BSSId":"B8:69:F4:B8:C6:DD","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":13,"Downtime":"0T00:25:14"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Start to open (or close) the shutter, press stop and Homekit shows the percent of the rolling shutter position

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The "opening" or "closing" continuous message are showed

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-08-11:
This issue can probably not be solved (soon? ever?) from Tasmota side since none of the developers use such a set up. Everyone is invited to find the reason and a solution for. If the code size is minimal and its not a breaking change there is a chance that a provided PR get merged.

Jason2866, 2020-08-11:
One thing. Tasmota does not work with all features without correct time and date.
Maybe thats the reason for your issue

stefanbode, 2020-08-14:
Can you upload the seriallog? best with loglevel 4. This will give some insight if your configuration send the right MQTT commands. Also a screenshot from the error would be helpful. I have no Homekit, therefore debugging will be a bit tricky but willing to help.

stefanbode, 2020-08-14:
As @Jason2866  mentioned:
```
22:21:18 MQT: stat/avatto-cd27/STATUS10 = {"StatusSNS":{"Time":"1970-01-11T22:21:18","Switch3":"OFF","Shutter1":{"Position":46,"Direction":0,"Target":47}}}
22:21:18 MQT: stat/avatto-cd27/STATUS11 = {"StatusSTS":{"Time":"1970-01-11T22:21:18","Uptime":"10T21:58:31","UptimeSec":943111,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":14,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Wifi":{"AP":1,"SSId":"HomeLAN-IoT","BSSId":"B8:69:F4:B8:C6:DD","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":13,"Downtime":"0T00:25:14"}}}
```
The date 1970 looks quite a bit strange here. I had a similar timesync problem, but do not remember what was the problem. I think a reboot of my Fitz router fixed the issue. Without a proper time many things can go wrong.

Zer0x00, 2020-08-14:
Could you use [MQTT Explorer](http://mqtt-explorer.com/) and show us the values of the shutter?

ascillato2, 2020-08-16:
@GerSant

Hi, any news on this?

closed 2020-08-20 by ascillato2