arendst/Tasmota #9073: SM2135E not working with Tuya TYLC5 (esp8285)

CrudelisPL, 2020-08-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [*] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [*] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [*] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [*] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Polux G45, 5,5W, 470lm, E14
- [ ] Tasmota binary firmware version number used:8.4
  - [*] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: Tuya-convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
```
13:34:38 RSL: STATE = {"Time":"2020-08-11T13:34:38","Uptime":"0T00:25:31","UptimeSec":1531,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"00000047B8","HSBColor":"249,100,0","White":100,"CT":404,"Channel":[0,0,0,28,72],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"DOM","BSSId":"30:5A:3A:6E:15:B8","Channel":6,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:34:38 CMD: Backlog Template; Module; GPIO 255
13:34:38 SRC: WebConsole from 192.168.1.200
13:34:38 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
13:34:38 SRC: Backlog
13:34:38 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
13:34:38 RSL: RESULT = {"NAME":"A65 SMDWWCW+RG","GPIO":[0,0,0,0,0,0,0,0,181,0,180,0,0],"FLAG":0,"BASE":18}
13:34:39 SRC: Backlog
13:34:39 CMD: Group 0, Index 1, Command "MODULE", Data ""
13:34:39 RSL: RESULT = {"Module":{"0":"A65 SMDWWCW+RG"}}
13:34:39 SRC: Backlog
13:34:39 CMD: Group 0, Index 1, Command "GPIO", Data "255"
13:34:39 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"181":"SM2135 Dat"},"GPIO13":{"0":"None"},"GPIO14":{"180":"SM2135 Clk"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
13:30:45 CMD: Group 0, Index 1, Command "STATUS", Data "0"
13:30:45 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_2F696C","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:30:45 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:21:38","StartupUTC":"2020-08-11T12:09:07","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2020-08-10T19:21:12","SaveCount":323,"SaveAddress":"FB000"}}
13:30:45 RSL: STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:05:28","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"380/699"}}
13:30:45 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":0,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["DOM",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00006000"]}}
13:30:45 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
13:30:45 RSL: STATUS5 = {"StatusNET":{"Hostname":"RGB LED BULB 001","IPAddress":"192.168.1.253","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"24:62:AB:2F:69:6C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:30:45 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-08-11T12:30:45","Local":"2020-08-11T13:30:45","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:38","Sunset":"20:12"}}
13:30:45 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-08-11T13:30:45"}}
13:30:45 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-08-11T13:30:45","Uptime":"0T00:21:38","UptimeSec":1298,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"00000047B8","HSBColor":"249,100,0","White":100,"CT":404,"Channel":[0,0,0,28,72],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"DOM","BSSId":"30:5A:3A:6E:15:B8","Channel":6,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Flash with tuya-convert
2. Make a config with SM2135

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-08-13:
[Reported](https://discord.com/channels/479389167382691863/479389167382691865/743253735697809452) as operational with a reworked [driver](https://cdn.discordapp.com/attachments/479389167382691865/743262157348536360/xlgt_04_sm2135.ino).

arendst, 2020-08-14:
Well the "NEW" code doesn't work for the "old" LSC bulb as the colors are different; LSC is GRB and Polux is BGR. The timing seems fine but I hate the interrupt toggles. Need further investigation.

arendst, 2020-08-14:
The latest commit contains most of @CrudelisPL code saused with Tasmota adjustements. The new code works fine with my LSC bulb but I couldn't test the Polux one.

Notice in order to use the Polux bulb you'll need a change in the template. The correct templates now are:
```
 * Action LSC SmartLed (GreenRedBlue)
 * {"NAME":"LSC RGBCW LED","GPIO":[0,0,0,0,0,0,0,0,181,0,180,0,0],"FLAG":0,"BASE":18}
 * Polux E14 (BlueGreenRed) - Notice GPIO00 = 9 (Switch1)
 * {"NAME":"Polux RGBCW E14","GPIO":[9,0,0,0,0,0,0,0,181,0,180,0,0],"FLAG":0,"BASE":18}
```
where the LSC template is unchanged but the Polux adds a ``switch1`` on GPIO0 just for detecting the needed RGB sequence change. The switch won't pop up in any status or state message as it is discarded as soon as the driver is initialized. Another Tasmota quirk which I like.

closed 2020-08-16 by ascillato2

Ganbit666, 2020-08-16:
I also have "Polux G45 5.5W 470lm E14" and I can't make it work either.
I tried
"{" NAME ":" Polux RGBCW E14 "," GPIO ": [9,0,0,0,0,0,0,0,181,0,180,0,0]," FLAG ": 0," BASE ": 18} "
And it still doesn't work. I opened it and they all match up. It looks as if SM2135E is not receiving a signal from ESP8285.
I also changed mapping RGBWcWw to B, G, R, Ww, Wc - "SetOption37 55"

Please help

CrudelisPL, 2020-08-16:
I will compile later. But try this one:
[https://github.com/CrudelisPL/Tasmota](url)

sfromis, 2020-08-16:
The great update from CrudelisPL is already included in the precompiled binary from http://thehackbox.org/tasmota/

Ganbit666, 2020-08-16:
Thank you 
  Dnia 16 sierpnia 2020 18:02 sfromis   napisał(a):  
 The great update from CrudelisPL is already included in the precompiled binary from  zasobygwp.pl thehackbox.org

Ganbit666, 2020-08-16:
Thank you 
  Dnia 16 sierpnia 2020 15:44 CrudelisPL   napisał(a):  
 I will compile later. But try this one:  
 github.com