arendst/Tasmota #9077: White Slider for MagicHome RGB ZJ-WFMN-B V1.1 LED

Sineos, 2020-08-12:
### PROBLEM DESCRIPTION
The above mentioned device is RGBW. I would like to get a white slider in addition to the already present color slider.
- Device is setup according to: https://templates.blakadder.com/magichome_ZJ-FWMN-A_RGB.html
- RGB and White Split is defined with `SetOption37 128`
- The `White X` command will produce an error

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): MagicHome RGB ZJ-WFMN-B V1.1 LED
- [x] Tasmota binary firmware version number used: 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
13:45:07 MQT: stat/TM_TerrasseLED/RESULT = {"NAME":"MagicHome RGB","GPIO":[0,0,0,0,0,37,0,0,38,39,0,0,0],"FLAG":0,"BASE":34}
13:45:07 MQT: stat/TM_TerrasseLED/RESULT = {"Module":{"34":"MagicHome"}}
13:45:08 MQT: stat/TM_TerrasseLED/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"147":"ALux IrRcv"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"39":"PWM3"},"GPIO13":{"0":"None"},"GPIO14":{"37":"PWM1"},"GPIO15":{"159":"ALux IrSel"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
n/a

```
- [x] Provide the output of this command: `Status 0`:
```
13:46:26 MQT: stat/TM_TerrasseLED/STATUS = {"Status":{"Module":34,"DeviceName":"Tasmota","FriendlyName":["TM_TerrasseLED"],"Topic":"TM_TerrasseLED","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:46:26 MQT: stat/TM_TerrasseLED/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:25:16","StartupUTC":"2020-08-12T11:21:10","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2020-08-11T19:51:42","SaveCount":81,"SaveAddress":"FB000"}}
13:46:26 MQT: stat/TM_TerrasseLED/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:05:28","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"413/699"}}
13:46:26 MQT: stat/TM_TerrasseLED/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SweetHomeSpot24",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000180060000005A00000000000000","000000C0","00006000"]}}
13:46:26 MQT: stat/TM_TerrasseLED/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
13:46:26 MQT: stat/TM_TerrasseLED/STATUS5 = {"StatusNET":{"Hostname":"TM_TerrasseLED-3516","IPAddress":"192.168.178.58","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"DC:4F:22:F9:0D:BC","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:46:26 MQT: stat/TM_TerrasseLED/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.22","MqttPort":1883,"MqttClientMask":"TM_TerrasseLED","MqttClient":"TM_TerrasseLED","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:46:26 MQT: stat/TM_TerrasseLED/STATUS7 = {"StatusTIM":{"UTC":"2020-08-12T11:46:26","Local":"2020-08-12T13:46:26","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:10","Sunset":"20:37"}}
13:46:26 MQT: stat/TM_TerrasseLED/STATUS10 = {"StatusSNS":{"Time":"2020-08-12T13:46:26"}}
13:46:26 MQT: stat/TM_TerrasseLED/STATUS11 = {"StatusSTS":{"Time":"2020-08-12T13:46:26","Uptime":"0T00:25:16","UptimeSec":1516,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"FF2200","HSBColor":"8,100,100","Channel":[100,13,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SweetHomeSpot24","BSSId":"B6:FB:E4:F4:13:A4","Channel":11,"RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
13:47:43 CMD: weblog 4
13:47:43 MQT: stat/TM_TerrasseLED/RESULT = {"WebLog":4}
13:47:43 CFG: Saved to flash at FA, Count 82, Bytes 4096
13:47:47 CMD: white 100
13:47:47 SRC: WebConsole from 192.168.178.44
13:47:47 CMD: Group 0, Index 1, Command "WHITE", Data "100"
13:47:47 MQT: stat/TM_TerrasseLED/RESULT = {"Command":"Error"}
```

### TO REPRODUCE
Freshly flashed device

### EXPECTED BEHAVIOUR
White slider just like the color picker to appear

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
n/a

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-08-12:
its very difficult split W from RGB on an RGB only controller. what you're trying to do is create a white channel out of nowhere since the linked hardware doesn't support it

closed 2020-08-12 by blakadder

Sineos, 2020-08-13:
@blakadder 
This is in fact a RGBW controller and I have connected a RGBWW led-stripe.
The model no. printed on my PCB is exactly `ZJ-WFMN-B V1.1`

[image]

Why it is named RGB on https://templates.blakadder.com/magichome_ZJ-FWMN-A_RGB.html I do not know.

Sorry for the confusion and any help would be appreciated.

blakadder, 2020-08-13:
It's not the name that's the issue, it's the fact that the template supports only 3 channels or RGB

Sineos, 2020-08-13:
I guess the link you pointed to should be: https://templates.blakadder.com/magichome_ZJ-WFMN-B.html

Unfortunately this does not work as well. Also it seems that it is IR controlled while the ZJ-WFMN-B V1.1 version (at least mine) is RF controlled.

After some trial and error, finally following template works for me:
`{"NAME":"MagicHome RGBW RF (ZJ-WFMN-B V1.1)","GPIO":[0,0,0,0,147,38,0,0,39,40,37,0,159],"FLAG":0,"BASE":34}`

Thanks for the support.

blakadder, 2020-08-14:
Thanks for the template ;)

JackyKNX, 2020-10-22:
I've got exactly same model and it works fine (all 4 channels) with the template 
{"NAME":"MagicHome RGBW RF (ZJ-WFMN-B V1.1)","GPIO":[0,0,0,0,147,38,0,0,39,40,37,0,159],"FLAG":0,"BASE":34}

However RF controller doesn't work for me (I've checked with the same module before flashing and RF works fine).

Any ideas ? Any rules necessary to make RF working ? 

Thanks ...